MarlinFirmware/Marlin #13333: [BUG] eeprom not shoing all settings marlin 2.0

jon012198, 2019-03-08:
it is only showing info steps filament advanced 
here is my configs
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2946418/Marlin.zip)
i am running marlin 2.0
[image]
[image]
[image]
[image]

RudyBenner, 2019-03-08:
I am also having problems with EEPROM not displaying all the settings.
When I connect with a terminal program (Putty) and do M501
the output looks like it may have extra line feed characters.
I have reverted to several older versions with the same result.
Yet, I have no problem with the new SBASE board.
I will attach some files.
Pronterface also shows the same botched up formatting.

I use Repetier Host

[BUG.zip](https://github.com/MarlinFirmware/Marlin/files/2947640/BUG.zip)

kAdonis, 2019-03-08:
Yes, I see this behavior

> like it may have extra line feed characters

in commit from 9. March
M501 is showing every parameter, but is difficult to read
I flashed marlin from 3. March and M501 is working as expected
(Ramps/Re-Arm)

jon012198, 2019-03-09:
Ok I am glad I am not the only one

RudyBenner, 2019-03-09:
Yesterday it was fine.

kAdonis, 2019-03-09:
@thinkyhead The cause for this issue is in #13311 
The new SERIAL_ECHOLNPAIR(...) sends an EOL after **every** pair
That is probably not as it should be, but I dont find an elegant way to fix this

p3p, 2019-03-09:
I can barely parse thinkyheads magic macros but I think moving the EOL to the top level macro should fix it, 

`serial.h line 135`
```c++
#define SERIAL_ECHOLNPAIR(...) do{_SELP_N(NUM_ARGS(__VA_ARGS__),__VA_ARGS__); SERIAL_EOL(); }while(0)
```

RudyBenner, 2019-03-09:
Thank you sir, we seem to be working again. 
rudy

closed 2019-03-09 by thinkyhead

fape, 2019-03-09:
I got compile errors:
Latest marlin2.0, latest paltformio. Only changes motherboard to BOARD_BIGTREE_SKR_V1_3 and serial1 to -1.

```
In file included from Marlin\src\inc/MarlinConfig.h:49,
from Marlin\src\Marlin.h:24,
from Marlin\src\Marlin.cpp:31:
Marlin\src\Marlin.cpp: In function 'void setup()':
Marlin\src\inc/../core/serial.h:115:64: error: expected 'while' before ';' token
#define _SELP_4(a,b,...)  do{ _SEP_2(a,b); _SELP_2(__VA_ARGS__); }while(0)
^

Marlin\src\inc/../core/serial.h:110:27: note: in expansion of macro '_SELP_4'
#define __SELP_N(N,...)   _SELP_##N(__VA_ARGS__)
^~~~~~
Marlin\src\inc/../core/serial.h:111:27: note: in expansion of macro '__SELP_N'
#define _SELP_N(N,...)    __SELP_N(N,__VA_ARGS__)
^~~~~~~~
Marlin\src\inc/../core/serial.h:137:32: note: in expansion of macro '_SELP_N'
#define SERIAL_ECHOLNPAIR(...) _SELP_N(NUM_ARGS(__VA_ARGS__),__VA_ARGS__)
^~~~~~~
Marlin\src\Marlin.cpp:913:3: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
SERIAL_ECHOLNPAIR(MSG_FREE_MEMORY, freeMemory(), MSG_PLANNER_BUFFER_BYTES, (int)sizeof(block_t) * (BLOCK_BUFFER_SIZE));
```

RudyBenner, 2019-03-09:
This error was corrected, unfortunately, he made some more changes and broke it.
Patience.

fape, 2019-03-09:
I know. I fixed it in #13343

RudyBenner, 2019-03-09:
I reverted that file. Now it compiles.

commit ef80a517 (2019-03-09): Fix SERIAL_ECHOLNPAIR

Fix #13333
commit c7d618b4 (2019-03-09): Fix SERIAL_ECHOLNPAIR

Fix #13333