MarlinFirmware/Marlin #13338: [BUG] mb ! +3(

guestisp, 2019-03-08:
marlin 1.1.8
I know it's an outdated version, but for no apparent reason, a 2:45 hours print was interrupted with the string `mb ! +3(` on the last line (Geeetech A20M)

As you can see from the attached photo, hotend and hotbed temperatures where still on.

Any idea ?

guestisp, 2019-03-08:
[image]

ManuelMcLure, 2019-03-09:
Printing from SD or through USB? Have you looked in the gcode file for that string, or even that the gcode file is complete?

thinkyhead, 2019-03-09:
Please test the `bugfix-1.1.x` and/or `bugfix-2.0.x` branches to see whether it helps. If the problem has been resolved then we can close this issue. If the issue isn't resolved in either of those branches yet, then we should investigate further.

guestisp, 2019-03-09:
Im unable to replicate and i don't know the source of this error (It this an error or just carbage shown on screen?)

guestisp, 2019-03-09:
Printing from SD, gcode is complete and that string isn't in the gcode file.
When it happened, printer was frozen (no movements) but hotend and hotbed still turned on

boelle, 2019-03-13:
@guestisp so this error only came up once?

maybe it was bad gcode and not a fault with marlin

guestisp, 2019-03-13:
I think it was a defective SD card.

closed 2019-03-13 by guestisp