MarlinFirmware/Marlin #13339: double quatation mark in german language file

crix26, 2019-03-08:
### Description

Line 237 in Language_de.h in 1.1.9_bugfix has an erroneous double quatation Mark at the End

### Steps to Reproduce

Try to compile Marlin 1.1.9_bugfix with german lcd Language

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]
Does not Compile
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-03-09:
is the problem the same in bugfix 2.0? that is where development happens :-)

FanDjango, 2019-03-09:
@boelle  Looks to me like @thinkyhead commited a fix on bugfix-1.1.x for this some 15 hours ago.

boelle, 2019-03-09:
yeah but 2.x is still what gets updated most

1.1.x is kind of on its way to the grave

closed 2019-03-09 by thinkyhead