MarlinFirmware/Marlin #13342: [TMC2130][PSU on/off] These drivers do not work with PSU off

Amachete, 2019-03-09:
Hello

After a lot of tinkering with these drivers and the wiring and a lot of reading, and googling, I have just realized that my setup does not work with these drivers very well.

My setup has a PSU on/off feature, so If motors are not necessary PSU turns off, in fact PSU is off by default that's why I have no response with the RAMPS only on USB.

JFI:  I have the PCB PROTECTORS pcb on.

In summary:
Marlin should advice if the PSU is OFF, there is not possible the communication with the drivers, so M122 will not work.

If you need some more information let my know.

I am also in the search for an electronic solution to these apart of solving the wiring problems that I have, so if someone can guide me,.....help me out!!!, I am desperate with this....

thank you all.

boelle, 2019-03-09:
attach 2 config files as a zip file

Ludy87, 2019-03-09:
How should Marlin know that the power supply is switched off?
The voltage / current from the USB is too low to control the drivers properly.
This is not a problem of Marlin, but of your energy management.

kAdonis, 2019-03-09:
I have been using different TMCs with AUTO_POWER_CONTROL enabled, and it is working several months now. You just need to make sure the drivers have 12V( or 24V) when booting marlin, so they can be configured properly. Afterwards you may switch the PSU off and on at your discretion, the TMCs stay configured.
My config is:
```
/**
 * Select your power supply here. Use 0 if you haven't connected the PS_ON_PIN
 *
 * 0 = No Power Switch
 * 1 = ATX
 * 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)
 *
 * :{ 0:'No power switch', 1:'ATX', 2:'X-Box 360' }
 */
#define POWER_SUPPLY 1

#if POWER_SUPPLY > 0
  // Enable this option to leave the PSU off at startup.
  // Power to steppers and heaters will need to be turned on with M80.
  //#define PS_DEFAULT_OFF

  #define AUTO_POWER_CONTROL        // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS           // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define POWER_TIMEOUT 30
  #endif

#endif
```
The only drawback is, as you said, M122 does not work when the PSU is off
a warning message from marlin in this case is not a bad idea

What might be a problem is:
Watterott (the stepper driver manufacturer) and Trinamic dont recommend to have Logic Supply Voltage connected **without** 12/24V Motor Supply Voltage, so there is a risk to damage the drivers, but it works for me a long time now. Watterott has these SilentStepstick-Protectors with a diode from Vio to Vm but I never used them

Amachete, 2019-03-10:
hello 

As requested my config files. I had to disable the LCD, in order to make probes.....

By default my setup has the PSU off. so Marlin should know that it needs to have the pin 12, in  state high before doing nothing with these drivers.

Let me know if you  need something else.

Bob-the-Kuhn, 2019-03-10:
The TMC2130 SilentStepStick uses the motor voltage to power the internal analog & digital logic.   The VIO pin just powers the I/O drivers.  SPI communications requires that both be present.

You **might** be able to put a Schottky diode between the motor voltage and +5V to provide power when the power supply is off.  Some possible problems with this method:
- The TMC2130 would be operating under it's recommended range so there's no guarantee.  
- Some power supplies look like a diode drop to ground when powered off.  You won't have enough +5V power to back drive this type of supply.
- There may be enough other things tied to the motor voltage that your +5V won't have enough muscle to drive it through the diode.

kAdonis, 2019-03-10:
@Amachete 
there are still no config files

Amachete, 2019-03-10:
Sorry, I was in a hurry and I forgot to press upload key!!!

:-)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2949804/Marlin.zip)

Amachete, 2019-06-14:
I will close this issue, because now with 

//#define PS_DEFAULT_OFF

I do not have problems with the TMC drivers....

closed 2019-06-14 by Amachete