MarlinFirmware/Marlin #13345: [BUG] BLTouch v3.0 not working properly with marlin homing

paulpenney, 2019-03-09:
### Description

BL Touch new 3.0 units not working with marlin on homing. Probe responds to M codes for deployment, stowing, and shows no errors, but when it comes to levelling the probe doesn't seem to properly communicate that the Z endstop has been reached. Tried on three different 3.0 probes out of a batch of 100. Tried on old and new marlin builds and machines with existing BL touch 2.1 units that function fine with the same results. 

### Steps to Reproduce

1. Install BL Touch 3.0 
2. Self test is OK and no flashing lights or errors. 
3. Start homing G28.

**Expected behavior:**

 Probe should hit bed and be sucked up, triggering the Z endstop. 

**Actual behavior:** [What actually happens]

Probe deploys properly and Z lowers to meet bed. The probe touches the bed and sucks up into the unit as expected and glows red, but the Z axis continues to move downward as if it didn't recognize the action.

#### Additional Information

This is on the latest build of marlin. 
Others on reddit report similar behaviour. 
Tested three new units out of an order of 100. 
Here is the new [DATASHEET](https://i.imgur.com/CtYJagJ.jpg) for the 3.0 BL touch, which hasn't made it to Antclabs website yet. 
This has been tested on machines that we know work with a BL touch 2.1. We home successfully with the 2.1, swap it for the 3.0 unit, homing fails, go back to 2.1 and its fine. That machine runs with marlin 1.1.9. Also tested on CR10S pro with InfinityAutomations firmware with the same results. No wiring connection issues. 

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2948383/Archive.zip)

InsanityAutomation, 2019-03-11:
Are you able to deploy the probe, run M43 W and tap the pin with a finger and see if the input pulses at all? Your orders are much bigger than mine! I thought I got alot when I got 5!! lol

paulpenney, 2019-03-11:
> Are you able to deploy the probe, run M43 W and tap the pin with a finger and see if the input pulses at all? Your orders are much bigger than mine! I thought I got alot when I got 5!! lol

Unfortunately I am not - Requires Pins Debugging and unfortunately on the ender there is zero room for any more code, literally bytes away from maximum capacity on the memory. I don't have another testbed on me currently. Still awaiting a solution from Paris in S. Korea. 

We are a pretty big shop with service and can burn through in a few weeks or so - people love the BL touch! I install probably at least one or two a day at the shop. 

Here is a video of the behaviour. https://drive.google.com/file/d/1zcYgOuyuHnCjl3d1abK5lXgTjrE1WS-d/view?usp=sharing Just to reiterate - it will report a Z triggered on an M280 P0 S60.

InsanityAutomation, 2019-03-11:
So first thought is the sensor itself having swapped polarity on signal (black / white) lines. But the line "it will report a Z triggered on an M280 P0 S60." mostly rules that out. Next thought is pulse width. Id need to get one to hook up to my scope to see if thats different, and potentially shorter causing the endstop noise filtering to ignore it.

paulpenney, 2019-03-11:
> So first thought is the sensor itself having swapped polarity on signal (black / white) lines. But the line "it will report a Z triggered on an M280 P0 S60." mostly rules that out. Next thought is [...]

Oh yeah the connections are fine - swapping in a 2.1 unit works perfectly, and we've tested four 3.0 units in a row now.

Paris did respond saying "After v2.2, because we use ceramic resonator, so we don't accept too large error than before." - Not sure what it means. I hope they didn't decrease the pulse width because it was already tiny at 5ms I believe. Also: On these tests, endstop noise filtering is not enabled.

Attached her is a better copy of the manual for the V3.0 
[BLTouchGuideV30.pdf](https://github.com/MarlinFirmware/Marlin/files/2953637/BLTouchGuideV30.pdf)

FanDjango, 2019-03-12:
Seems to me there are some changes from V2.2 to V3.0:
V2.2 Servo Timing:
[image]
V3.0 Servo Timing:
[image]
Although at first glance it looks downward compatible, I would consider playing around with the M280 commands on the V3.0 to see if EACH one works as expected.

FanDjango, 2019-03-12:
> aris did respond saying "After v2.2, because we use ceramic resonator, so we don't accept too large error than before."

Sounds like they might be more exacting on the incoming PWM ratio that determines the servo angle. As they have moved to an external resonator for the MCU of the BLTouch, they don't need such a big leeway in detecting the desired angles setting - and that also gives them to additional new positions to use for two new "command" (at 140° and at 150°.). If Marlin is slightly "off", something might go wrong.

FanDjango, 2019-03-12:
In both the 2.2 and the 3.0 datasheets, they state the pulse witdh for detection is 10ms.

paulpenney, 2019-03-12:
> In both the 2.2 and the 3.0 datasheets, they state the pulse witdh for detection is 10ms.

Thanks for all your insights above. I agree that it appears (without a scope to confirm of course) that the two new commands are causing some difficulty - as if not issuing one of those commands after an M280 Px S10 command doesn't arm the unit for sensing.

Can someone explain to me exactly how marlin activates the BL touch for sensing during homing? Looking at the code there are five definitions for PWM positions: 

` #define BLTOUCH_DEPLOY    10
  #define BLTOUCH_STOW      90
  #define BLTOUCH_SELFTEST 120
  #define BLTOUCH_RESET    160
  #define _TEST_BLTOUCH(P) (READ(P##_PIN) != P##_ENDSTOP_INVERTING)`

How and in what order does marlin use these in the sensing process? If the bl touch is deployed (10), does pushing it up automatically trigger an endstop event? And what is the purpose of Reset (160).

FanDjango, 2019-03-12:
Actually, after giving this some more thought, this:

> The probe touches the bed and sucks up into the unit as expected and glows red, but the Z axis continues to move downward as if it didn't recognize the action

points toward the probe actually performing its actions correctly. But it is the passing on of the touch-event to the controller board that is failing.

My thoughts: Configuration of pull-up action of the output stage in the probe is missing or insufficient. Even though it works for a 2.2 probe, maybe you don't have a pull-up strong enough (The probe is normal-state LOGIC-LOW according to the datasheet). You could actually measure (and compare to V2.2) with a multimeter in qiescent state. 

And they explicitly state the following in the new datasheet, as if it were of dire importance:

`#define ENDSTOPPULLUPS                         // BLTouch Smart V3.0 and Later  #`

From your description I would assume they have modified the output stage, therefore also the new commands to allow selecting the desired configuration and thus obviating the resistors or the little circuit board trace to cut when selecting 3.3V or 5V logic. Open-Drain works across all voltages, I suppose YOU should use the 5V Logic setting OR better, you should make sure you have actually activated your pull-ups on that input-pin. Which you have, according to your config files.

It's funny though: Why is (a little but further down)  your
`#define Z_MIN_PROBE_ENDSTOP_INVERTING false`

whilst on my machine:

`#define Z_MIN_PROBE_ENDSTOP_INVERTING true`

Aha! Maybe. So on your machine the endstops are COM to ground, NC to input pin. On my machine, 'tis normal OPEN. In that case maybe you should use the 5V logic setting of the probe instead of open drain (default, means you are missing the 140° init command) or consider changing #define Z_MIN_PROBE_ENDSTOP_INVERTING to true.

Worth a try?

paulpenney, 2019-03-12:
So  I just tried inverting the Probe Endstop and setting it to true -- no dice, same behavior. 

I agree with you that the probe is simply in the wrong state, I just don't know how to properly impliment the 5V logic setting with the 140 INT command. I am pretty sure I have to squeeze it in here between BLTOUCH_DEPOLY and BLTOUCH_STOW:

Marlin_Main.cpp line 2095: 
`      bltouch_command(deploy ? BLTOUCH_DEPLOY : BLTOUCH_STOW);`

The only problem is I have no idea how to interpret the syntax of that command and insert a 140 command. Its beyond my moderate level of arduino knowledge. Any ideas on your end?

paulpenney, 2019-03-12:
Scratch that, I'm not so sure anymore lol. Its a lot of complex code!

FanDjango, 2019-03-12:
Can you try a manual M280 Px S140 and then try a G28 Z? ?

FanDjango, 2019-03-12:
But try the S10 and S90 first, to confirm you have the right "x"

FanDjango, 2019-03-12:
I understand that perhaps the S140 setting might be needed EVERY time you deploy, but maybe it is kept in that state until next RESET or power cycle. Actually one should ask the ANTCLABS people all these things

paulpenney, 2019-03-12:
Tried all of the above, I definitely am addressing the right pin as M280 10 and 90 work great, and 60 returns a Z triggered when I look at the pinstates. I tried both S140 and S150 before homing but no luck, I am guessing its not programmed to save state. They responded to us on Sunday but we haven't heard back from them yet - I'll direct them to this thread! 

In the mean time - If I needed to insert those commands just after deployment for Z probing, any idea where that line of code might be? I can just insert a bltouch_command(140) wherever that is, but its a matter of finding it.

InsanityAutomation, 2019-03-12:
> Especially the 5V Logic ZMIN (option) position makes me a little suspicious - what the heck is it for? 

Originally they had a solder jumper for 5v or 3.3v logic. Seems they replaced that with the angle command to avoid damage due to people soldering in order to be more compatible with some 32bit boards that are 3.3v instead of 5v logic for inputs.

Big question now is how we can get a couple of these in developers hands to mess with for awhile. perhaps the support rep can confirm vendors in the US these have been sent to that I may be able to get one from? Worst case, maybe someone can bring one to MRRF?

paulpenney, 2019-03-12:
> Originally they had a solder jumper for 5v or 3.3v logic. Seems they replaced that with the angle command to avoid damage due to people soldering in order to be more compatible with some 32bit [...]

You may be able to get one from antclabs themsevles if they can't figure it out - try giving them an email at their gmail, the address is on the datasheet attached to a post above.

Bob-the-Kuhn, 2019-03-12:
>  it will report a Z triggered on an M280 P0 S60

This means the hardware is hooked up correctly.

The video shows that the Marlin software is not detecting the pulse from the BLTouch.

Long shot - Sometime enabling **ENDSTOP_INTERRUPTS_FEATURE** is helpful. 

---

The basic BLTouch probing and homing sequences are the same:
1. Deploy probe.
2. Endstop code samples the probe input.  This phase will be aborted if probing and Z get too low.
3. When two consecutive active samples are detected the BLTouch is considered to have been activated.
4. The probe is then stowed and the Z stepper(s) are stopped.
5. Depending on the options selected, the Z axis may be re-positioned and steps 1-4 redone.

___

Is there US stock of the new BLTouch?

A quick look on Amazon didn't show anything claiming to be V3.0

Anyone willing to part with a V3.0?   I can use PayPal to cover the cost & shipping.

InsanityAutomation, 2019-03-12:
Just sent an email out. Ill update here what I get back. So far, it sounds like we need to initialize the collector on startup and reset based on the state of the endstup pullup, and potentially add a config option to override and menu button to resend. From there, testing on it processing signals correctly needs to be performed.

paulpenney, 2019-03-12:
> Long shot - Sometime enabling **ENDSTOP_INTERRUPTS_FEATURE** is helpful.

One of the first things I tried unfortunately. No cheddar. 

I'm not sure if US stock of BL touches, it appears we are basically at the release stage for them - at first I thought they were fakes, no reference on the website at all, but we're in touch with Paris and she says they are legit and redesigned.

InsanityAutomation, 2019-03-12:
> One of the first things I tried unfortunately. No cheddar.
>
> I'm not sure if US stock of BL touches, it appears we are basically at the release stage for them - at first I thought they were fakes, no [...]

Dont you love being patient zero? lol Pretty bad to get hit with that qty of them as well...

paulpenney, 2019-03-12:
I'm just playing around here and hacking away at code, but I have some results that may offer some clues into the cause of the issues with 3.0: 

In Marin_Main.cpp at line 3133 / 3171

  #if HOMING_Z_WITH_PROBE && ENABLED(BLTOUCH)
    // BLTOUCH needs to be deployed every time
    if (axis == Z_AXIS && set_bltouch_deployed(true)) return;
     **bltouch_command(60);**
  #endif

If I manually add in a BL touch 60 command (Test) it will actually correctly sense the endstop properly, although it acts jankier because I am sure the code isn't supposed to work this way. But either way, it is putting the Bltouch in a state to report the endstop properly to marlin and indeed it gets sensed. If I change these values to 140 or 150 though, all that happens is the plunger gets sucked up, stays sucked up, and the Z axis keeps dropping.

Bob-the-Kuhn, 2019-03-12:
I was the one who came up with the basic system in use today to reliably interface to the BLTouch.  I'm very familiar with the low level hardware and software and I have the tools needed to see what the hardware is doing.

There a 90+% probability I can have a test image available within 6 hours of getting my hands on a unit.

Once I have code that works on a couple of different CPUs I'll do a PR so that people can grab a copy and try it.  Once we get feedback from several people then we'll move the changes into the bugfix branches.

@Roxy-3D - do you still have the direct contact info?

@paulpenney - are you willing to part with one of your units?  If yes please send it to:
Bob Kuhn
1034 Wisteria Trail
Austin, TX 78753
(512) 934-1943

I can use PayPal to send you $$ to cover your costs and/or return it to you..

Bob-the-Kuhn, 2019-03-12:
From my experience with the V2.0 units:
- The 3.3V/5V option just selects what voltage the output pulse will be.  
- You still need to supply 5V to the unit to have reliable operation.. Running the unit at 3.3V will result in unreliable operation.

paulpenney, 2019-03-12:
Bob, I would try Paris @ antclabs@gmail.com to see if she can ship you one, I'm sure that she has a lot of stake in getting this integrated with marlin as well considering that a lot of her customer base is people upgrading enders, CR10s, D9s and all the other marlin-based printers

InsanityAutomation, 2019-03-13:
As a quick update, antclabs did not respond to my email as of yet however another distributor who got hit with the same bundle of new sensors and is seeing the same problem is getting one over to me. I'll have it hooked up as soon as it gets here and hopefully have something working this weekend that we can merge giving everything that we've seen here. The aim will be to keep them plug and play with something that does not cause an issue with 2.1, and I have a bunch of those to swap back and forth however I will add a configuration option if needed. Given that these are being sold as direct replacements, there may be some merit to a run-time option if needed however I really hesitate to do that and hopefully I can find a way around it.

paulpenney, 2019-03-13:
> As a quick update, antclabs did not respond to my email as of yet however another distributor who got hit with the same bundle of new sensors and is seeing the same problem is getting one over to [...]

UPDATE: We got a positive response from Antclabs. The pin C7 capacitor leading to the Z endstop needs to be removed from the board with pliers. Doing this solved the problem, at least for the Enders. This is not the ideal solution though, as it permanently modifies the board, so we're hoping there is a software solution that will crop up.

paulpenney, 2019-03-13:
Or maybe its a resistor. Anyways its the bigger yellow component.

FanDjango, 2019-03-13:
Removing a C7 capacitor. Hmmm. If removing it should help, it very probably (but not totally necessarily) is between ground and the output signal as a slight noise filter. WITH this capacitor, a 10ms signal might have less distinct rising and falling edges, thereby reducing the timeperiod in which the signal is above the logic 1 threshhold. Perhaps the samples that Marlin takes are then too far apart?

FanDjango, 2019-03-13:
This reminds me of #13128 where the OP solved his noise problem by putting a small cap across the signal and ground. Works on that machine, but who knows how small that cap may be. Now ANTClabs have put a cap (same value or more?) on by default (did they read that thread?) and that seems to be detrimental for "normal" Marlin machines.

FanDjango, 2019-03-13:
Could you post the original text of their reply and also post a foto of the V.30 PCB with the C7?

InsanityAutomation, 2019-03-13:
I can measure the component with my lcr-meter and put a scope on it before and after to see what it does to the pulse width. I should get the part here sometime tomorrow, but I'll do a little bit of logic testing to see what I can do with it un-modified first.

InsanityAutomation, 2019-03-15:
just a quick rig up to my scope, didn't really set up any triggering and this is an ancient analog scope without logging anyway. Definitely seeing the curve on the pulse. Overall length appears to be the same however the slope on front is definitely cutting into the duration.

[image]
[image]

Bob-the-Kuhn, 2019-03-15:
I still don't see the problem.

Looks like about a 2mS rise time.  If it's still a 10mS pulse then we should have 8mS at logic 1 which is well above the 5mS Marlin is designed to catch.

Strange - the pulse height is only 2V.  I would expect more.  What happens if you put a 1K pullup on it?

FanDjango, 2019-03-15:
If the unit is in "Open-Drain" Mode per default, shouldn't the configured pull-up of the input-port-pin "pull it up" further?
Alternatively, would a switch to 5V mode, via 140° servo make the situation better?

InsanityAutomation, 2019-03-16:
Pull request is up! I dont have a 3.3v logic level board here to verify 100% with, but looks to work plug and play between versions now. Added menu items and everything.

FanDjango, 2019-03-16:
So it actually was necessary to utilize the new BLTouch V3.0 mode commands. Sheesh, they sure risked it not working on a huge percentage of the printers out there (MARLIN pre todays bugfix vers), what got into them?

juliandroid, 2019-03-16:
I've received BLTouch 3.0 yesterday, but I was expecting actually 2.2 (it wasn't published on the Antclabs site). However I have received a feedback from Creality support that their latest firmware 1.60.7 is not stable - and it was mentioned on their official video as counterpart to their BLTouch kit for Creality printers.
Today, I've checked and they don't show 1.60.7 as downloadable version, but 1.60.71 instead. It is interesting to know whether they fixed the same issue or whether their stability issue was something else.
@InsanityAutomation Could you please confirm that this new commit should be enough fix the issue with homing?

Roxy-3D, 2019-03-20:
@paulpenney @FanDjango   Can you give me your shipping addresses?  Antclabs would like to send the people in this thread a Bl-Touch v3.0 to help insure their probe is well supported.    @Bob-the-Kuhn  (I already have your address!)       

There are two easy ways to get me your address without posting it publicly here.   You can join www.3dprintboard.com and send me a private message with it.   Or...  You can bring up Discord, join the Marlin server, and send me a private message there.

FanDjango, 2019-03-20:
@Roxy-3D: I sent you a discord as requested

paulpenney, 2019-03-20:
@Roxy-3D I sent you a PM on the 3d Printing Forum.. your name there is simply "Roxy"? Thanks again,

Paul

Roxy-3D, 2019-03-20:
Thank You.   I've sent everybody's address to Antclabs.   I'm guessing it will take several weeks for the v3.0 Probes to arrive.

CWSpear, 2019-03-20:
Dang... looks like I might be another "victim" of issues with BLTouch v3.0. As seen in the pic, that means I've got 3.0, yeah?

[image]

I bought it off Amazon: https://www.amazon.com/gp/product/B076PQG1FF/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

I am pretty sure my issues are the same as @paulpenney, but my observation was slightly different:

He wrote,

>Probe deploys properly and Z lowers to meet bed. The probe touches the bed and sucks up into the unit as expected and glows red, but the Z axis continues to move downward as if it didn't recognize the action.

I would say that mine continues gets stowed (sucked up), but then redeploys ~half a second later and then stows again when it hits the surface. It's at _this_ point that it registers the Z endstop or whatever and stops going down. So it won't drill into the bed indefinitely. I can hold down the plunger as its testing and it will definitely stop. but it's like it takes half a second of it being unable to fully deploy for it to register.

CWSpear, 2019-03-20:
This thread is a bit technical for me... is there a (temporary?) fix to get the v3.0 sensors to work correctly that someone could sum up?

FanDjango, 2019-03-20:
Official fix according to the vendor for now seems to be to remove a capacitor (see above). Not recommended, can damage the hardware and difficult to undo.
Next recommended possible fix: Use the newest bugfix-2.0 download to get the V3.0 support (add the needed config statements). This means downloading, modifying the configuration files accordingly and uploading to your printer. @InsanityAutomation has got this working and it wouldn't be committed if it weren't worth trying 🥇 

I am still trying to get a V3.0 here somewhere through normal channels (out of curiousity) but no chance. I still find it difficult to believe that the normal pre-BLTouch V3.0 Marlin does not work with the probe when pin-pull-ups are enabled. If it needs the new BLTouch V3.0 commit (which sets 5V mode instead of open-drain-mode), then as a next step I would like to see a schematic of the output stage (is it more than just the output pin of the ATTINY13A, what resistors and capacitors are on that output pin?). Can be measured but I can't find a V3.0 here, as I said.
It's sort of time to stop experimenting and to get down to some hard facts from ANTClabs. I know they need some secrecy because of the blatant cloning that they are experiencing but maybe just a small part of the details might help.
I myself am equipped with both 5V and 3.3V boards and V2.x BLTouches, they work fine with Marlin. Got some on order but who knows what version will arrive.

CWSpear, 2019-03-20:
I have a CR-10 for what it's worth.

I installed the latest from the `bugfix-2.0.x` branch, and set up the BLTouch, uncommenting out `BLTOUCH_FORCE_5V_MODE` and `BLTOUCH_V3`.

It didn't seem to make a difference.

Also, as a side note, when I hit the different options in the new menu, I see stuff like,

```
echo:enqueueing "M280 P0 SBLTOUCH_5V_MODE"
```

in the serial monitor log, when I would expect

```
echo:enqueueing "M280 P0 S140"
```

Maybe something isn't getting converted correctly in the code? This is true for all the commands, like self-test, stow and deploy. I can execute the commands directly and they work as expected.

But I'm still seeing the delayed response, even if I send `M280 P0 S140` to ensure it's in 5V mode.

CWSpear, 2019-03-20:
Ok, scratch that. It does seem to be working when I do a `G28 Z` 🎉 

Does still seem to be some kinks, such as the issues I listed above and it doesn't seem to want to take my Z offset into consideration, but maybe I'm doing something wrong (first time with 2.x firmware) (is `G28 Z` supposed to take into consideration Z offset? This is actually the first time I've really gotten this far).

InsanityAutomation, 2019-03-21:
@CWSpear I just opened a pull request to fix the menus.

For a Creality machine, can try my pre-configured branch here - its caught up the the current changes and open PR's

https://github.com/InsanityAutomation/Marlin/tree/Creality_2.0_Devel

Roxy-3D, 2019-03-21:
@paulpenney @FanDjango 

I apologize!    It turns out I also need to give Antclabs your phone numbers so they can fill out the Custom's forms.

juliandroid, 2019-03-22:
@InsanityAutomation My printer is CR-10S Pro what is the easiest way to back port BLTouch 3.0 commits?
Currently Creatily offers only source code of V1.60.3 (Marlin1.1.6) : https://www.creality3d.cn/download/source-code_c0001

Any suggestions?

InsanityAutomation, 2019-03-22:
@juliandroid Ill have a patch in the Tinymachines builds in the next couple days or so, and when its proven will be published as B6. Testing is in progress now :)

Bl0wnp0ny, 2019-03-22:
@InsanityAutomation 
looking forward to the tinymachines fix for my CR-10S Pro.  I ordered my BLtouch on Amazon.  I just installed it and spent a few hours tracing the wiring and troubleshooting.  Finally found this thread not knowing there was a new version of the BLtouch.  Sure enough I have V3.0.  I feel like Antclabs did a terrible job on this rollout of new hardware.  I'm down one machine on productivity.

Kaisertron, 2019-03-23:
So I think this must be the same problem Im getting. Video here https://youtu.be/eAfLy4Y4R9M

Bl0wnp0ny, 2019-03-23:
> So I think this must be the same problem Im getting. Video here https://youtu.be/eAfLy4Y4R9M

yep.  the sensor will say V3.0 on the side with the connector.

Kaisertron, 2019-03-23:
Yep sure does. It's a version 3.0. Ordered off Amazon. Just for clarification the fix is listed above, correct?

InsanityAutomation, 2019-03-23:
@Kaisertron grab the code from the open pull request. Should have a v3 running nicely.

Kaisertron, 2019-03-23:
@InsanityAutomation ok sorry if I sound like an idiot I just need to flash the firmware in the current state on the page you listed or change setting for my Ender 3. Besides the obvious x and y probe offset.

Bob-the-Kuhn, 2019-03-23:
I did some playing with a couple of the V3 units.  As best I can tell the main differences between the V3 and V2 are:
- Open drain is the new default rather than a totem pole output (this was reported earlier by others)
- The output driver on V3 (after issuing S60 then S140 commands) is very weak compared to the V2.  

I measured the following output voltages with a resistor to ground on two V3 units and a V2.  The V3 was given S60 then S140 commands before taking the measurements.
```
 R      V3       V2
open   4.47V    3.95V
10K    0.96V    3.90V
1.2K   0.15V    3.59V
```

These numbers translate into an output impedance of about 30K on the V3 and 0.1K on the V2.

Roxy-3D, 2019-03-23:
Wow!   That concerns me.   30k ohms on a signal wire that is going be run next to stepper motor wires and heater wires makes me think it will be picking up electrical noise.

Bob-the-Kuhn, 2019-03-23:
30K is OK if there are no capacitors on the line.

Noise pickup on long lines is also a concern.

On the other hand it would be similar to a mega2560 & RAMPS with mechanical switches and no external pullup resistors.  In that case the only pullup would be the Rpu.  I don't know if that is a common system or if it has a reputation for false/missing triggers.

[489 more replies]

closed 2019-06-14 by thinkyhead