MarlinFirmware/Marlin #13346: Conflicting declaration 'volatile bool wait_for_user'

rschaeuble, 2019-03-10:
### Description

Commit 60e82e3929bfd402384e7dea9d737796c96fa07f introduces a compile error:

```
In file included from ...\sketch\src\HAL\HAL_AVR\MarlinSerial.cpp:58:0:

...\sketch\src\feature\emergency_parser.h:35:22: error: conflicting declaration 'volatile bool wait_for_user'

 extern volatile bool wait_for_user, wait_for_heatup;

                      ^

In file included from ...\sketch\src\HAL\HAL_AVR\MarlinSerial.cpp:44:0:

...\sketch\src\marlin.h:336:15: note: previous declaration as 'bool wait_for_user'

   extern bool wait_for_user;
```

In `emergency_parser.h`,  `wait_for_user` is declared as volatile. Since the latest commit, `Marlin.h` now declares the same variable as non-volatile. This results in a conflicting duplicate definition, which the compiler rejects.

I have tried both Arduino IDE 1.8.8 and 1.9-beta, with identical results.

noderat, 2019-03-10:
Confirm, I am also getting this duplicate definition error.

AbuMaia, 2019-03-10:
Another confirmation.

Vertabreak, 2019-03-10:
confirmed.

Moyster, 2019-03-10:
fixed here : https://github.com/MarlinFirmware/Marlin/pull/13349

closed 2019-03-10 by thinkyhead