MarlinFirmware/Marlin #13347: Connecting arduino mega to serialport for external controls

wookie666, 2019-03-10:
Trying to connect an arduino mega to serial ports AUX1 on MKS GEN L board. 
I have tried a few different configs 

```
/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1 // is the USB port

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 0 // is the serialport for the TFT
```

I have the baud rate correct on both units. 
they match

I have tried serial port 0 and serial port 2 0 but it still wont work. What do i need to do to be able to communicate with my board using serial.

7eggert, 2019-03-10:
Did you remember to swap RX and TX?
Also: https://github.com/MarlinFirmware/Marlin/issues/10219
HTH

wookie666, 2019-03-10:
Yep they were correct

boelle, 2019-03-13:
@wookie666 read this link: https://github.com/MarlinFirmware/Marlin/issues/10219

maybe it can help you

closed 2019-07-11 by thinkyhead