MarlinFirmware/Marlin #13351: [BUG] bugfix-2.0.x mixing extruder and filament change feature do not work seamlessly

brandstaetter, 2019-03-10:
### Description

When mixing extruder and filament change features are enabled, the filament change menu shows entries for E1 and E2, but klicking on E2 does nothing. E1 moves the extruders according to the defined mix

### Steps to Reproduce

1. enable ADVANCED_PAUSE_FEATURE
2. enable MIXING_EXTRUDER with MIXING_STEPPERS 2
3. navigate to the menu Filament change

**Expected behavior:** Entries for E1 and E2 really work on the specified steppers and ignore any mixture settings

**Actual behavior:** E2 does not work, E1 moves the steppers according to the configured mix settings

(Ping me if I need to attach my configs, don't think it's necessary here)

Vertabreak, 2019-03-10:
Confirmed 
I would add if you set mix 100% to E1 or E0 and use E1 filament change the one set 100% is used.

tcm0116, 2019-03-26:
I've noticed this same thing. It's been on my to-do list to look into it some more. I think the main thing to consider is that the filaments need to be changed one at a time as pushing too much filament into a mixing extruder when one of the other inputs is empty will create a huge mess!

tcm0116, 2019-04-06:
@brandstaetter - I was working on fixing this when the control board in my machine blew up :-(

I have pushed the changes to the branch linked below, if you want to give them a try. `M600`, `M701`, and `M702` should work as expected. `M702` also supports `FILAMENT_UNLOAD_ALL_EXTRUDERS` when no `T` parameter is provided, and should unload all of the filaments simultaneously. The LCD code was already trying to erroneously send T parameters to the filament change commands, so it should work from the LCD and via the manual G-Codes.

https://github.com/tcm0116/Marlin/tree/2.0.x-mixing_filament_change

boelle, 2019-06-22:
missing label

thinkyhead, 2019-07-11:
Attempts have been made to fix up this combination, but mixing is still a bit of a hack, so some work may still be needed. Please test the latest 2.0.x code to see where it stands.

boelle, 2019-09-24:
@brandstaetter have you tested with latest 2.0.x ?

brandstaetter, 2019-11-01:
Printer is still non-functional, no time to fix it unfortunately. Is there noone else who could test it?

boelle, 2019-11-01:
i personal have problems with just the filament change (M600) option on a normal extruder

it skips the unload option, and i have posted on discord about it but so far no luck

my configs are also posted on discord in the  #marlin-v2 channel

boelle, 2019-12-02:
@shitcreek @thinkyhead could it be that PR #13803 broke this?

boelle, 2019-12-07:
@brandstaetter since 2.0 was just released a few days ago has this changed this issue at all?

Vertabreak, 2020-01-19:
i would note that it does function its just weird and labeled funny.

boelle, 2020-01-19:
yep, my org problem is also gone, it works as its programmed

boelle, 2020-01-24:
will close as i tested it and works

closed 2020-01-24 by boelle