MarlinFirmware/Marlin #13353: [BUG] reprap stepper expander X3

bjbeare, 2019-03-10:
### Description

I am using the stepper expander X3 for my (three stepper) Z axis. Due to a pin conflict, I have to use a custom cable... it would be nice to figure out how to do this with a standard ribbon cable to the AUX1 and AUX 2 connectors.

The conflict is with the X3-DIR pin on the stepper expander. This pin is connected to D2 on the MEGA connector... the MEGA connector labels this as TX (the serial port). I don't use the serial port (just the USB) for communications with Marlin. The MEGA 2560 schematic labels this pin as TXD0(PE1). How can I tell Marlin (pins_RAMPS.h) to leave this pin for X3_DIR rather than configure it for a UART?

### Steps to Reproduce

#define Z_DIR_PIN       59
#define Z_STEP_PIN      64
#define Z_ENABLE_PIN    44

#define Z2_DIR_PIN      63
#define Z2_STEP_PIN     40
#define Z2_ENABLE_PIN   42

#define Z3_DIR_PIN      2
#define Z3_STEP_PIN     57
#define Z3_ENABLE_PIN   58

**Expected behavior:** [What you expect to happen]
X3_DIR on the stepper expander would flip between 0 and +5 when I issue
G0 Z10
G0 Z-10

**Actual behavior:** [What actually happens]
X3_DIR on the stepper expander does not alter level.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

7eggert, 2019-03-10:
According to this, it's pin 1, not pin 2 (which is the Z_MAX_PIN)
reprap.org/wiki/Stepper_Expander#Using_RAMPS
reprap.org/wiki/File:SE-RAMPS.JPG

bjbeare, 2019-03-11:
Yes, I've tried pin #1 in the #define. The 2 there was a cut-paste from an experiment I ran (I should have corrected it before posting, missed it). I'm currently using #3 (x-min pin) and turned off the pullup to access the pin.

Pins 0 and 1 are routed to AUX1.. but are RX and TX and seem to be already in use elsewhere in Marlin.

... I'll correct the original post's #define.

AnHardt, 2019-03-11:
The assumption RX and TX would be free for alternate uses when using USB, on a Arduino MEGA, is wrong. 
USB on the Arduino MEGA is realized by an extra chip. This chip is connected to the 2560 by the serial lines.

bjbeare, 2019-03-11:
Thank you @AnHardt 
The stepper expander 3 isn't "perfect" for the 2560 then. A cable modification is, in fact, required and I have the correct approach.

I'll close the issue.

closed 2019-03-11 by bjbeare

lloydw88, 2020-07-05:
Im using stepper expander x2, I want it to control the 2 extra stepper motors.  Im relatively new to this so Im reaching out for help.  Ive got it hooked up corretly as far as I know.  I have the ramps 1.4 board, a thc-01 hotend.  ive gotten hotend working and reading, fan works as well.  It moves on all axis.  I believe i had the 2 extra motors wired wrong at first, but i was able to switch extruders in repetier and one of the 2 spun when i pushed the extrude button.  the other did not.  also one was hot.  so I  got looking at connections and realized I had one backwards.  now that they are wired right, they arent getting hot but i get no motion out of them.  Any help would be appreciate.

lloydw88, 2020-07-15:
> Thank you @AnHardt
> The stepper expander 3 isn't "perfect" for the 2560 then. A cable modification is, in fact, required and I have the correct approach.
> 
> I'll close the issue.
Is this error on the stepper expander x2 as well?  when i take a multimeter to the pins, for the first stepper, 2 of the pins show me 12v and for the second stepper I only get 12v from one of the pins.  Is this normal?  shouldnt there be 2 12v sources per stepper motor?  I think my problem is in the stepper expander x2, but not entirely sure if i verified right.  I believe all my wires are hooked up correct, and the pins assigned correctly as well, just cant seem to get all 3 motors turning.  I have switched the cables to test all stepper motors, and all stepper motors work.  As I stated coming off the x2 stepper expander board, the 2a, 2b, 1a, and one 1b, on one motor, I get 2 12v coming off one set of pins out of the 4 for the stepper motor, and on the other set coming off the board I only get 1 with a 12v reading.  Any help would be much appreciated.