MarlinFirmware/Marlin #13354: Problem With Simplify3d

kaijrock23, 2019-03-10:
Hi, I have a little problem and I feel sorry to bother you but I really don't know how to solve, I own an ender 3 pro, I printed different things with both cura and simplify3d using the exact same settings profile and all printed perfectly so far.
I upgraded from stock Marlin to Marlin 1.1.9, and now I have issues with my prints quality, solved with cura with "control jerk" turned off, but I have still problems with simplify3d.
I don't know how to call it, raw surface or a bit of layer shifting.
Here is a photo of the issues that I encounter with simplify3d is the unfinished model I did just to show the raw surface/layer shifting.
The second photo is the same model with the same settings printed in cura.
I tried using default creality's jerk value and the ones that came with the Marlin 1.1.9 no success, simplify3d just have that behavior, ringing/layer shifting/raw surface I don't even know how to classify this issue.
[image]
[image]
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2949720/Marlin.zip)

boelle, 2019-03-10:
try bugfix 2.0 also, 1.1.9 is old and not updated as much

2.0 is both 8 bit and 32 bit

kaijrock23, 2019-03-10:
I have bed levelling and other small things added, since the bugfix 2.0.0 contains less files, it's safe to add just what I need and then flash over the 1.9.0 bugfix that I have already installed?

kaijrock23, 2019-03-10:
Arduino:1.8.8 (Windows Store 1.8.19.0) (Windows 10), Scheda:"Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

sketch\src\lcd\menu\menu.h: In function 'load':

sketch\src\lcd\menu\menu.h:165:18: internal compiler error: Segmentation fault

     static void  load(void *ptr, const int32_t value) { *((type_t*)ptr) = unscale(value);     }

                  ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.19.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1

If I verify the marlin 2.0bugfix it gives me this issue.

> try bugfix 2.0 also, 1.1.9 is old and not updated as much
> 
> 2.0 is both 8 bit and 32 bit

boelle, 2019-03-10:
bugfix 2.0.0 contains same amount of files, they are just split up differently

also upgrade to arduino ide 1.9 or vscode+platformio

InsanityAutomation, 2019-03-11:
Arduino IDE 1.8.8 is fine, just get rid of the windows store version. It is known to be buggy and crash compilation very often. Please download it directly from the Arduino website.

kaijrock23, 2019-03-16:
I upgraded to Marlin 2.0, same exact issue with simplify3d, with cura still all ok.

kaijrock23, 2019-03-16:
I always used the tested setup file in the default configuration files.

kaijrock23, 2019-03-17:
Does simplify3d use jerk and acceleration from the printer or from it's own settings?

robbycandra, 2019-04-03:
Actually you can compare the gcode file produced by cura vs simplify3d and see the differences between them.

kaijrock23, 2019-04-03:
How?

guestisp, 2019-04-05:
I think I have the same here.
Upgraded from 1.1.8 to 2.0.x and now surfaces are rough and "shifted". Slic3r exibit this issue, Simplify3D (with the Benchy) much, much less (but not perfect), Cura a little bit cleaner than Slic3r, but worse than S3D.

[image]

kaijrock23, 2019-04-06:
With cura I solved turning jerk control and acceleration off

guestisp, 2019-04-06:
So you rely on firmware accelerations. which settings do you have ?

kaijrock23, 2019-04-06:
[image]

kaijrock23, 2019-04-06:
I use the 1.1.6 ones they are from creality.

guestisp, 2019-04-06:
I don't think it's the slicer acceleration control to add these artefacts. My slicer use the same acceleration and jerk values as per firmware, so leaving them disabled won't change anything

I've started to get these trouble when using Marlin 2. I'll try to revert to Marlin 1.1.8 and see

kaijrock23, 2019-04-19:
Guestisp ti posso chiedere una cosa? Ma in simplify hai tipo negli script un gcode con scritto REPLACE M204 S ecc?

guestisp, 2019-04-19:
No, perché?

kaijrock23, 2019-04-19:
Perché sto continuando a fare prove, con cura ho risolto del tutto, disattivando jerk e acceleration lasciando il tutto nelle mani di Marlin, in simplify in starting gcode ho messo tutti i parametri presenti in Marlin, e sembra aver risolto il problema, in post processing avevo pure un gcode m204 S con valori alti e in più S non si usa più. Per questo chiedevo, visto che nessuno ci ha saputo aiutare ho continuato a cercare una soluzione concreta al problema.

guestisp, 2019-04-20:
Ma hai anche te quelle bande orizzontali usando Slic3r ? Considera che Slic3r di default fa override dei settaggi del Marlin, quindi nelle impostazioni della stampante, su Slic3r, devi impostare gli stessi valori che hai in marlin.

kaijrock23, 2019-04-20:
Non uso slic3r, quindi non saprei, dimmi i settaggi che usi magari c'è qualcosa di troppo alto.

guestisp, 2019-04-20:
500 di accelerazione, junction deviation 0.02

kaijrock23, 2019-04-20:
Hai provato usando il jerk invece di junction? X e y 10, E 0.3 e Z5?

guestisp, 2019-04-20:
Si, non cambia nulla.

closed 2019-07-11 by thinkyhead