MarlinFirmware/Marlin #13358: [BUG] Level corners mode stays active after display timeout

Santobert, 2019-03-10:
### Description
The level corners mode stays active after timeout has reset the display and the info display is shown. After pressing the button again, the printer goes to the next corner and shows a weird mixture of the level corners menu and the main menu:
[image]

### Steps to Reproduce
1. Select level corners
2. Go through the process until the printer reached the first corner
3. Wait there until the timeout resets the display and the info display is shown
4. Hit the button

**Expected behavior:** When hitting the button from the info display the main menu should be shown

**Actual behavior:** The printer travels to the next corner and a weird display is shown

#### Additional Information

shitcreek, 2019-03-12:
Could you select 'back'? What version of Marlin are you using?

Santobert, 2019-03-12:
Yes, I can select 'back' and it works as expected. It leads back into the 'level bed' menu.
I am using the latest marlin 2.0 bugfix version.

boelle, 2019-03-13:
maybe attach your 2 config files as a zip file?

Santobert, 2019-03-19:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2982763/Marlin.zip)

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
This has been fixed, so this one can be closed.

closed 2019-07-17 by Santobert