MarlinFirmware/Marlin #13359: [BUG] (short description)

ruggb, 2019-03-10:
### Description
f6ab62b with RH Repetier Host
CoreXY with 3 point ABL and nozzle Z endstop

I am not sure how I can describe this but I will try
I do a G28 - everything good
I do a G29 - everything seems good till it starts to go park then it appears totally lost and makes a lot of noise as in maybe stepping too fast. 
At this point it appears lost. However, if I repeat G29, Marlin seems to know where it is, but RH does not. Marlin appears to be telling RH that Y =150 and X = 4.99. It is actually suppose to be X= 5 and Y = 295 and it appears close to that.

G27 works as expected, however.......
If I do G28, then G27, at this point RH does not know where it is == says 0, 0
any step toward 0 for X or Y will run it all the way back to 0.

I see nothing in the RH log where it is receiving a position from Marlin after a G29 or G27

[mod files.zip](https://github.com/MarlinFirmware/Marlin/files/2949993/mod.files.zip)

boelle, 2019-03-10:
edit tittle

right now it's: [BUG] (short description)

thinkyhead, 2019-03-10:
Repetier Host tends to hide position and temperature updates in its console. When troubleshooting, just use the dumb Serial Monitor in the Arduino IDE so that you see *all* output.

thinkyhead, 2019-03-10:
Try adding this line to the end of `Nozzle::park` in `nozzle.cpp` and see if it fixes the issue:
```diff
        do_blocking_move_to_z(MAX(park.z, current_position[Z_AXIS]), fr_z);
    }

    do_blocking_move_to_xy(park.x, park.y, fr_xy);

+   report_current_position();
  }
```

ruggb, 2019-03-11:
Not sure what you are telling me

  void Nozzle::park(const uint8_t z_action, const point_t &park /*= NOZZLE_PARK_POINT*/) {
    const float fr_xy = NOZZLE_PARK_XY_FEEDRATE;
    const float fr_z = NOZZLE_PARK_Z_FEEDRATE;

Where does it go?

ruggb, 2019-03-11:
sketch\src\libs\nozzle.cpp: In static member function 'static void Nozzle::park(uint8_t, const point_t&)':
complie error

sketch\src\libs\nozzle.cpp:184:33: error: wrong type argument to unary plus

     +   report_current_position(); //** added per Thinkyhead
                                               ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

7eggert, 2019-03-12:
The plus is not to be included in the file but a sign to indicate the line that needs to be added.

The diff utility (or git) will produce short snippets with these signs (and information about which file and line numbers) and the patch utility is able to apply these changes. Developers use it all the time and thus expect people to know, too.

ruggb, 2019-03-12:
Bad expectation!

That fixed the G27 issue, but G29 is still lost. When doing the park it still grinds away and shows what appears to be the last probe test position or there about, but it is at/near park position.

ruggb, 2019-03-12:
maybe that report line is needed at the end of G29 BEFORE a park is executed.

ruggb, 2019-03-12:
It does that on the park command I have as the end code in Repetier also.

ruggb, 2019-03-15:
at the end of a print it appears to go to the park position then vibrate in the Y direction. X=5 and it is not hitting the X stop.
It says it is at Y=295 and it still has open space == it can actually go farther than 300 which is the bed size I have set.

closed 2019-03-15 by ruggb