MarlinFirmware/Marlin #13361: [BUG] Junction Deviation increases ringing on first segment of layer

lekv, 2019-03-10:
### Description
I'm trying to get rid of ringing on my Ender 3 Pro. Setting Junction Deviation to 0.01 resulted in very little ringing but caused bulging corners (this is expected). Increasing JD to 0.02 and then 0.05 causes ringing on the X-axis only on the starting segment of the layer:

I'm printing [this 25mm cube](https://www.thingiverse.com/thing:2267549) with .4 line and wall width, .2 layer height, at 35mm/sec for both the first layer and the wall.
```
View from the TOP
    v-- ringing happens here
+----+  <- Layer starts here, printing counter clockwise
|    |
|    |
+----+
 ^-- ringing does not happen here
```
The next image shows how the ringing increases at two points towards the top of the cube, when I changed JD to 0.02 and then to 0.05.
[image]

This image shows the other X-axis side of the cube where corners got smoother (hard so see in the pic) but the ringing did not increase.
[image]

### Steps to Reproduce
Print the cube as described above with Junction Deviation set to 0.01. Then increase the JD to 0.02 and then 0.05 during the print.

These are my settings (at the end of the print).
```
Send: M503
Recv: echo:  G21    ; (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E94.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3200 Y2200 Z200 E5000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1000.00 R1000.00 T10000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 J0.05
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H210 B60 F255
Recv: echo:  M145 S1 H240 B0 F255
Recv: echo:PID settings:
Recv: echo:  M301 P22.16 I1.67 D73.54
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: ok
```

This is my config file:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2950119/config.zip)

Most notably S-Curve acceleration is enabled in addition to JD.

**Expected behavior:**
The corners should become sharper. Once the JD value gets too large, the ringing should appear consistently around the corners.

**Actual behavior:**
The corners get sharper, but ringing only appears at the start of the first segment of the layer.

shitcreek, 2019-03-12:
Have you tried Linear Advance? It should help with corners.
http://marlinfw.org/docs/features/lin_advance.html

lekv, 2019-03-12:
Yes, I've tried to calibrate Linear Advance but the extruder would make large jumps and skip steps when trying to calibrate it with Junction Deviation enabled. Is what I am describing a known issue with Junction Deviation on first segments of a layer?

volca02, 2019-03-15:
There will be a small segment of Z-only movement when changing layers, I wonder if that confuses the JD algorithm.

boelle, 2019-06-22:
missing label

thinkyhead, 2019-07-11:
We'll have to get some logging of this to see what's happening. Maybe previous XY move information is being left behind instead of being thrown out on Z-only moves.

boelle, 2019-09-24:
@lekv is this still an issue?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle