MarlinFirmware/Marlin #13364: [BUG] LPC1768 build with disabled LCD fails on Adafruit NeoPixel/Adafruit_NeoPixel.cpp

Andrei-Pozolotin, 2019-03-11:
config snapshot:
https://github.com/random-builder/Marlin/tree/9d33ca5673e6852914552927e6b62c0fa251b8a8

build error:
```
Compiling .pioenvs/LPC1768/CMSIS/driver/lpc17xx_ssp.c.o
Compiling .pioenvs/LPC1768/CMSIS/driver/lpc17xx_systick.c.o
.piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
.piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:153:3: error: 'noInterrupts' was not declared in this scope
noInterrupts(); // Need 100% focus on instruction timing
^~~~~~~~~~~~
.piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:153:3: note: suggested alternative: 'detachInterrupt'
noInterrupts(); // Need 100% focus on instruction timing
^~~~~~~~~~~~
detachInterrupt
.piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:2037:3: error: 'interrupts' was not declared in this scope
interrupts();
^~~~~~~~~~
.piolibdeps/Adafruit NeoPixel/Adafruit_NeoPixel.cpp:2037:3: note: suggested alternative: 'intptr_t'
interrupts();
^~~~~~~~~~
intptr_t
Compiling .pioenvs/LPC1768/CMSIS/driver/lpc17xx_timer.c.o
Compiling .pioenvs/LPC1768/CMSIS/driver/lpc17xx_uart.c.o
```

kAdonis, 2019-03-11:
See #13329
Update PlatformIO frameworks and libraries

Andrei-Pozolotin, 2019-03-11:
"Update PlatformIO frameworks and libraries" resolves the issue, thank you.

closed 2019-03-11 by Andrei-Pozolotin