MarlinFirmware/Marlin #13367: Z moves as I ask Y to move

AlphonseZeratul, 2019-03-11:
Hi,

Marlin 1.60.3 for CR10S-PRO.

This gcode :
 
G28 X0 Y0 Z0 ;
G91 ; 
G1 F3000 Z30 ;
G90 ;
G28 X0 Y0 ; 
G1 F3000 Y200 ; 

is making Z go down (last G1).
To fix it, I've done this : 

G28 X0 Y0 Z0 ;
G91 ; 
G1 F3000 Z30 ;
G90 ;
G28 X0 Y0 ; 
**G92 Z0;**
G1 F3000 Y200 **Z0**; 

Bug ?

BR

boelle, 2019-03-11:
attach 2 config files as zip file

boelle, 2019-03-11:
and use bugfix 2.0

reloxx13, 2019-03-11:
there is no `x0 y0 z0` param for `g28`.
its `g28 x y z` or just `g28`

try without it to be sure not invalid gcode is confising the printer.
also do a power cycle off/on.

(not sure how it was in ur old marlin version, this info is from current marlin docs)
http://marlinfw.org/docs/gcode/G028.html

and u dont need to write the semicolon. its for comments like
`G28 ;homing now`

AlphonseZeratul, 2019-03-12:
> 
> 
> and use bugfix 2.0

?

AlphonseZeratul, 2019-03-12:
> 
> 
> attach 2 config files as zip file

Don't have them ... maybe Creality have.

reloxx13, 2019-03-12:
> > and use bugfix 2.0
> 
> ?

update to 1.9.0 or/and (try both) 2.0.x
than ull have the configurations, too.

AlphonseZeratul, 2019-03-12:
> 
> 
> there is no `x0 y0 z0` param for `g28`.
> its `g28 x y z` or just `g28`
> 
> try without it to be sure not invalid gcode is confusing the printer.
> also do a power cycle off/on.
> 
> (not sure how it was in ur old marlin version, this info is from current marlin docs)
> http://marlinfw.org/docs/gcode/G028.html
> 
> and u dont need to write the semicolon. its for comments like
> `G28 ;homing now`

I've done quickly. I've done a lot of trys. CURA is creating G28 X0 Y0, not G28 X Y (I've to try).
I've deleted comments, and simply forgot to delete ; when copy/paste.

7eggert, 2019-03-12:
While the marlin documentation says "G28 X Y Z", you can actually say "G28 X42 Y23 Z80" or just "G28". I made the same mistake while hunting a different bug.

I tested this on last week's bugfix-2.0 firmware, using the arduino serial console. My Anet A8 moves the Y axis as it should.

boelle, 2019-03-13:
@AlphonseZeratul bugfix 2.0: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

boelle, 2019-03-13:
>Don't have them ... maybe Creality have.

well you have them or else you would not be able to compile and flash firmware

configuration.h and configuration_adv.h is what we need in a zip file so we can check if your config is correct

closed 2019-05-17 by thinkyhead