MarlinFirmware/Marlin #13368: [BUG] MKS-SBASE/LPC1768 servo energized despite DEACTIVATE_SERVOS_AFTER_MOVE

Andrei-Pozolotin, 2019-03-11:
### Description

Fails to auto-disable servo pwm signal after servo has completed the motion.

### Steps to Reproduce

1. build with `#define DEACTIVATE_SERVOS_AFTER_MOVE`

2. verify servo is working via `M280 P1 S0`, etc

3. try to move servo actuator with hand

**Expected behavior:** servo shaft moves freely after the move

**Actual behavior:** servo continues to stay energized, can not move with hand after move is complete

#### Additional Information

perhaps due to use of non-default pin mapping
see #13314

config snapshot: [Configuration.h#L2078](https://github.com/random-builder/Marlin/blob/f531df1c494b3f85b2c986d14f2993c0e3f3ad14/Marlin/Configuration.h#L2078)
https://github.com/random-builder/Marlin/tree/f531df1c494b3f85b2c986d14f2993c0e3f3ad14

AnHardt, 2019-03-11:
`DEACTIVATE_SERVOS_AFTER_MOVE` should read 'Deactivate servo pulsing after move' - but that's a bit long.
What the servo makes from this is not defined.
Years ago I tested a bunch of different servos. Some depowered the motor. Some left the power on and tried to hold the position determined by the last received pulse(s). That may depend on, if the servos electronic is analog or digital. Could also depend on the manufacturer, series or type of the servo.
If the servo does not depower, there is no reason to be worried about. 
If `DEACTIVATE_SERVOS_AFTER_MOVE`  works as intended, there can't be pulse jittering, because there should not be pulses.

----

But - if the servo is not depowerd, there is another possible reason for a jittering servo angle. If VCC (5-6V) is varying a lot - in the right (wrong) frequency - that may be visible/feelable. There is no way to improve that by Marlin.

Andrei-Pozolotin, 2019-03-11:
1. I verified 5v servo supply quality - there is only 15mv ripple
a) the jitter I reported in #13314 occurs only when steppers are moving
b) there is no jitter when there is no stepper activity that could be attributed to 5v supply

2. I measured that servo pwm signal 
is still present **after** the move is completed, as follows:
```
M280 P1 S0   -> period=20ms, pulse=0.53ms, peak-to-peak=3.3v
M280 P1 S90  -> period=20ms, pulse=1.46ms, peak-to-peak=3.3v
M280 P1 S180 -> period=20ms, pulse=2.39ms, peak-to-peak=3.3v
```

drzejkopf, 2019-03-11:
It doesn't work, because you have enabled BLTouch, go to src/inc/Conditionals_LCD.h and comment line 468
#undef DEACTIVATE_SERVOS_AFTER_MOVE

Andrei-Pozolotin, 2019-03-29:
1. perhaps it should be documented here about possible configuration interference: generic servo vs bltouch servo:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h#L2105
```
//#define DEACTIVATE_SERVOS_AFTER_MOVE
```

2. experiments show that `DEACTIVATE_SERVOS_AFTER_MOVE` is not needed by BLTOUCH here,
sensor works just fine on LPC1768, perhaps this should be made optional instead:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L455
```
  #undef DEACTIVATE_SERVOS_AFTER_MOVE
```

boelle, 2019-06-22:
missing label

thinkyhead, 2019-07-11:
If the `#undef DEACTIVATE_SERVOS_AFTER_MOVE` is not needed, great! It was added after some discussion where it wasn't completely clear-cut whether to enable it or not. https://github.com/MarlinFirmware/Marlin/issues/3902#issuecomment-222881064 Upon review, it does seem safe to either remove that line, or even (ironically) to replace it with `#define DEACTIVATE_SERVOS_AFTER_MOVE`.

AnHardt, 2019-07-11:
//#define DEACTIVATE_SERVOS_AFTER_MOVE

was the suggestion of the first original manual for the BL-Touch (at least for the Kickstarter version). 
(But they have been wrong in that many points. Why not also in this.)

boelle, 2019-08-19:
@Andrei-Pozolotin have you tried to test with latest bugfix 2.0.x to see if the issue us still present?

boelle, 2019-09-24:
@Andrei-Pozolotin is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

Vertabreak, 2019-11-29:
i'm gonna retest this and if i see any gain ill send PR.

InsanityAutomation, 2019-11-29:
> i'm gonna retest this and if i see any gain ill send PR.
> it does seem silly to prevent the use of the option on the bltouch unless it causes some crazyness.

Antclabs does not recommend it and after testing as klipper turns off pwm, it found absolutely 0 change in function, just the input light on the bltouch goes out. No change to repeatability either.

Vertabreak, 2019-11-29:
that test was on a real bltouch? not a clone?

Vertabreak, 2019-11-30:
nope makes the 3dtouch not function at all.