MarlinFirmware/Marlin #13372: [BUG] (short description)

FelixHauser, 2019-03-11:
### Description

Delta Height Calibration has changed by aprox 15mm from Marlin 1.1.9 to Marlin bugfix 2.0

### Steps to Reproduce

1. Buy or build a Delta Printer
2. Install Marlin 1.1.9 on a compatible board
3. Run Delta Autocalibration and write down the Delta Height
4. Update to Marlin Bugfix 2.0
5. Copy paste the parameters obtained on 1.1.9 to Marlin 2.0
6. Issue a G1 Z5 command (optional)
7. Watch as your printing bed gets a tattoo because of the nozzle is ramming the bed (optional)
8. Run another Delta Autocalibration
9. See how without changing any hardware, the height of your delta printer has been cut off by aprox 15mm

**Expected behavior:** 

I expected the Delta Settings to be the same for both versions of Marlin

**Actual behavior:**

Apparently, the settings are not the same. At least the delta height

#### Additional Information

So, yeah. This is what happened yesterday. I upgraded the control board to a 32b BIQU SKR 1.1
and without giving it any though, I copy pasted the settings (which were working on Marlin 1.1.9) from one Marlin to the next (2.0). I tried to issue a G1 Z5 command ant the nozzle scratched the my ultrabase bed. 
I don't know if it is a bug or a feature, but the delta height has been cut off by 15mm (aprox). I have the impression that in Marlin 2.0 there is like a Endstop offset, that lowers the head some mm. Not that the height is of much importance, but yeah... I'd like to know if it's a bug.

Thx

reloxx13, 2019-03-11:
i tried it myself to use a 1.1.9 config in 2.0.x and it wont work, many variable names have changes so alot were missing or wrong spelled.

compare ur 2.0.x config with one of the example configs for ur board or if its not in examples use the delta/generic for comparing.

boelle, 2019-03-11:
maybe attach 2 config files so the smart people have a chance of figuring if settings are right etc etc etc

FelixHauser, 2019-03-11:
How the smart people will be able to tell me if my delta_height is right, is beyond my comprehension

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2954503/Archive.zip)

reloxx13, 2019-03-12:
if u still have it, please post the 1.1.9 config, too.

FelixHauser, 2019-03-12:
thanks for the interest, but given that @reloxx13 told us that it happened the same to him, I´m soing to go ahead at take it as normal and close the issue. Thanks.

closed 2019-03-12 by FelixHauser

reloxx13, 2019-03-12:
nooo, u misunderstood me :S