MarlinFirmware/Marlin #13374: EXAMPLE CONFİGURATİON FOR DELTA TYPE 3D PRİNTER

ebubekirn, 2019-03-11:
which one should i choose example configuration for marlin firmware. I have delta type 3d printer. But there are 7 different type for delta. There are : Anycubic, FLSUN, generic, Hatchbox_Alpha, kossel_mini, kossel_pro, kossel_xl.
My printer's height is 1 meter. Can you help me for this problem??

reloxx13, 2019-03-11:
u dont need to use the example config.
take one which is almost like urs and adjust the config for ur printer.
tempsensor, steppers, board, endstops, size, etc.

read the configuration.h and configuration_adv.h, everything is described quite good in it.

http://marlinfw.org/docs/hardware/boards.html#board-list

could be outdated at some points if u use 2.0.x : 
http://marlinfw.org/docs/configuration/configuration.html#configuration.h

ebubekirn, 2019-03-11:
thank you bro

reloxx13, 2019-03-12:
pls close this if u r done, since u opened another issue this one is "done".

keep it clean ;p

closed 2019-03-12 by ebubekirn