MarlinFirmware/Marlin #13375: endstop settings

ebubekirn, 2019-03-12:
İ am using delta type 3d printer. Can anybody help about the endstop settings. Normally endstop settings like that:
//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
//#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG // a Z probe
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.

Do ı have to change something?

shitcreek, 2019-03-12:
I grabbed this from the "generic" delta config file
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/2955064/Configuration.h.zip)
Hope it helps. Also, this blog might have some useful information: http://wp.boim.com/?p=54

`// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
//#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG // a Z probe
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG

#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true  // set to true to invert the logic of the probe.`

7eggert, 2019-03-12:
You should understand the meaning of the settings before changing them. Also you should test them by typing "M119" at the arduino serial console.

There are NC and NO type sensors and switches.

NO means no connection / 5 v if the switch is detected or the sensor is active and connection to GND / 0V if the switch is pressed or the sensor detects proximity. This is called "INVERTING" because usually 5 V (or 3.3 V) means an active signal. These need a pullup resistor, some mainboards support internal resistors or have these soldered-on.

NC is the opposite. For these, you need a pulldown resistor. AVR does not support this internally, but there may be one soldered on.

A probe using the Z endstop pin needs to be of the same type as the pin it's using, otherwise they'd interfere.

Some probes do send a signal of e.g. 12 V, higher than the mainboard logic can accept without damage. Be sure to have a level converter for these kind of probes - an optocoupler is recommended. If you are using a diode, the signal will be at 0.7 v and the AVR is not guaranteed (my manufacturer) to detect it correctly. Many sensors will need 6 V to work correctly (and thus send overvoltage).

"(most common setup)." - I'm not sure this is true anymore, but I only encountered a limited variety of china printers by forum.

kAdonis, 2019-03-12:
@7eggert 
This is not quite correct.
It is possible to use NC Switches like you describe, but I'm using NC (normally closed) switches with COM to GROUND and these also need  pullup resistors. This is what is described as "most common setup", because it has the advantage, that if a cable comes loose the switch shows triggered, the axis doesnt move when homing, and you know that something is wrong.
With a NO (normally open) switch and a loose cable, you realize an error after the axis reached its mechnical limit with possible damage.

7eggert, 2019-03-12:
Sorry, you are correct.

ebubekirn, 2019-03-12:
I have another problem about my endstops. When the arms touching the endstops they are not stop. They still trying to go up. İs this about NO NC settings? Can you help me? And ı am using three wire mechanical endstops. How must ı connect to my ramps 1.4 board? İn the inthernet generally 2 wire types. Please give some advices :) Thank you

7eggert, 2019-03-13:
What is your mainboard type? I'll assume RAMPS 1.4 (I'm currently in the process of changing my Anet to RAMPS).

Since I have a cartesian printer, I googled and found that: reprap.org/forum/read.php?178,495853 "Deltas never have min endstops". Thus your endstops would need to be connected to the max endstop pins. Make sure to not connect the switch to VCC and GND.

On my RAMPS, the sensor pins are blue, GND is black and VCC is red. The red pins should be unused, and the `#define ENDSTOPPULLUPS` should be set.

Having three pins, you can use NC or NO,  kAdonis suggested using NC if possible.

Some switches say "NC" and "NO" on the corresponding pins - use that pin and the third one, connect these to the signal pin GND. Don't configure inverting endstops. (If you had two-pin switches or an unlabeled switch, you can measure or guess.)

Open the arduino console, verify the baud setting (if you see a lot of lines starting with "echo", the baud setting is correct. On my RAMPS, I had to change the configuration from 250000 baud to 125200 and re-upload to make it work). Manually press each endstop (or gently move the mechanics, whatever is easier for you) and compare the result of M119 with the actual status of the switches. Do the values being displayed change by pressing a switch? Do they match your expectation?

(If you have two-pin switches, the status will be inverted. If you have three-pin switches and connect the NC and NO wire, ist should not work at all. If no switch is connected, all should read as "TRIGGERED").

If you have a voltage meter, you can measure the voltage at the switch's pins. It should read 5 V (or 3.3 V) and change to 0 V while pressing the switch for NO switches and the other way around for NC switches. (It may display negative voltages instead of positive, that's no problem.) If you always have 0 V, you may need to turn on the endstop settins or you may have connected the switch to the wrong pins.

My printer after homing (thus touching all the switches):
(I type M119)
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED

Move X away from the switch:
Reporting endstop status
x_min: open
y_min: TRIGGERED
z_min: TRIGGERED

Therefore I know that the X min switch is working correctly. I'd continue with Y and Z doing the same.

boelle, 2019-03-13:
@ebubekirn figured it out?

since this is more a question than a bug please consider using discord or facebook

7eggert, 2019-03-13:
I think the answer should be in the documentation / moved to the wiki. (after being verified).
Also incidentally I stubled on this: www.thingiverse.com/thing:2906283

closed 2019-05-18 by thinkyhead