MarlinFirmware/Marlin #13376: Extremely slow extrusion with PIDTEMPBED enabled

brainyron, 2019-03-12:
I recently upgraded my Ender-3 with a MKS Gen L board.  After getting it working with the base Ender-3 config, I went through and turned on features I'd like to use since I now have more program space.  As soon as I flashed that firmware, I found my extruder was running dead slow.  To the order of a single 20mm extrusion (feedrate 300) taking several minutes to complete, during which it would not respond to any other Gcodes.  Through a lengthy process of elimination, I found that this was caused by PIDTEMPBED being enabled.

I'm using Marlin 1.1.9 and confirmed that this issue persists in the 1.1.x-bugfix branch as well.

boelle, 2019-03-12:
also on bugfix 2.0?

reloxx13, 2019-03-12:
> * Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

brainyron, 2019-03-13:
@boelle I haven't tried bugfix-2.0.  I've been avoiding it until it's closer to release.  If you think the issue might be fixed there I can try loading it temporarily.

brainyron, 2019-03-13:
@reloxx13 Configuration files are here: [https://drive.google.com/open?id=1IwlHu9_LHM6D0G_ZZ4qPwOjIH89mBz96](https://drive.google.com/open?id=1IwlHu9_LHM6D0G_ZZ4qPwOjIH89mBz96)

boelle, 2019-03-13:
you can also attach configs here :-)

[image]

Yuppi911, 2019-06-13:
I've encountered the same problem on my Anet A8 but also all axes motors were running dead slow after i've enabled PIDBEDTEMP. Did use of bugfix-2.0 solve this problem? And what can be the cause of it?

Rubberduck84, 2019-06-17:
My Extruder it's also dead slowly.
I don't can find the problem.

thinkyhead, 2019-06-18:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Rubberduck84, 2019-06-18:
i will test it this week thanks for the files

boelle, 2019-06-22:
@Rubberduck84 what are the test results?

Rubberduck84, 2019-06-22:
I having troubles with complying marlin 2.0 have use now marlin 1.1.9 and I have reset my eprom memory and the extruder working again 😀😀😀

Yuppi911, 2019-07-01:
Also the same situation - after resetting eeprom all motors started working normally

boelle, 2019-08-19:
@brainyron @Rubberduck84 can you confirm what @Yuppi911 did solves the problem?

Rubberduck84, 2019-08-19:
@boelle,

Eeprom reset Fix the problem

boelle, 2019-08-28:
i will close this one since it seems to be fixed by resetting the eeprom

we can always reopen if it turns out not fixed

closed 2019-08-28 by boelle