MarlinFirmware/Marlin #13377: [BUG] (RAMPS pin assignment)

mbm60, 2019-03-12:
Hi

in header file pins_RAMPS.h and in section "MKS_MINI_12864" , pin 64 is assigned to KILL_PIN wrongly. pin 41 is right.

boelle, 2019-03-12:
you could submit a PR yourself to fix this :-D

boelle, 2019-03-20:
you are not right, it is set to 41

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_RAMPS.h#L519

```
    #elif ENABLED(MKS_MINI_12864)   // Added in Marlin 1.1.6

      #define DOGLCD_A0         27
      #define DOGLCD_CS         25

      // GLCD features
      // Uncomment screen orientation
      //#define LCD_SCREEN_ROT_90
      //#define LCD_SCREEN_ROT_180
      //#define LCD_SCREEN_ROT_270

      #define BEEPER_PIN        37
      // not connected to a pin
      #define LCD_BACKLIGHT_PIN 65   // backlight LED on A11/D65

      #define BTN_EN1           31
      #define BTN_EN2           33
      #define BTN_ENC           35

      #define SD_DETECT_PIN     49
      #define KILL_PIN          41
```

closed 2019-05-01 by mbm60