MarlinFirmware/Marlin #13381: [FR] Want to set enable pins high early in boot.

bjbeare, 2019-03-12:
### Description

My printer does a "bump" on Z axis whenever I apply power. My guess is that Marlin is initializing the STEP and DIR pins before the ENABLE pins for the (three) Z axis steppers. Once Marlin is fully booted, the "bump" is gone and the steppers operate correctly. An alternate explanation is that the stepper expander board actuates the drivers with noise during power on and before the pull up is put on the enable pin.

### Steps to Reproduce

 I am using a stepper expander X3 with:

#define Z_DIR_PIN       59
#define Z_STEP_PIN      64
#define Z_ENABLE_PIN    44

#define Z2_DIR_PIN      63
#define Z2_STEP_PIN     40
#define Z2_ENABLE_PIN   42

#define Z3_DIR_PIN      3
#define Z3_STEP_PIN     57
#define Z3_ENABLE_PIN   58

#### Additional Information

Looking at the code, I do see what appears to be the initializers for Z*_ENABLE_PIN in src/module/stepper.cpp:

  #if HAS_Z_ENABLE
    Z_ENABLE_INIT;
    if (!Z_ENABLE_ON) Z_ENABLE_WRITE(HIGH);
    #if Z_MULTI_STEPPER_DRIVERS && HAS_Z2_ENABLE
      Z2_ENABLE_INIT;
      if (!Z_ENABLE_ON) Z2_ENABLE_WRITE(HIGH);
    #endif
    #if ENABLED(Z_TRIPLE_STEPPER_DRIVERS) && HAS_Z3_ENABLE
      Z3_ENABLE_INIT;
      if (!Z_ENABLE_ON) Z3_ENABLE_WRITE(HIGH);
    #endif
  #endif

The code "seems" to be right... provided it is executed early enough (before the STEP_PINs are initialized). If it isn't a code issue, I suppose that I could put a pull-up on the stepper expander's enable pins. I've not tried this yet.

Has this issue been seen? Any advice?

Thanks!

boelle, 2019-03-13:
i dont have the issue, but it can be that the power rails do not come up in the correct order

5V/3.3V has to come up before 12V/24V or else the stepper control signals can be in any state hence you can hear strange sounds

its worth checking out what comes first, control signals or power to move steppers

bjbeare, 2019-03-13:
Thanks @boelle . I have an oscilloscope on order... I'll be able to look at timings in a few weeks.

bjbeare, 2019-03-17:
Problem wasn't the code. The stepper expander doesn't include pull-up resisters on the enable lines. I added them (tacked to the back of the board) and the problem is resolved.

closed 2019-03-17 by bjbeare