MarlinFirmware/Marlin #13384: [BUG] M290 causes instant reboot

Ryanf55, 2019-03-13:
### Description

Printer: Anycubic Kossel Linear+ Delta
Marlin Version: 2.0.x bugfix downloaded February 16, 2019
Config File: (both config and config_adv are in the zip folder)

### Steps to Reproduce

1. Upload the latest Marlin 2.0.x bugfix and my config file on a STOCK Anycubic Kossel Delta linear +. 
2. Turn printer on
3. Home it with g28;
4. Send a babystep command with M290 Z0.25;
5. Watch the printer reboot.

**Expected behavior:** 
I expected the printer to change the z position slightly. It should stay in the babystep menu while during a print. If G-code sends the command, the printer should keep printing.

**Actual behavior:**
Every time a babystep command is issued in G-code or through the menu during a print, the printer reboots. 

#### Additional Information
1) Zip file attached. All changes are tagged by me from default. 
2) This has been confirmed by another user on RepRap forums here:
https://reprap.org/forum/read.php?415,847369,849037#msg-849037
3) Repeatable? Yes, it reboots every single time.

Please let me know if you require further information on this bug.

boelle, 2019-03-13:
is it also the case with 2.0.x from today?

its updated many times a day (just look at commit history) and it might well be that is has been fixed during the last month

Ryanf55, 2019-03-13:
Boelle, 

Thank you for the reply. I will have time to try with the latest version on Thursday. I will let you know the results. 

I did search the commit history for M290 and did not see anything in the past year or so.

Ryanf55, 2019-03-16:
Hello again.

I just downloaded bugfix 2.0.x from today (March 15th). Sending M290 with my config files still causes a reboot. Please let me know what I can try. 

Thank you!

Stady234, 2019-04-10:
any updates? I am running into the same issues...

thinkyhead, 2019-04-10:
Unable to reproduce with the current `bugfix-2.0.x`.

Stady234, 2019-04-13:
My issues were solved via this thread adjustment... https://github.com/MarlinFirmware/Marlin/issues/13300

Basically all my various machine run on 4988 drivers and as the thread points out.. the MINIMUM_STEPPER_PULSE in more recent commits was changed from 2 to 1.  Which leads to bad math and a system lockup.   I made the change as the poster described in my stepper.cpp file, recompiled and all is good.  No more lockups on my machines.

thinkyhead, 2019-05-17:
With the fix above now merged, the reboot should no longer occur. Please close this issue if the problem is solved… and if not we’ll fix more things.

closed 2019-05-18 by thinkyhead