MarlinFirmware/Marlin #13385: Manual bed leveling, incorrect Z height.

nique88, 2019-03-13:
Well hello,

I've got an issue I cannot figure out.
Since 1 month I use manual mesh bed leveling. After I printed something that had to be very accurate, the issue came up.

I printed a calibration cube to be sure. The x and the y are 19.8mm but the Z is 18.85mm.

I changed the hardware from the Z with my other printer but still the same measurement.
When I put the nozzle on the bed and set manually 1mm steps I measure exactly 1mm with the feeler gauge.

I think it has something to do with the MBL. I tried also different fade heights but nothing changed.
From what I can see is that the first couple mm of the prints are squished to much on eachother.

Also this difference in Z is not linear. On every object with different heights the difference in Z is around 1.15mm.

When I print with standard bed leveling I get the measurements spot on.

Who has the answer for this trouble?

Thanks in advance

Printer: ender 3
Software: Cura
Firmware: marlin 1.1.9

reloxx13, 2019-03-13:
m420 s1 in startcode to apply the mesh?

measure z without printing, move it to 50mm, mark it with tape on the the profile, move up for 100mm, mark again and measure the distance ( hope u understand what i mean :p ).

nique88, 2019-03-13:
The m420 I have in my start code.

I know where you want go to. The calibration of my Z is ok. I did the measurement and everything is ok. Also without mesh bed leveling the cali cube comes out nearly perfect.

Tonight I will add my config.

Thanks

shitcreek, 2019-05-15:
@nique88  Has your issue been resolved? Can we close this issue?

nique88, 2019-05-15:
It's not resolved.
But I use standard bed leveling now. It gave me headache!

Greetings

boelle, 2019-06-22:
missing label

closed 2019-07-11 by thinkyhead