MarlinFirmware/Marlin #13389: [2.0] MKS Gen L - layer fan not working with FAN_SOFT_PWM (MOSFET LED is barely lit) - works with 1.1.8

dayaan82, 2019-03-13:
Almost explained it all in the description.

FAN_SOFT_PWM enabled.. layer fan not working anymore.. the MOSFET LED is barely lit and not bright like the extruder and bed ones..

Works normally with FAN_SOFT_PWM disabled.

This option works on 1.1.8.

TarruTarru, 2019-03-16:
Yes, I have exactly the same problem. Board = MKS Gen 1.4, with Marlin 2.0.x from September 2018,  which I have on my hard disk SOFT_PWM works fine. With Marlin 2.0.x downloaded yesterday - SOFT_PWM does not work. All settings in Configuration.h and Configuration_adv.h are the same.

thisiskeithb, 2019-03-20:
A friend of mine was having similar issues with their MKS GEN 1.4 board and a 24v Mechatronics B5015E24B 5015 radial fan. I purchased the exact same fan and confirmed the same issue using Marlin 1.1.9 & the latest Bugfix-2.0 branch from the [1440187](https://github.com/MarlinFirmware/Marlin/commit/144018736aa1e0d466e0c0693cf7992190e4981f) commit from a couple of days ago.

I had to enable FAST_PWM_FAN and disable FAN_SOFT_PWM, otherwise the fan wouldn't spin at any value between 1 & 255. It would just make a faint clicking sound and the mosfet LED would barely light up.

bossman74, 2019-03-25:
I have the same issue with this 12v Brushless 5015 blower.  But no issues when using Marlin 2.0.x from last August, no noise at all speeds.  This looks like a recent thing.

On the latest branch, I only got it running with `FAN_SOFT_PWM` commented.  And then, it's noisy when not running at full speed.

[https://goo.gl/ePVwPx](url)

bossman74, 2019-03-27:
Anyone found a solution to this ?  I reverted back to an older 2.0.x bugfix until then.

TarruTarru, 2019-03-27:
For all of you who would like to use the latest version of Marlin (where SOFT_PWM does not work) but have a problem with a loud fan.

I used simple hardware in the form of a low-pass filter. After connecting this you can turn off SOFT_PWM (use Hardware PWM) and the whistling noise is gone.
[image]

bossman74, 2019-03-28:
In the latest commit from yesterday, my fan now works with `FAN_SOFT_PWM` and without the noise.

Thank you to whoever fixed it !

closed 2019-05-18 by thinkyhead

jasonmarquette, 2020-02-05:
I have an issue with my parts fan not running after upgrading to marlin 2.0.3. I also have an MKS_GEN_L board. I'm using a BLTouch 3.1

My Fan will not turn on via any slicer settings, however I can control it on the Ender 3 directly and it will turn on manually. I was really hoping enabling FAN_SOFT_PWM would solve my issue. I have searched everywhere. Any Idea's?

thinkyhead, 2020-02-07:
@jasonmarquette — Please file a new issue and be sure to fill in the whole template.