MarlinFirmware/Marlin #13390: [BUG] M191 is disabled in source

Andrei-Pozolotin, 2019-03-13:
M191 is disabled in source:

https://github.com/MarlinFirmware/Marlin/blob/67bee06e4371fca6511aae99125337b6e63560d8/Marlin/src/gcode/gcode.cpp#L400

```
      #if HAS_HEATED_CHAMBER
        case 141: M141(); break;                                  // M141: Set chamber temperature
        //case 191: M191(); break;                                // M191: Wait for chamber temperature to reach target
      #endif
```

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
Chamber heaters are getting more common. This may need a review soon.

boelle, 2019-09-24:
@Andrei-Pozolotin is this still disabled? chamber control has just been added to graphics (GLCD)

Andrei-Pozolotin, 2019-10-08:
@boelle : yes, still disabled:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/gcode.h#L621
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/gcode.cpp#L447

boelle, 2019-10-12:
should be fixed when https://github.com/MarlinFirmware/Marlin/pull/15531 is merged

GMagician, 2019-10-17:
This may be closed, above PR has been merged

closed 2019-10-21 by boelle