MarlinFirmware/Marlin #13395: [FR] Add support for Fysetc Mini 12864 display

bipsendk, 2019-03-14:
### Description

Add support in Marlin 2.0 for Fysetc Mini 12864 display

### Details

Fysetc Mini 12864 is a Minipanel Clone - but (at least one model) have the ability to change colors using RGB LED's.

The LED control is done via pins in the EXP1 port - so some tinkering is needed to the pins file for at least RAMPS and FYSETC_F6_13.
Besides this, a single file or two probably needs minor adjustments as well.

#### Additional Information

Information about required changes exist at https://wiki.fysetc.com/Mini12864_Panel/
Modifying the code according to the instructions provided by the vendor is not that difficult - the challenging part is to get dependencies modified, so that configuration validation still will work, and the user only will have to consider modifying configuration.h and configuration_adv.h

I have a display to test with (on a Fysetc F6 board), in case someone is able to implement the needed patches - so that a new display type (maybe named FYSETC_MINI_12864_RGB ) can be added.

Bob-the-Kuhn, 2019-03-15:
There's a person wanting to use this display on a Re-ARM board.  Here's my first pass at the needed changes:
1. The Fysetc display has the shrouds on the EXP1 & EXP2 connectors reversed from the RepRap norm.  I suggest CAREFULLY prying up the shrouds on the LCD adapter, rotating them 180 degrees and re-installing.  See the [RepRap Re-ARM WIKI]( https://reprap.org/wiki/File:Re-ARM_LCD_adapter_MKS_Rev_4.jpg) for a picture.
2. Replace the file **pins_RAMPS_RE_ARM.h** with this file: [pins_RAMPS_RE_ARM.h.txt](https://github.com/MarlinFirmware/Marlin/files/2969484/pins_RAMPS_RE_ARM.h.txt) (remove the ".txt") 
3. The Fysetc display uses +5V but the LCD adapter supplies +3.3V so a custom cable is needed.  As long as a custom cable is being created I suggest also breaking out the R, G & B LED signals.  My recommendation is to use the Re-ARM ethernet connector J12 to get the four signals.

```
                                        EXP1 cable 
                 Re-ARM     Re-ARM     -------------        		
function        pin/port   connector   RepRap   MKS   		
+5V              N/A        J12-14       10      1   		
RGB_LED_R_PIN    P1_04      J12-10        6      5   		
RGB_LED_G_PIN    P1_00      J12-11        7      4   		
RGB_LED_B_PIN    P1_01      J12-12        8      3   		
```
4. Assuming my recommendation is followed, the **configuration.h** changes would be:
```
#define MINIPANEL

#define RGB_LED
//#define RGBW_LED

#if ENABLED(RGB_LED) || ENABLED(RGBW_LED)
  #define RGB_LED_R_PIN   P1_04   // (D77)
  #define RGB_LED_G_PIN   P1_00   // (D78)
  #define RGB_LED_B_PIN   P1_01   // (D79)
  #define RGB_LED_W_PIN -1
#endif
```
5. The **configuration_adv.h** changes are the same as in the Fysetc WIKI

FYI - I like using the [Adafruit P1199 rainbow (octopus) cable](https://www.adafruit.com/product/1199) when doing custom 10 pin cables.

bipsendk, 2019-03-15:
The issue of connectors on EXP1 and EXP2 are well-known - and often depends on the brand of board/adapter being used to connect the display.
As for the 3.3/5V hack/cable-btreakout, this is only required for board/adapters not supplying +5V on EXP1 (probably an issue multiple boards may have).

What about the change in ULTRALCD_IMPL_DOGM.H as listed on the Fysetc wiki page - no need to implement that part ?

Bob-the-Kuhn, 2019-03-15:
I know that Re-ARM has the 5.0/3.3V issue.  The RAMPS-FD adapters route 5V to the correct pin so DUE is OK.  Don't know about any of the other 32 bit boards.

On the set contrast, I think I've handled it by including the statement  **#define LCD_CONTRAST_MAX 255** in the pins file.  Can't be sure until someone actually tries it.

Bob-the-Kuhn, 2019-03-15:
I still have work to do on the LCD_CONTRAST.  Tried it on a Re-ARM but the M250 command wasn't available which means LCD_CONTRAST hasn't been setup properly.

And here's the pins file with the compiler warnings fixed.

[pins_RAMPS_RE_ARM.h.rev 2.txt](https://github.com/MarlinFirmware/Marlin/files/2969702/pins_RAMPS_RE_ARM.h.rev.2.txt)

Bob-the-Kuhn, 2019-03-15:
I ended up hacking in **FYSETC_MINI_12864_PANEL** as a new LCD controller.  It may have come at the expense of **MINIPANEL** so it is not the final code.

Here's the three files that need to be changed: [FYSETC_MINI_12864_PANEL.zip](https://github.com/MarlinFirmware/Marlin/files/2969848/FYSETC_MINI_12864_PANEL.zip)

There is one change to the instructions.  In **configuration.h** do **NOT** enable **MINIPANEL** but instead add this line `#define FYSETC_MINI_12864_PANEL`

bipsendk, 2019-03-15:
I think the missing M250 originates from this:

```
#if HAS_GRAPHICAL_LCD
  /**
   * Default LCD contrast for Graphical LCD displays
   */
  #define HAS_LCD_CONTRAST (                \
       ENABLED(MAKRPANEL)                   \
    || ENABLED(CARTESIO_UI)                 \
    || ENABLED(VIKI2)                       \
    || ENABLED(AZSMZ_12864)                 \
    || ENABLED(miniVIKI)                    \
    || ENABLED(ELB_FULL_GRAPHIC_CONTROLLER) \
  ) 
```

An entry with
`|| ENABLED(MINIPANEL) \`
probably needs to be added..  That is why a special entry perhaps is needed for the FYSETC mini display...

bipsendk, 2019-03-15:
I finally got it working on my F6 board....

[image]

Now - I made changes to some files, but cannot remember exactly which ones... And I am no export in using git to contribute any changes to the sourcecode tree...

bipsendk, 2019-03-15:
These are my changes - I have only prepared for Fysetc F6 board.. In case other board should be supported with this display, the pins.h file for these boards probably needs to be edited as well..

Can someone do sanity check of what I have done?

[fysetc_mini_12864.zip](https://github.com/MarlinFirmware/Marlin/files/2972552/fysetc_mini_12864.zip)

bipsendk, 2019-03-15:
@Bob-the-Kuhn  - the diff file I just added will automatically define the RGB pins, so that might be used as an inspiration to modify the pins_RAMPS_RE_ARM.h file as well....

co8fs, 2019-03-15:
Hi 
Can anybody help me please.
I have a Rumba board and an MKS MINI12864 display board but is not working, only backlite.
Please.
I am running Marlin V2.0
Thanks.

Sebastian
co8fs@yahoo.com

bipsendk, 2019-03-16:
Maybe some kind of sanity check will be needed, so that using the FYSETC_MINI_12864 only will be valid on pins files, that have been prepared for it ?

hargany, 2019-03-18:
@bob-the-kun I tried the files, defined fystec Mini panel, etc...but I still have compilation errors as if I did not have a defined LCD. I have to disable all advanced options and basics. and the rgb control. I do not know where I'm wrong

Bob-the-Kuhn, 2019-03-18:
Compilation errors I can help with.

Please ZIP up your Marlin directory and then post it here.  That way I know I have everything.

hargany, 2019-03-18:
[Marlin-bugfix-2.0.x.mini7z.zip](https://github.com/MarlinFirmware/Marlin/files/2977988/Marlin-bugfix-2.0.x.mini7z.zip)
i hope i don't do to many mistakes

Bob-the-Kuhn, 2019-03-18:
I didn't include all the needed files before.  Here's the corrected ZIP.

[FYSETC_MINI_12864_PANEL (corrected).zip](https://github.com/MarlinFirmware/Marlin/files/2980932/FYSETC_MINI_12864_PANEL.corrected.zip)

Bob-the-Kuhn, 2019-03-18:
@hargany - Here's the two files that need to be replaced:
[fixes.zip](https://github.com/MarlinFirmware/Marlin/files/2980937/fixes.zip)

In addition to the file I forgot to include, there was a line missing in configuration.h

Bob-the-Kuhn, 2019-03-19:
@co8fs 

Sebation - please open a new issue for your problem.  It's too confusing to discuss two different problems in the same thread.

When you open the issue please include the following:
A ZIP containing **configuration.h**, **configuration_adv.h** and **pins_YOUR_BOARD.h**
A description of how the display is connected to your controller.

hargany, 2019-03-19:
@Bob-the-Kuhn -j'ai compilé sans probleme ma version et une" propre ";je fais les branchements re-arm au mieu ce soir ou demain.
je pourrais tester cet écran dans la semaine, j'espère,sur ramps fd et due+fd adaptateur

grafixcz, 2019-03-24:
Hi,
I successfully connected this panel to Rumba board with Marlin 1.1.9.

Unfortunately, the LED CONTROL menu does not allow you to save the selected color display backlight. The display backlight is always in the default setting after restart. It can be solved by direct setting of color channel R, G or B when selected for LCD_BACKLIGHT_PIN one of LCD_PINS R, G or B.

The bigger problem is that when the display is connected (powered), the firmware upload does not work. Communication with the board over Repetier (and similar) it works. Once the display is disconnected, the firmware upload is functional.

Any idea for a solution? Thanks.

bipsendk, 2019-03-24:
@grafixcz - https://www.facebook.com/groups/197476557529090/permalink/352352678708143/ 
Remove R4 and C1 on the display...

grafixcz, 2019-03-24:
Thanks for the advice, it works :)

Brainc0re, 2019-04-04:
I have a CR10S-Pro which normally use a touchscreen. But i didn't like and replaced it with a 12864 reprap full graphic controller. That controller/display works since weeks. So i thought, that this mini12864 will work too. But im not able to run it :/ Not even the backlight works (Pin1 and Pin2 are correct - i can meassure 5V - Pin1 + // Pin2 -)
I made the changes in Marlin which are provided by fysetc on their website.

Is it possible, that this display wont work on the creality mainboard?

Current Firmware (reprap 12864): https://drive.google.com/open?id=1HoQVlum-DM5lGhCuu1yqCfW6UupuqbK6
FW for mini12864: https://drive.google.com/open?id=11KdF0nT0fa-EBrvGuiYhUTILRo4qJgjR
Mainboard layout: https://drive.google.com/open?id=1Ya5zHF_WQpw7XY_BoL0zs20UwexBvtwI

Attention: Both FWs has the same *.ino name but different folder names.

John824, 2019-04-10:
Hello!
Can anyone suggest how to adjust the contrast?
Fee Mega 2560, Ramps 1.4.
On Marlin 1.1.9 firmware, everything works well.
[image]
On the Marlin firmware bugfix - 2.0. X can not increase the contrast. 
[image]
Wherever I found, I set LED_CONTRAST_MAX, LED_CONTRAST_MIN and DEFAULT_LCD_CONTRAST = 255. But this did not change anything.

doughy2002, 2019-04-12:
> Hello!
> Can anyone suggest how to adjust the contrast?
> Fee Mega 2560, Ramps 1.4.
> On Marlin 1.1.9 firmware, everything works well.
> [image]
> On the Marlin firmware bugfix - 2.0. X can not increase the [...]

hi,  i cannot get the lcd backlight working or any leds, i have an mks gen l which is what i was sold the display with, could you share your pin settings for this please on marlin1.1.9

John824, 2019-04-12:
Hello!
I unfortunately did not deal with mks gen l.
But it seems to me that you first need to check the outputs on the mks gen l on the EXT 1 connector in accordance with the datasheet. To get started, just try turning the EXT 1 and EXT 2 connectors 180 degrees. And then set up everything according to this instruction https://wiki.fysetc.com/Mini12864_Panel/.

By the way, I managed to adjust the contrast on a bunch of Mega 2560 + RAMPS 1.4 + Marlin 2.0.X.
You need to add the following data to the #if ENABLED (FYSETC_MINI_12864) line in the Conditionals_LCD.h file:

#define LCD_CONTRAST_MIN 0
#define LCD_CONTRAST_MAX 255
#define DEFAULT_LCD_CONTRAST 200

in the same file, in the line #define HAS_LCD_CONTRAST
change the #define LCD_CONTRAST_MAX and #define DEFAULT_LCD_CONTRAST values ​​to 255.

After these settings, you can change the contrast of the display through the terminal using the command М250 С ... (instead of the points, the desired contrast value is from 0 to 255).

The last problem remains. How to run this display on DUE + Hacked RAMPS?

doughy2002, 2019-04-12:
So just cut the tabs off the connector and turn it upside down?

John824, 2019-04-12:
You can not cut off the tongues.
With them, too, the connector in the socket will fit.
But the thought you understood correctly.

DkOuvE, 2019-04-13:
> Hello!
> Can anyone suggest how to adjust the contrast?
> Fee Mega 2560, Ramps 1.4.
> On Marlin 1.1.9 firmware, everything works well.
> [image]
> On the Marlin firmware bugfix - 2.0. X can not increase the [...]

Try this:
[image]

John824, 2019-04-13:
Thank!
Already happened on the Mega 2560.
On the Arduino DUE with the same connection, it is impossible to start the display. Trying to deal with pins.
[image]
[image]

Bob-the-Kuhn, 2019-04-14:
Are you sure you have +5V to the display?  Due usually supplies 3.3V instead of 5V on most connectors with Vcc on them..

John824, 2019-04-14:
Yes of course. This is the first thing I checked.
[image]

John824, 2019-04-14:
Although you gave me the idea to check the other voltages (and not just Vcc).
The voltage between pins 16, 23, 25, 27,29 and GND on the Arduino DUE is 3.3 V. On the Mega 2560 between the same pins 5 V.
It turns out the display is not possible to force to work with DUE?

Bob-the-Kuhn, 2019-04-14:
I just got the display working with a RAMPS_FD_V1 shield.  I had to make the following change to **pins_RAMPS_FD_V1.h**
```
  #if ENABLED(MINIPANEL)
//    #define DOGLCD_CS      25
//    #define DOGLCD_A0      27
    #define DOGLCD_CS        17
    #define DOGLCD_A0        16
  #endif
```

John824, 2019-04-14:
I was immediately installed in the firmware.
[image]

I think that on your board (RAMPS_FD_V1) there is a level converter.

Probably I will need to use something like this: https://ru.aliexpress.com/item/High-Speed-8-Bit-Bidirectional-Voltage-Level-Translator-Module-for-Arduino/32583237273.html?spm=a2g0v.10010108.addToCart.1.68db61d7ITNay9&pvid=1032ed2e-1e58-4e5f-b13f-65546d6e1631&gps-id=pcDetailCartBuyAlsoBuy&scm=1007.12908.99722.0&scm-url=1007.12908.99722.0&scm_id=1007.12908.99722.0

What do you think will help?

Bob-the-Kuhn, 2019-04-14:
Yes, the LCD adapter for the RAMPS_FD_V1 has a 5V logic translator.

You may be right on the 3.3V logic compatibility.  I just tried it on my Re-Arm and all I get is a blank display.  I'll play with it more later today.

Bob-the-Kuhn, 2019-04-15:
I have it working on my Re-Arm board with only 3.3V logic signals.

It even works with a 3.3V supply.  It's just a little dimmer than when on a 5V supply.

Here's the pinout of the connectors:
```

 PIN NUMBERS AS SHIPPED (MKS style)                                                                           
                                                     
            EXP1                      EXP2                      
    BEEPER  10 9  BTN_ENC    MISO     10 9  SCK          
    DOG_CS   8 7  DOG_A0     BTN_EN2   8 7  SD_CSEL      
    N/C      6 5  RED        BTN_EN2   6 5  MOSI         
    GREEN    4 3  BLUE       SD_DET    4 3  RESET BUTTON 
    GND      2 1  Vcc        GND       2 1  RESET BUTTON 
                                                                                                              
                                                                                                              
                                                                                                              
 PIN NUMBERS WITH SHROUDS ROTATED 180 DEGREES (RepRap style)                                                  
                                                                                                              
            EXP1                      EXP2                      
    BEEPER  1  2  BTN_ENC    MISO     1  2  SCK          
    DOG_CS  3  4  DOG_A0     BTN_EN2  3  4  SD_CSEL      
    N/C     5  6  RED        BTN_EN2  5  6  MOSI         
    GREEN   7  8  BLUE       SD_DET   7  8  RESET BUTTON 
    GND     9 10  Vcc        GND      9 10  RESET BUTTON 
```

The pins needed to have something show up on the display are:
DOG_CS
DOG_A0
SCK
MOSI
GND
Vcc

John824, 2019-04-15:
I have an empty screen with the same pinout. Probably our displays are different. Your running from 3.3 V, my not.

Bob-the-Kuhn, 2019-04-15:
Doesn't matter if it is 3.3V or 5V.  All the voltage affects is the brightness of the display.

Please provide the following & I'll see if I can reproduce your setup tomorrow.
- A ZIP with configuration.h, configuration_adv.h and pins_YOUR_BOARD.h
- A connection list
- If you've rotated the shields on the display or not.

John824, 2019-04-15:
Files that you requested in the attachment. This configuration works with the Mega 2560, but does not work with the DUE.
Did not quite understand which list of connections is needed? After all, I do everything on the file pins.
I turned the connectors on the display.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3081675/Marlin.zip)

co8fs, 2019-04-15:
Hi Guys
Can somebody tell me where can I order one of this FYSETC Mini12864 panel?
Thanks.

Sebastian

Brainc0re, 2019-04-15:
https://www.aliexpress.com/item/MINI12864-LCD-Display-Screen-mini-12864-Smart-Display-128x64-5V-Supports-Marlin-DIY-With-SD-Card/32851658211.html?spm=a2g0s.12269583.0.0.4a2133b26TumBT

Bob-the-Kuhn, 2019-04-16:
@John824 - I found part of the problem.  On DUE, MOSI is pin D75 and SCK is D76.  Those are the middle pins on the 6 pin SPI connector near the CPU.

Right now the signals all look good but I still have just the backlight on.  This display is definitely sensitive to timing.  Or maybe it's noise due to the long cables I'm using.

co8fs, 2019-04-16:
Thank You Brainc0re
I ordered one with RGB.I have an MKS MINI12864 and a Rumba Board, But I was not able to make work. Can anybody help me with this.Thanks.
Sebastian

 
 
https://www.aliexpress.com/item/MINI12864-LCD-Display-Screen-mini-12864-Smart-Display-128x64-5V-Supports-Marlin-DIY-With-SD-Card/32851658211.html?spm=a2g0s.12269583.0.0.4a2133b26TumBT

thinkyhead, 2019-04-16:
I received one of these displays and will get into testing it more on Wednesday.

John824, 2019-04-16:
Bob-the-Kuhn, i also paid attention to this.
Yesterday I tried to use these contacts (75 and 76) instead of 51 and 52. This did not work.
I tried to use the level converter, but apparently did not connect it correctly, because even when the input on the converter was turned off, it produced 5 V at the output.

nebman, 2019-04-16:
Just a quick question: If you just apply only power (VCC & GND) to the display, will it light up or something?  
I unsuccessfully tried to get this display working but after some measuring around I'd suspect at least the voltage regulator on mine is broken.

doughy2002, 2019-04-16:
Mine didn't work either. I ended up bridging 2 pins on the chip to get blue led on.

Brainc0re, 2019-04-16:
Mine didn't work too when GND and VCC is connected. I hope the display works with the SKR V1.3 board - when it arrives.

nebman, 2019-04-16:
I use the SKR v1.3 too. The current pins file does not have the definitions yet. But there is another issue where it was posted, if you dont want to add it yourself.

bipsendk, 2019-04-16:
@Brainc0re - I have it working with SKR v1.3

bipsendk, 2019-04-16:
I added a couple of changed files to https://github.com/MarlinFirmware/Marlin/pull/13365 - don't know if some of that has been merged into the repo yet..

bipsendk, 2019-04-16:
OBS - Please observe, that on v2.1 of the display Fysetc has changed to Neopixel control - see https://www.facebook.com/groups/197476557529090/permalink/363916580885086/

nebman, 2019-04-16:
Thanks to @bipsendk for this essential hint of the newer hardware versions!  

Got my V2.1 screen working now on the SKR v1.3. Colors work immediately but on boot but contrast settings are not working, I have to issue a M250 C255 to get readable contrast

nebman, 2019-04-16:
@Brainc0re no you need to `#define FYSETC_MINI_12864`. also make sure you have the lastest git code as today some commits were added

then if you have a v2.1 RGB screen enable the NEOPIXEL options as seen in the facebook link, only change `#define NEOPIXEL_PIN    P1_21`

bipsendk, 2019-04-16:
Maybe a new 
#define FYSETC_MINI_12864_v21
directive is needed for the neopixel version - if some sanitychecks should be added to ensure backlight is enabled on power-up ?

Brainc0re, 2019-04-16:
> @Brainc0re no you need to `#define FYSETC_MINI_12864`. also make sure you have the lastest git code as today some commits were added

I see ;) I used the marlin version from the bigtreetech github...

nebman, 2019-04-17:
@Brainc0re which board and which display hardware revision do you use? from the screenshot I guess it's the SKR v1.3? please dont mistake the two options of MKS and FYSETC, as they use different pins

* get the latest git version of bugfix-2.0.x
* add the right pins to your pins file (either off @bipsendk 's [link ](https://github.com/MarlinFirmware/Marlin/pull/13365) or [here is a link](https://pastebin.com/gHvHrpmU) what I used)
* enable NEOPIXEL options if you have a v2.1 screen or RGB LED options for v2.0 (and maybe below?)

Brainc0re, 2019-04-17:
@nebman 

I tried it yesterday. Board v1.3 and latest marlin build. With the latest build and the standard pin definition, i get the error above. Do i use the pin definition from the [https://github.com/MarlinFirmware/Marlin/pull/13365](here) - same error DOGLCD_CS and DOGLCD_A0. With the pin definition from [https://pastebin.com/gHvHrpmU](here) i can compile it with no error.

I get my board tomorrow. I hope it works at the beginning :D And thank you to all of you for your help

John824, 2019-04-17:
Hello!
Help please fix the compilation error.

[image]

A bunch of DUE + RURAMPS4D + Marlin 2.0.x (last)

VadimPV, 2019-04-17:
Good afternoon, please help me. I can't turn it on.  RGB backlight on the display Fysetc mini 12864 panel v2.1 (marlin 1.19, Ramps 1.4) the image is displayed on the display and the backlight does not scatter, I made all the settings according to the instructions on the https://wiki.fyel.cc/Mini12864_Panel/

[23 more replies]

closed 2019-04-20 by Bob-the-Kuhn