MarlinFirmware/Marlin #13398: USB Connection and gcode execution

murdock62, 2019-03-14:
Hello, and sorry for my bad english.

I noticed 2 curious little things with the 1.1.9 and 1.1.x bugfix versions.
For information the tests were carried out on my 3 creality Ender-3 which I replaced the motherboard with a CR-10S card. (ATMEGA 2560)

First point the USB communication:
When I'm in version 1.1.8 and I plug my 3 ender on the same USB port of my pc with a hub, no problem everything works perfectly.

If I pass one of the 3 machines in 1.1.9 and the other two remain in 1.1.8, this is the drama. In a few minutes the machines lose all their connection with the PC.

If I only connect two machines (a 1.1.9 and a 1.1.8) it works a little better, but the communication is very slow. Connection / disconnection requests can take a good ten seconds to execute.

If I plug the two 1.1.8 together and the 1.1.9 on an isolated USB port all is well.

Second point, the execution of the gcode:
My startup sequence is as follows:

G90
M82
M106 S0
G28 ; home all axes
G1 Z15 F3000 ; lift
G1 X0 Y0 F7800 ;
M140 S50 T5 ;
M190 S50 T5 ;
M104 S200 T0 ;
M109 S200 T0 ;
G92 E0 ; reset extrusion distance
G1 F1500 E-5 ;
G1 F200 E5;
G1 F1500 E-5 ;
G92 E0 ; reset extrusion distance
M117 Nettoyage buse ;
M300 S300 P100 ; 
G4 P500 ; 
M300 S300 P100 ; 
G4 P500 ; 
M300 S300 P100 ;
G4 P5000 ;
M117 Bed calibration ;
G29 ;
...

But the interesting part of this sequence is this one:
G92 E0 ; reset extrusion distance
G1 F1500 E-5 ;
G1 F200 E5;
G1 F1500 E-5 ;
G92 E0 ; reset extrusion distance
M300 S300 P100 ; 
G4 P500 ; 
M300 S300 P100 ; 
G4 P500 ; 
M300 S300 P100 ;
G4 P5000 ;

With version 1.1.8 I had the following thing
-Movement of the extruder.
-Stopping of the extruder.
-Beep
-Short break
-Beep
-Short break
-Beep
-long break

With version 1.1.9 I get the following thing
-Movement of the extruder.
-Beep
-Stopping of the extruder.
-long break
-Beep
-Short break
-Beep
-long break

Curious, no?

Many thanks for your help.
Best regards,

François.

Here is my config files :
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2968834/config.zip)

boelle, 2019-03-15:
have you tried bugfix 2.0?

murdock62, 2019-03-15:
> have you tried bugfix 2.0?
I just tested, same problem ...

boelle, 2019-03-15:
i think it could be bad usb cables or a bad hub, 

how long are the cables?
is it a powered hub?

murdock62, 2019-03-15:
They are armored cables of 1m80. I tested with two hubs one was powered and not the other, same worries. Moreover if the pb was linked only to the usb, why would not it happen with version 1.1.8?

Besides, it certainly does not have a link, but do you have an explanation for the second point? (This one even produces with a single machine directly connected to the pc)

Many thanks for your help !

boelle, 2019-03-15:
1,8 Meters is ok, i know from experience you will only get in to trouble when you get to 5 meters
but if the hub is an unpowered one then the cable from pc to hub and from hub has to be combined, example 2.5 meters from pc to hub and 2.5 meters from hub to printer gives 5 meters

but in your case we can rule out cables

will have to think about this, @thinkyhead can you maybe give a clue what is going on?

boelle, 2019-03-15:
so you have 3 machines that has a cr-10s board in them

i looked at your config and its not correct

you have
```
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif

// Optional custom name for your RepStrap or other custom machine
// Displayed in the LCD "Ready" message
#define CUSTOM_MACHINE_NAME "Ender-3"
```
it should be
```
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_CREALITY
#endif

// Optional custom name for your RepStrap or other custom machine
// Displayed in the LCD "Ready" message
#define CUSTOM_MACHINE_NAME "Ender-3"
```

example config for cr-10s here: 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/CR-10S/Configuration.h
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/CR-10S/Configuration_adv.h

and ender 3:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/Ender-3/Configuration.h
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/Ender-3/Configuration_adv.h

my suggestion is to use the cr-10s config files and take settings from ender3, but only take things that matter like steps/mm and endstop configuration

reloxx13, 2019-03-15:
use m400 to wait finishing move
http://marlinfw.org/docs/gcode/M400.html

murdock62, 2019-03-15:
> use `M400` to wait till (E) move is finished
> http://marlinfw.org/docs/gcode/M400.html

It is odd that the versio 1.1.8 systematically waiting while the 1.1.9 does not.

> #define MOTHERBOARD BOARD_RAMPS_CREALITY

I took the settings of my CR-10S in 1.1.8.At the time there the map "RAMPS_CREALITY" did not exist it seems to me.

> my suggestion is to use the cr-10s config files and take settings from ender3, but only take things 

> that matter like steps/mm and endstop configuration

Yes that's what I did. I just made some changes like feeding the fan of the hotend by the mosfet planned to manage the feeding of the second hotend

boelle, 2019-03-15:
> did not exist it seems to me.

that is because you dont use bugfix 2.0, it has been added after 1.1.8

murdock62, 2019-03-15:
> that is because you dont use bugfix 2.0, it has been added after 1.1.8

Yes when I configured 2.0 (and 1.1.9 for that matter) I kept the habits of version 1.1.8.

It's hard to lose your own habits. :p

murdock62, 2019-03-16:
> use `M400` to wait till (E) move is finished

But the fact that in version 1.1.8 he was waiting without asking for the end of any movement and not on 1.1.9 is voluntary or is it an anomaly?

reloxx13, 2019-03-16:
thats the reason m400 exists

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

thinkyhead, 2019-07-11:
> But the fact that in version 1.1.8 he was waiting without asking for the end of any movement and not on 1.1.9 is voluntary or is it an anomaly?

This was a major improvement that was deployed with 1.1.9. I removed the unnecessary `synchronize` for commands like `G92` by creating something called a "sync block" that sets the position as part of planner processing instead of in-place. This eliminated the need to wait for the planner to finish processing before changing the internal position.

closed 2019-07-11 by thinkyhead