MarlinFirmware/Marlin #13399: Bltouch Anet SERVO0_PIN Pin conflict with ANET_FULL_GRAPHICS_LCD

gustavomassa, 2019-03-15:
Printer: Anet A6
Firmware: Marlin 1.1.9

After struggling for a day, I found out the problem with my official BLTOUCH, the SERVO0_PIN 27 has conflicts with the ANET_FULL_GRAPHICS_LCD.

BLTOUCH Wiring for Anet Board: [https://www.antclabs.com/wiring3](https://www.antclabs.com/wiring3)

BLTOUCH recommended settings for Marlin: [https://docs.wixstatic.com/ugd/f5a1c8_c22571a27da74a5096efbfe8757f2f97.pdf](https://docs.wixstatic.com/ugd/f5a1c8_c22571a27da74a5096efbfe8757f2f97.pdf)

pins_ANET_10.h
```
#if ENABLED(ULTRA_LCD)
  #define LCD_SDSS           28
  #if ENABLED(ADC_KEYPAD)
    #define SERVO0_PIN       27   // free for BLTouch/3D-Touch
    #define LCD_PINS_RS      28
    #define LCD_PINS_ENABLE  29
    #define LCD_PINS_D4      10
    #define LCD_PINS_D5      11
    #define LCD_PINS_D6      16
    #define LCD_PINS_D7      17
    #define BTN_EN1          -1
    #define BTN_EN2          -1
    #define BTN_ENC          -1
    #define ADC_KEYPAD_PIN    1
  #elif ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) || ENABLED(ANET_FULL_GRAPHICS_LCD)
    // Pin definitions for the Anet A6 Full Graphics display and the RepRapDiscount Full Graphics
    // display using an adapter board  // https://go.aisler.net/benlye/anet-lcd-adapter/pcb
    // See below for alternative pin definitions for use with https://www.thingiverse.com/thing:2103748
    #define SERVO0_PIN       29   // free for BLTouch/3D-Touch
    #define BEEPER_PIN       17
    #define LCD_PINS_RS      27
    #define LCD_PINS_ENABLE  28
    #define LCD_PINS_D4      30
    #define BTN_EN1          11
    #define BTN_EN2          10
    #define BTN_ENC          16
    #ifndef ST7920_DELAY_1
      #define ST7920_DELAY_1 DELAY_NS(0)
    #endif
    #ifndef ST7920_DELAY_2
      #define ST7920_DELAY_2 DELAY_NS(63)
    #endif
    #ifndef ST7920_DELAY_3
      #define ST7920_DELAY_3 DELAY_NS(125)
    #endif
    #define STD_ENCODER_PULSES_PER_STEP 4
    #define STD_ENCODER_STEPS_PER_MENU_ITEM 1
  #endif
#else
  #define SERVO0_PIN         27
#endif
```
Anet A6 uses ANET_FULL_GRAPHICS_LCD, which goes to define SERVO0_PIN to pin 29 instead of pin 27. I've already soldered Anet board pin 27 directly to the Bltouch wire, as wiring documentation...

So, I need to force the ANET_FULL_GRAPHICS_LCD to use another pin for LCD_PINS_RS and leave pin 27 for the BLTOUCH.

What LCD_PINS_RS is used for?
Do I have other options?

7eggert, 2019-03-15:
Your instructions regarding which pin to solder were wrong. Thus change
 #define SERVO0_PIN       29   // free for BLTouch/3D-Touch
 #define LCD_PINS_RS      27
to
 #define SERVO0_PIN       27   // switched around 'cause I was fooled
 #define LCD_PINS_RS      29

and cut the corresponding ribbon cables at a convenient place to separate the wires. (Third from red vs. seventh from red (Red == 1), if I'm not mistaken)

gustavomassa, 2019-03-15:
> Your instructions regarding which pin to solder were wrong. Thus change
> #define SERVO0_PIN 29 // free for BLTouch/3D-Touch
> #define LCD_PINS_RS 27
> to
> #define SERVO0_PIN 27 // switched around 'cause I was fooled
> #define LCD_PINS_RS 29
> 
> and cut the corresponding ribbon cables at a convenient place to separate the wires. (Third from red vs. seventh from red (Red == 1), if I'm not mistaken)
> 
> This is not a Marlin problem, though. It's the difference between A8 and A6.

@7eggert 
The Bltouch official documentation just says to use pin 27 for Anet, I think they need to update the doc, since I should use pin 29 rather 27.
I think I've already tried changing LCD_PINS_RS to pin 29, I will try again later.
Right now I'm using the BLTouch on pin 27, without the LCD, I've already cut the LCD ribbon cable 27, that is being used by the BLTouch.

Do you know where is the pin 29 located on the Anet Board 1.5? If you know, I could change the BLTouch  to use pin 29.
I thought that Anet A6 and A8 uses the same board and pins... Do you know any other differences?

7eggert, 2019-03-15:
The A6 does not officially support the BLTouch, there are three original firmwares:
A8, A8 Autolevel and A6.

The board versions differ in the power connectors, positions of the connectors, availability of adjusting motor currents, the rest is compatible AFAIK even for 1.7.

gustavomassa, 2019-03-16:
@7eggert 
Setting #define LCD_PINS_RS 29 did not work, LCD display not working correctly. Any other ideas

boelle, 2019-03-16:
million $$$$: why not use bugfix 2.0? it has config files ready to go for the A6
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Anet/A6

7eggert, 2019-03-16:
gustavomassa did connect his bltouch to pin 27 by soldering, thus the definitions on ANET_FULL_GRAPHICS_LCD will not work for him (it uses pin 27 instead of pin 29, leaving the later free). Now he needs to change the wiring on the ribbon cable and to swap the definitions of pin 27 and 29.
This has to be done in `pins_ANET_10.h` since it overrides the definitions from Config*.h

OTOH, if @gustavomassa needs to cut and solder anyway, he should just use pin 29 and re-connect pin 27.

Also on my test compile, the result was too big. Beyond copying the A6 example config to the main directory, I changed this↓. Then I did a test compile but not more - it should work after setting the location of the probe, but I can't be sure.
```
diff --git a/Marlin/example_configurations/Anet/A6/Configuration.h b/Marlin/example_configurations/Anet/A6/Configuration.h
index d407aa8..54e5f97 100644
--- a/Marlin/example_configurations/Anet/A6/Configuration.h
+++ b/Marlin/example_configurations/Anet/A6/Configuration.h
@@ -75,7 +75,7 @@
 // startup. Implementation of an idea by Prof Braino to inform user that any changes made to this
 // build by the user have been successfully uploaded into firmware.
 #define STRING_CONFIG_H_AUTHOR "(Ralf_E, ANET A6 config)" // Who made the changes.
-#define SHOW_BOOTSCREEN
+//#define SHOW_BOOTSCREEN
 #define STRING_SPLASH_LINE1 SHORT_BUILD_VERSION // will be shown during bootup in line 1
 #define STRING_SPLASH_LINE2 WEBSITE_URL         // will be shown during bootup in line 2
 
@@ -142,7 +142,7 @@
 #define EXTRUDERS 1
 
 // Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
-#define DEFAULT_NOMINAL_FILAMENT_DIA 3.0
+#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
 
 // For Cyclops or any "multi-extruder" that shares a single nozzle.
 //#define SINGLENOZZLE
@@ -417,9 +417,9 @@
  * heater. If your configuration is significantly different than this and you don't understand
  * the issues involved, don't use bed PID until someone else verifies that your hardware works.
  */
-#define PIDTEMPBED
+//#define PIDTEMPBED
 
-//#define BED_LIMIT_SWITCHING
+#define BED_LIMIT_SWITCHING
 
 /**
  * Max Bed Power
@@ -768,9 +768,9 @@
 /**
  * The BLTouch probe uses a Hall effect sensor and emulates a servo.
  */
-//#define BLTOUCH
+#define BLTOUCH
 #if ENABLED(BLTOUCH)
-  //#define BLTOUCH_DELAY 375   // (ms) Enable and increase if needed
+  #define BLTOUCH_DELAY 375   // (ms) Enable and increase if needed
 #endif
 
 /**
@@ -1083,7 +1083,7 @@
  */
 //#define AUTO_BED_LEVELING_3POINT
 //#define AUTO_BED_LEVELING_LINEAR
-//#define AUTO_BED_LEVELING_BILINEAR
+#define AUTO_BED_LEVELING_BILINEAR
 //#define AUTO_BED_LEVELING_UBL
 //#define MESH_BED_LEVELING
 
@@ -1091,7 +1091,7 @@
  * Normally G28 leaves leveling disabled on completion. Enable
  * this option to have G28 restore the prior leveling state.
  */
-//#define RESTORE_LEVELING_AFTER_G28
+#define RESTORE_LEVELING_AFTER_G28
 
 /**
  * Enable detailed logging of G28, G29, M48, etc.
@@ -1598,7 +1598,7 @@
  * just remove some extraneous menu items to recover space.
  */
 //#define NO_LCD_MENUS
-//#define SLIM_LCD_MENUS
+#define SLIM_LCD_MENUS
 
 //
 // ENCODER SETTINGS
diff --git a/Marlin/example_configurations/Anet/A6/Configuration_adv.h b/Marlin/example_configurations/Anet/A6/Configuration_adv.h
index 68c011c..133b2c6 100644
--- a/Marlin/example_configurations/Anet/A6/Configuration_adv.h
+++ b/Marlin/example_configurations/Anet/A6/Configuration_adv.h
@@ -54,7 +54,7 @@
 #if DISABLED(PIDTEMPBED)
   #define BED_CHECK_INTERVAL 5000 // ms between checks in bang-bang control
   #if ENABLED(BED_LIMIT_SWITCHING)
-    #define BED_HYSTERESIS 2 // Only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
+    #define BED_HYSTERESIS 0.2 // Only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
   #endif
 #endif
```

gustavomassa, 2019-03-19:
> OTOH, if @gustavomassa needs to cut and solder anyway, he should just use pin 29 and re-connect pin 27.
Totally agree, the problem is, where is the pin 29 on the board hahaha
My printer and the BLTouch are working very well, without LCD, I would like to use LCD just for status screen..

> PS, my power replacement power supply does not like PIDTEMPBED (it sings), and the Anet power supply are known to explode.
I'm using the original power supply, with mosfet for bed and PIDTEMPBED, it is working very well..

Here is my repo with current settings:
https://github.com/gustavomassa/Marlin

7eggert, 2019-03-19:
Look here:
https://github.com/gustavomassa/Marlin/blob/1.1.x/Marlin/pins_ANET_10.h#L235

The bltouch manual was for the ZONESTAR_LCD, using the N/C pin ("not connected")
You have ANET_FULL_GRAPHICS_LCD

The connector is ANET-LCD..., you want ANET-LCD.7 which is the seventh ribbon wire* on the cable counting from red = 1. Between the wire you used and the wire you should have used, there are three wires.

* guessing the name

boelle, 2019-06-22:
missing a label

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

kiroke, 2019-08-02:
I hope someone reads this post, I have anet A8 main board anet 1.7 firmware installed Marlin 1.1.9, full graphics display the one with the knob on the right under the buzzer and below still reset button.
I would like to mount the bl touch but the two connectors of the LCD display and J3 are occupied by the display cables which change should be made to keep the display and bltouch?
thanks in advance to those who explain me .....

InsanityAutomation, 2019-08-02:
Typically you use a Pin27 board (found on ebay or from th3d) to break out the buzzer pin to use for the BLTouch.

7eggert, 2019-08-02:
Pin #29 if you have graphics display, see pins_ANET_10.h

kiroke, 2019-08-03:
Grazie delle info, allora quale è il pin 29? si trova sulla scheda del display? funzionerà con la mia scheda anet ver. 1.7 e con marlin 1.1.9 come ho detto uso il display full graphic (quello per anet a6) ovvimanete dovrò saldare il filo del bl touch sulla parte posteriore del display o devo tagialere il filo corrispondente? scusatemi ma sono alle prime armi.....

kiroke, 2019-08-03:
volevo aggiungere che attualmente il display full graphic funziona perfettamente con marlin 1.1.9