MarlinFirmware/Marlin #13400: [FR] (Hour Meter, and other counters...)

rsilvers129, 2019-03-15:
The Zortrax M200 keeps tract of printing hours. I love to know this for logging and maintenance purposes. It is much like an odometer on a car.

So my feature request is for the EEPROM to store number of hours and tenths that the machine is printing. Number of print starts. Maybe number of print aborts. Number of prints completed.

reloxx13, 2019-03-15:
```
/**
 * Print Counter
 *
 * Track statistical data such as:
 *
 *  - Total print jobs
 *  - Total successful print jobs
 *  - Total failed print jobs
 *  - Total time printing
 *
 * View the current statistics with M78.
 */
//#define PRINTCOUNTER
```

boelle, 2019-03-20:
@rsilvers129 what @reloxx13 said ^^^

rsilvers129, 2019-04-25:
Thanks. I was not aware.

closed 2019-04-25 by rsilvers129