MarlinFirmware/Marlin #13401: [FR] periodic nozzle auto clean

Andrei-Pozolotin, 2019-03-15:
### Description

Marlin already implements G12 nozzle clean command
http://marlinfw.org/docs/gcode/G012.html

Some high end 3d printers do periodic nozzle auto clean
by automatically executing an equivalent of G12 every few layers.

Is it feasible to introduce similar feature in Marlin?
Where in Marlin code one would start to implement this?

reloxx13, 2019-03-15:
could be doable with postprocessing script which inserts g12 every x layers.
but i dunno if u can execute g12 mid printing in the gcode file.

Andrei-Pozolotin, 2019-03-15:
thank you for the cura plugin link, I am still interested in firmware-level solution

boelle, 2019-03-16:
insert it in start or stop Gcode of the slicer, how would firmware know when to do it?

personal i take a single sheet of toilet paper and fold it to form a small pad
then every 2nd print i carefully wipe the nozzle while its still semi hot (~150C) and it takes off all the molten filament without scratching the nozzle

Andrei-Pozolotin, 2019-03-16:
thank you for the input. do you by chance know a "safe place" in marlin source 
to inject auto-clean hook code w/o messing up the rest of the system (planner, motion advance, etc)?

boelle, 2019-03-16:
you dont have to put in in the firmware

just trigger the clean from slicer like i suggested?

boelle, 2019-06-22:
missing a label

dcwalmsley, 2020-09-02:
So I'm a bit of a NOOB when it comes to writing scripts but I am in agreement that I would like to implement G12 every so many layers (operator's choice) or after so many minutes throughout the print.  In other words repeat cleaning process periodically while print is running to ensure clean nozzle and no residual or build up on nozzle tip that can blemish the print.  This does not replace dialing in the temperature, feed rate, retraction distance and speed, but to help mitigate print failures.

I'd appreciate either an option to include this into firmware or some way to develop this into GCODE for slicing.  

Thanks,
Doug W

vajonam, 2021-11-08:
If you are running Prusa Slicer or Super Slicer you can add this 

`{if (layer_num != 0)}{if (layer_num % 40) == 0}G12 P1 S2{endif}{endif}` simple mod function to do a G12 Wipe every 40 layers. skipping the first or 0th layer. 

quick explanation.

`layer_num` - Slicer's variables that is available for custom GCode
`40` - clean ever 40th layer. 
`G12` - Nozzle Clean
`P1` - Triangles
`S2` - Two Strokes

You can change that however often you like. This is assuming you have G12 in the start g-code, which does a more thorough wipe, this particular is just a quick wipe of `S2` two strokes
 
Here is Pursa/Super Slicer example, but others will have similar places where you can enter it in.
[image]