MarlinFirmware/Marlin #13402: marlin not reporting position after G28 + G0 Z20

ruggb, 2019-03-15:
Marlin-bugfix-2.0.x.9a515cb
RH 2.1.3
not reporting position after this script
G28
G0 Z20

It works perfect EXCEPT RH reports Z is at the offset -0.10

RH log
N11 G28
N12 G0 Z20
X:0.00 Y:0.00 Z:-0.10 E:0.00 Count A:0 B:0 Z:-80
ok
ok

If I then do a G0 Z30 Rh reports Z=30

If I do home Z or home all Then do a G0 Z20 RH reports the correct Z=20.

RH log
N11 G28 z0
X:0.00 Y:0.00 Z:-0.10 E:0.00 Count A:0 B:0 Z:-80
ok
N12 G0 Z20
ok

Is this a Marlin or an RH issue?

boelle, 2019-03-16:
this might sound like a stupid Q, but what is RH ? google did not help me

ruggb, 2019-03-16:
sorry, I didn't start with Repetier Host (RH)

boelle, 2019-03-16:
hmm... read the first post again

what happens if you use pronterface and use M114 to figure where the printer think it is after each command?

that could be used to rule out if its RH or Marlin that is acting up

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-07-24:
@repetier care to comment?

repetier, 2019-07-25:
After a manual G28 with no commands following the host can interpret the following coordinates as homing position.
N11 G28
N12 G0 Z20
X:0.00 Y:0.00 Z:-0.10 E:0.00 Count A:0 B:0 Z:-80

So what happens here is host sees coordinates and thinks it is still good to interpret them as homing coordinates overwriting the last G0 Z20. The following G0 Zx fixes the error then for host. So problem is more that first G0 move was send and host did not disable parsing returned coordinates.

That is a general problem making it nearly impossible to interpret returned coordinates. You do not know which command returns them and if they are still valid at the time you get them.

This is not really a bug in marlin. Repetier-firmware does the same. We even had removed this for a while but then we got many complains about wrong coordinates after homing. Mostly because users did not set homing position in configuration, but also because calibration changed sometimes position especially for z max homing.

So nothing to do on marlin side. I find it still useful to have them after homing and maybe I find a way to solve the case here as well. Normally it is no issue during a print it is ignored and most users do not directly move z after homing. I guess this is a script in host, which would explain why it is treated as manual command with coordinate parsing.

boelle, 2019-08-19:
will close this one as its not related to marlin or repetier

closed 2019-08-19 by boelle