MarlinFirmware/Marlin #13410: Marlin\src\module\temperature.cpp:826:46: error: operator ',' has no right operand[BUG] (short description)

dekra54, 2019-03-16:
When I try to compile the current Marlin 2.0_bugfix with #define PID_DEBUG  enabled i get the following compile error: 
Marlin\src\module\temperature.cpp:826:46: error: operator ',' has no right operand

from 
https://github.com/MarlinFirmware/Marlin/blob/09eb30dc05a727e3816ab5a407f02ada18700e23/Marlin/src/module/temperature.cpp#L821-L829

`      #if DISABLED(PID_OPENLOOP)
        SERIAL_ECHOPAIR(
          MSG_PID_DEBUG_PTERM, work_pid[HOTEND_INDEX].Kp,
          MSG_PID_DEBUG_ITERM, work_pid[HOTEND_INDEX].Ki,
          MSG_PID_DEBUG_DTERM, work_pid[HOTEND_INDEX].Kd
          #if ENABLED(PID_EXTRUSION_SCALING),
            MSG_PID_DEBUG_CTERM, work_pid[HOTEND_INDEX].Kc
          #endif
        );`
I tryed to remove the , after  #if ENABLED(PID_EXTRUSION_SCALING) and add it to MSG_PID_DEBUG_DTERM, work_pid[HOTEND_INDEX].Kd
but this causes additional errors.

boelle, 2019-03-16:
please edit title so it becomes: [BUG] Marlin\src\module\temperature.cpp:826:46: error: operator ',' has no right operand

boelle, 2019-03-16:
and then attach your 2 config files + notepad with compile errors in a single zip file

Ludy87, 2019-03-16:
@bolle The mistake is identified, look into the PR

boelle, 2019-03-16:
ahh so this has been fixed but not closed?

commit 59353546 (2019-03-16): Fix BUTTON_EXISTS and misplaced comma (#13411)

Fixes #13410 and redeclaration of `BUTTON_EXISTS`
closed 2019-03-17 by dekra54