MarlinFirmware/Marlin #13413: Auto level sensor proximity

manarsak, 2019-03-16:
Hi, I want to use this sensor pl-P802 to fix the bed but I do not know how to use it so that the bed will automatically work. I use Marlin2.0

manarsak, 2019-03-16:
I want to run the sensor as it works in the Repetier program but I can not connect the sensor to z max endstop

I want him to check the bed automatically

boelle, 2019-03-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

sjasonsmith, 2020-11-10:
Too little information. Closing.

closed 2020-11-10 by sjasonsmith