MarlinFirmware/Marlin #13414: Babystepping with M851 offset reversed with FIXED_MOUNTED_PROBE.

KevMags, 2019-03-16:
### Steps to Reproduce

Use   FIXED_MOUNTED_PROBE with  BABYSTEP_ZPROBE_OFFSET from Bugfix 2.0 March 7.
Use babystepping to adjust nozzle, exit babystepping and observe change to Z offset.  

**Expected behavior:** [What you expect to happen]
Since the nozzle is the probe, babystepping closer to the bed should decrease z offset.

**Actual behavior:** [What actually happens]
Babystepping the nozzle closer to the bed causes the numeric display to increase and when exiting babystepping the Z offset is increased.  When using the nozzle as probe the offset should decrease.

#### Additional Information

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/2973829/Configs.zip)

boelle, 2019-03-16:
define Z_PROBE_OFFSET_FROM_EXTRUDER 0 

needs to be a positive number

KevMags, 2019-03-16:
I had changed  //#define BABYSTEP_ZPROBE_OFFSET after it didn't work correctly.

By turning my LCD knob clockwise the nozzle gets closer to the bed. When doing so the display for the z offset increases and after exiting baby stepping my offset is increased.  It should decrease since I am using FIXED_MOUNTED_PROBE (nozzle as probe) and in this configuration smaller z offset moves nozzle closer to the bed.

InsanityAutomation, 2019-03-18:
This is actually just fine being a negative number. Its more often negative with retracting or stowing probes and positive with capacitive or IR.

Along with BABYSTEP_ZPROBE_OFFSET can you enable BABYSTEP_ZPROBE_GFX_OVERLAY and see if the graphical icons line up to actual motion?

boelle, 2019-03-20:
@KevMags btw, can you add [BUG] in title? [BUG] Babystepping with M851 offset reversed with FIXED_MOUNTED_PROBE.

boelle, 2019-07-17:
@KevMags did you try what @InsanityAutomation  suggested?

KevMags, 2019-07-18:
@boelle.  No, I am unconcerned about the value being positive or negative.  I just wanted my z offset to be changed in the proper direction according the the babystepping that I use.

When I use babystepping (with the nozzle being the probe) to move the nozzle closer to the bed I need the z offset to be decreased.  The firmware is correcting the z offset in the wrong (opposite) direction than what is needed.  Due to using the nozzle as the probe, the correction is exactly opposite of what would be needed with a typical probe.  The firmware appears to not be aware of this.

boelle, 2019-09-24:
@KevMags is this still an issue?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle