MarlinFirmware/Marlin #13415: changing E0_AUTO_FAN_PIN causes compile error.

KevMags, 2019-03-16:
-->

Using 2.0 Bugfix from March 12.
Setting #define E0_AUTO_FAN_PIN  to any number other than -1 causes compile error.  The bug fix from just a few days before works fine.  I am using a Ramps with fan extender.
### Steps to Reproduce
Change  #define E0_AUTO_FAN_PIN  to any pin number. Attempt compile, observe errors..

**Expected behavior:** [What you expect to happen]
To compile without errors.  I have used this feature for years and the March 7 bugfix worked correctly

**Actual behavior:** [What actually happens]
Numerous errors when compiling.

#### Additional Information

Using Ramps with fan extender.

[Configs and Error message.zip](https://github.com/MarlinFirmware/Marlin/files/2973849/Configs.and.Error.message.zip)

boelle, 2019-03-16:
first error to fix is

```
Multiple libraries were found for "U8glib.h"
```
delete either /Users/Kmags/Documents/Arduino/libraries/U8glib
or 
/Users/Kmags/Documents/Arduino/libraries/PU8glib

you cant have 2 copies of the same

not sure this solved it completely but non the less its an error that is easy to fix

boelle, 2019-03-16:
i use the re-arm+ramps and also the fan extender and i have no issues

#define E0_AUTO_FAN_PIN P1_21 

dont be alarmed over the funny number for the pin, that is just how they look with re-arm

p3p, 2019-03-16:
@KevMags The issue was fixed on 13th with this commit (https://github.com/MarlinFirmware/Marlin/commit/ad91476d26c098d83c8d48aefb98369bae08666b).

KevMags, 2019-03-16:
Deleting multiple U8glib libraries didn't help.
March 16 version fixed it.

closed 2019-03-16 by KevMags

boelle, 2019-03-16:
:-)

RoyBerntsenDesign, 2019-06-02:
I have the same issue again now with june 02/19 download
#define E0_AUTO_FAN_PIN 7 does not compile
[m2 error.txt](https://github.com/MarlinFirmware/Marlin/files/3244957/m2.error.txt)