MarlinFirmware/Marlin #13416: [BUG?] What dou you get NAN + number?

GMagician, 2019-03-16:
in `probe_pt` function there is a
`measured_z = run_z_probe() + zprobe_zoffset;`
but run_zprobe may return NAN when fails...
below this line there is a
`if (isnan(measured_z)) {`
but NAN + number what does it return?

p3p, 2019-03-16:
As a rule NaN propagates (as defined by the standard), if NaN is one of the values the result is always NAN, some std C++ functions will try to gracefully handle it and return a real value as a result even with NaN is passed as an argument, NaN values also have implicit conversion to bool, Nan != 0 therefore NaN == true, but conversion to int is undefined. fun times all round really.

So this is not a bug, NaN + value is always NaN.

GMagician, 2019-03-16:
@p3p this was more a question than a bug report...thanks for your time and explanation

Roxy-3D, 2019-03-16:
> As a rule NaN propagates (as defined by the standard), if NaN is one of the values the result is always NaN, 

And it gets a little more complicated.   You can not check for NaN with a comparison.   For example, This doesn't work:
```
float a;
a = NAN;
if (a==NAN) 
   printf("This will never print.\n");
```

p3p, 2019-03-16:
Ah yes forgot to mention the one, `NaN != NaN`, in fact that is how most checks for NaN work, just compare the variable to itself.