MarlinFirmware/Marlin #13418: [BUG] (Ramps-fd V1 TMC2130 marlin 2.0.x /TMC CONNECTION ERROR)

Miran81, 2019-03-16:
Hello. I'm using Ramps-fd v1+due+tmc2130 and ı searched everything but lcd shows tmc connection error 

x_cs pin 55(aux-2) and ı'm using servo pins for other pins.
[ramps-fd v1 marlin2.0.zip](https://github.com/MarlinFirmware/Marlin/files/2974502/ramps-fd.v1.marlin2.0.zip)
[image]
and when ı'm click SD CARD on screen, card is reseting why?another problem is  when ı plug it in arduino, lcd shows just squares and ı'm click reset button on board, lcd shows normally . why it wants reset firstly ?

coldtobi, 2019-03-17:
Do you have V_motor applied? On my FYSETC F6, I need power, otherwise I'll get the same error.

Miran81, 2019-03-17:
@coldtobi  ı did not operation about V_motor. I don't know what is it . which .h file ?

kAdonis, 2019-03-17:
V_motor  from your picture 12 or 24V

Miran81, 2019-03-17:
24 v wiring is ok but when ı open psu, one of tmc2130 is broken

Miran81, 2019-03-19:
I changed servo and aux-2 pins. Lcd screen TMC CONNECTION ERROR

Bob-the-Kuhn, 2019-03-21:
I think I know what is wrong. I made errors on the AUX 2 logical pins table in that diagram. Try using 60 instead of 55.

Here is a JPG with the AUX-2 logical pin numbers corrected.  I've removed the old JPG from the RepRap WIKI and replaced it with this one.

[image]

Miran81, 2019-03-21:
[cnf_cnfadv_ramps-fd-v1pins.zip](https://github.com/MarlinFirmware/Marlin/files/2992070/cnf_cnfadv_ramps-fd-v1pins.zip)
[image]
Thank u but the same error @Bob-the-Kuhn

kAdonis, 2019-03-21:
By looking at your picture I get the impression, that yout TMC2130 has long SPI Pins soldered, facing upwards **and downwards**, is that correct? This wont work on your board, make sure the four pins facing up have no contact to the socket

Miran81, 2019-03-21:
@kAdonis I check it . There is no problem

Bob-the-Kuhn, 2019-03-22:
I was able to reproduce the problem using your configuration files.  If you make your configuration_adv.h file match the following then your hardware will work as is.
```
  #define X_CS_PIN          60
  #define TMC_USE_SW_SPI
  #define TMC_SW_SCK        7  // S1
  #define TMC_SW_MOSI       6  // S2  SDI on TMC2130
  #define TMC_SW_MISO       5  // S3  SDO on TMC2130
```

Miran81, 2019-03-22:
@Bob-the-Kuhn  That's okey this lines but the same error. Lcd shows tmc connectıon error.What is your  Arduino software version ?

Bob-the-Kuhn, 2019-03-22:
You still have TMC_SW_SCK and TMC_SW_MISO swapped.  

I use PlatformIO to compile and download.  All it takes is four mouse clicks to do everything.

Miran81, 2019-03-23:
@Bob-the-Kuhn yessss ! thank u so much ! I did not understand mıso and mosı. similar words. Okey drivers is ok . I have some questions. When i use platformio, lcd shows just squares. I'm using arduino ide. When i click "No sd card" on the lcd, card is reseting why ? t

Bob-the-Kuhn, 2019-03-23:
What display are you using?  I don't recognize it from your picture.

Miran81, 2019-03-23:
@Bob-the-Kuhn 2004 Lcd

Bob-the-Kuhn, 2019-03-23:
Let me research the 2004.

Can you give me the link as to where you bought the 2004?

I'd only be guessing as to what is happening when compiled via Arduino.

Miran81, 2019-03-23:
#define Sdsupport and reprap_discount_smart_Contr... https://www.aliexpress.com/item/3D-Printer-2004-LCD-Controller-with-SD-card-slot-for-Ramps-1-4-Reprap-Display/32827226360.html?spm=a2g0s.9042311.0.0.27424c4dyiC84e  and sensors not working . I defined '1':"100k / 4.7k - EPCOS" #define TEMP_SENSOR_0 1. https://www.aliexpress.com/item/100K-Ohm-NTC-3950-Thermistors-Temperature-Sensor-With-Cable-Dupont-Head-For-Reprap-Mendel-MK2A-MK2B/32826939230.html?spm=a2g0s.9042311.0.0.57cf4c4dpwcTVQ

Bob-the-Kuhn, 2019-03-23:
Sorry - I'm getting my issues mixed up.

Yes, that is a RepRapDiscount Smart Controller.

I have one.  I'll try it on my test setup.  I may not have thecorrect adapter.  I've modified mine to work with the 128x64 full graphics controller.

It has a SD card slot on it.  Why do you want to disable it?

---

Can you be more specific about the sensor problems you're having?  Are the temperatures way off or are they pegged high (or low) or .... ?

If it's an accuracy problem You can try some of the other settings.  The following are possibilities:
```
3 : Mendel-parts thermistor (4.7k pullup)
11 : 100k beta 3950 1% thermistor (4.7k pullup)
```

What sensor do you have for the extruder(s)?  If you have an option for a PT100 then grab it.  Resolution of the thermistor type sensors is poor at extruder temperatures.  The PT100 is a better solution.

Bob-the-Kuhn, 2019-03-23:
What printer are you using?

Miran81, 2019-03-23:
I don't want disable sdcard. I want to use sd card. Now I have no sd card but when i click "no sd card" on lcd , card is reseted. When  normally time, it will no reset ? My printer is new design but cartesian.

Bob-the-Kuhn, 2019-03-23:
I have bad memories of designing & building my own 3D printer.  I was too new to 3D printing to understand the tradeoffs.  

---

I'm getting lines 1 & 3 as solid rectangles and lines 2 & 4 are blank.  Time to dig into it.

Miran81, 2019-03-23:
yep definitely difficult new desing printer. my extruder sensor https://www.aliexpress.com/item/100K-Ohm-NTC-3950-Thermistors-Temperature-Sensor-With-Cable-Dupont-Head-For-Reprap-Mendel-MK2A-MK2B/32826939230.html?spm=a2g0s.9042311.0.0.57cf4c4dpwcTVQ    #define TEMP_SENSOR_0 13 is 0/0 degree and  #define TEMP_SENSOR_0 3 is 0/16 but false ,so thermistor is not working.

Miran81, 2019-03-23:
by the way voltage regulator for 12v, I have 24v psu, so regulator is much heating. What is your suggestion for 24 v ? 7824 voltage regulator is ok ?

Bob-the-Kuhn, 2019-03-23:
If you haven't invested big $$ then I suggest you consider the Creality Ender 3.  For $200 you get a good printer that can make great prints.  Plus there's all kinds of upgrade plans/tutorials available.

---

DO NOT USE A 7824.  It is meant to supply 24V from a 27+V source.

I ended up removing the 7812 regulator and using an external buck converter to supply the +12V.

Miran81, 2019-03-23:
Ok . Thank you again for everything . I'm wait all advices . Can u share conf,confadv,pins_your board? Just i m wondering

Bob-the-Kuhn, 2019-03-23:
I'm not understanding the temperatures you're reporting.  For example "0/16" - is the "0" for TEMP_SENSOR_BED and the "16" for TEMP_SENSOR_0?

Just to be safe - the bed thermistor connects to the first pair of pins (labeled THERM0 in the picture), the first extruder is connected to the next pair, …

---

I use my system mostly to mimic problems users are having.  That means it's always being modified. The only part of my files that I think will be of any help to you is:
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
```

I then connect my USB cable to the native USB port on the DUE.  

The net result is all my host info & downloading is through the native port and, when an SD card is present, it automatically shows up as a drive on your PC.

Bob-the-Kuhn, 2019-03-23:
I just used Arduino to compile and my 2004 display sprang to life.  

Now I understand the temperatures.  They are from the LCD.  The first part is the reported temperature.  The second part is the setpoint for when the bed/extruder is enabled.

I'm thinking the PlatformIO issue is the wrong library is being pulled in.  Time to do some playing.

Bob-the-Kuhn, 2019-03-23:
It's definitely a library issue.  I've opened issue #13461 for this.

Miran81, 2019-03-23:
Okey Thanks a lot ..

Bob-the-Kuhn, 2019-03-23:
I just grabbed the files you posted two days ago and downloaded them.  Both temperature inputs are acting normally.

You have an open or high impedance in your thermistor connections.

Miran81, 2019-03-24:
I'm using new thermistor which is from link. I'm using without heater. Maybe problem is using without heater (extruder or heatbed) . I think it shows room temperature right ?

Bob-the-Kuhn, 2019-03-25:
For each of you strange readings, what is the thermocouple, what does the LCD say and what thermistor table are you using?

Miran81, 2019-04-14:
It shows -26/0 degree . may I use 4.7k pullup with thermistor ? @Bob-the-Kuhn

boelle, 2019-07-17:
@Miran81 is the bug still present?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

Miran81, 2019-07-21:
It did not solve and i changed the board. I mean i have one problem today and solved , tomorrow one more problem . So changed board

boelle, 2019-07-21:
so maybe it was hardware failure? what board did you get instead?

Miran81, 2019-07-22:
I think hardware problem. Lcd not working properly, tmc2130 not working i think pinsout wrong etc. I'm using ramps 1.4 now