MarlinFirmware/Marlin #13423: [BUG?] (Leveling offset differs between first and second G29)

viebvo, 2019-03-17:
### Description

After upgrading to the 2.x branch of Marlin i am trying to track down some issues with Mesh bed levelling and stumbled across something i suspect may be a bug in the levelling code:
There is roughly 0.1mm difference in Z-Offset between the first and second levelling runs after a G28

### Steps to Reproduce

G28
G29
17:08:32.753: 0 -0.338 -0.229 -0.207
17:08:32.754: 1 -0.177 -0.078 -0.111
17:08:32.755: 2 +0.080 +0.228 +0.147
G29
17:10:17.947: 0 -0.457 -0.347 -0.323
17:10:17.948: 1 -0.294 -0.195 -0.227
17:10:17.950: 2 -0.033 +0.118 +0.037
G29
17:12:02.604: 0 -0.460 -0.352 -0.327
17:12:02.604: 1 -0.299 -0.201 -0.232
17:12:02.608: 2 -0.036 +0.116 +0.031
G28
G29
17:17:13.494: 0 -0.335 -0.234 -0.210
17:17:13.494: 1 -0.179 -0.077 -0.111
17:17:13.498: 2 +0.075 +0.229 +0.143
G29
17:18:54.719: 0 -0.443 -0.339 -0.316
17:18:54.719: 1 -0.288 -0.182 -0.216
17:18:54.723: 2 -0.034 +0.122 +0.036
G29
17:20:23.361: 0 -0.442 -0.337 -0.313
17:20:23.361: 1 -0.290 -0.185 -0.217
17:20:23.364: 2 -0.037 +0.118 +0.034

**Expected behavior:** 

Z-Offset values should be roughly the same.

**Actual behavior:**

First G29 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2975420/Marlin.zip)
offset differs by approx 0.1mm compared to subsequent G29

boelle, 2019-03-17:
and its not the hardware that is at fault? ie loose screw etc?

boelle, 2019-03-17:
i see you have this enabled:

#define Z_MIN_PROBE_REPEATABILITY_TEST

what is the output when you send M48 ?

viebvo, 2019-03-17:
20:30:54.975: M48 Z-Probe Repeatability Test
20:31:24.663: Finished!
20:31:24.664: Mean: 0.006538 Min: 0.004 Max: 0.008 Range: 0.005
20:31:24.666: Standard Deviation: 0.001390
Different Position:
20:45:54.934: M48 Z-Probe Repeatability Test
20:46:24.651: Finished!
20:46:24.653: Mean: 0.019840 Min: 0.018 Max: 0.022 Range: 0.004
20:46:24.654: Standard Deviation: 0.001000

I believe i can roule out hardware related issues and i always have a G28/G29 combination in my gcode so the issue never appears during normal operation.
I cross-checked with another printer which is still running Marlin 1.9.x and strangely enough it shows the exact same behavior: 0.1mm difference between first levelling after G28 and all subsequent ones...
(i can upload all information if you like)

boelle, 2019-03-17:
yep
20:46:24.654: Standard Deviation: 0.001390
and 
20:46:24.654: Standard Deviation: 0.001000

it's a 0.000390mm difference

and there is nothing "not-normal" that maybe 1 G29 out of 5 is a bit off

we can only speculate in why, at least i know to little about it, i would be worried if the G29's where all over the place and never near the same

viebvo, 2019-03-17:
Sorry for my bad English understanding - so you think it is a mechanical issue?
My assumption is, that sensors like a SN08-N are never as accuarate as endstops or Bltouch sensors - but a deviation of 0.001x (i assume that is in mm) would be neglectable versus a difference of 0.1mm in above test situation?

boelle, 2019-03-17:
> so you think it is a mechanical issue?
yes, but it's only a problem if its every 2 G29 that differes so much, 1 out of 5 is not a problem

try and run M48 maybe 5 or times times if the Standard Deviation number is always near 0.001000mm then it must be something else

i would check my Z axis very carefull, but what i can tell you that its not in firmware

but again, if only 1 out of 5 G29 is off by 0.1 then i would not worrie

boelle, 2019-03-17:
you will never the same result every time, there will always be a small difference

but if it jumps all over the place you have a problem

viebvo, 2019-03-17:
Just to be sure that we are talking about the same:
- do you think that the probe repeatability test looks ok (see below)?
- did you take into account that my original test case (first G29 and second G29 - never any of the subsquent ones) shows a difference which is 2 orders of magnitude higher than the M48? (M48 deviation 0.001x mm vs. G29 difference 0.1x mm)?

Thanks for your patience!!

Some M48 in a row:

21:36:26.168: Mean: 0.002623 Min: -0.001 Max: 0.005 Range: 0.006
21:36:26.170: Standard Deviation: 0.001581
21:36:55.705: Mean: 0.002198 Min: -0.000 Max: 0.004 Range: 0.004
21:36:55.707: Standard Deviation: 0.001097
21:37:25.265: Mean: -0.000396 Min: -0.002 Max: 0.001 Range: 0.003
21:37:25.266: Standard Deviation: 0.000962
21:37:54.801: Mean: -0.004217 Min: -0.007 Max: -0.001 Range: 0.005
21:37:54.803: Standard Deviation: 0.002135
21:38:24.343: Mean: -0.010207 Min: -0.014 Max: -0.006 Range: 0.008
21:38:24.345: Standard Deviation: 0.002398

boelle, 2019-03-17:
> * do you think that the probe repeatability test looks ok (see below)?

yes

> * did you take into account that my original test case (first G29 and second G29 - never any of the subsquent ones) shows a difference which is 2 orders of magnitude higher than the M48? (M48 deviation 0.001x mm vs. G29 difference 0.1x mm)?

you cant compare M48 and G29

but your G29's look ok

viebvo, 2019-03-17:
Ok - got it, thanks a lot!

With a M48 V4 i can see the raw values of the repeatability test which should be comparable to the G29 output. RESTORE_LEVELING_AFTER_G28 is activated btw

Here's a full transcript of the initially mentioned issue with repeatability tests in between.
Please note the different Z values after the second G29.

As said before, this is something that doesn't affect me during daily prints as i always use G28/G29 at the start of each print. So please feel free to close this issue if you don't think it is something worth to be researched further!

Thank you!

**22:08:36.887: N28674 G28**
22:08:49.287: echo:Active Extruder: 0
22:08:49.290: X:129.00 Y:106.00 Z:5.00 E:1475.82 Count X:22720 Y:16960 Z:10600
22:08:53.980: N28675 G1 X100 Y100
22:08:58.816: N28676 M48 V4
22:08:58.819: M48 Z-Probe Repeatability Test
22:08:58.823: Positioning the probe...
22:09:03.495: Bed X: 68.000 Y: 100.000 Z: -0.042
22:09:05.997: 1 of 10: **z: -0.036** mean: -0.0363 sigma: 0.000000 min: -0.036 max: -0.036 range: 0.000
22:09:08.501: 2 of 10: z: -0.037 mean: -0.0366 sigma: 0.000236 min: -0.037 max: -0.036 range: 0.000
22:09:11.018: 3 of 10: z: -0.037 mean: -0.0368 sigma: 0.000385 min: -0.037 max: -0.036 range: 0.001
22:09:13.522: 4 of 10: z: -0.035 mean: -0.0365 sigma: 0.000698 min: -0.037 max: -0.035 range: 0.002
22:09:16.025: 5 of 10: z: -0.036 mean: -0.0363 sigma: 0.000667 min: -0.037 max: -0.035 range: 0.002
22:09:18.530: 6 of 10: z: -0.036 mean: -0.0363 sigma: 0.000609 min: -0.037 max: -0.035 range: 0.002
22:09:21.034: 7 of 10: z: -0.034 mean: -0.0359 sigma: 0.001140 min: -0.037 max: -0.034 range: 0.004
22:09:23.541: 8 of 10: z: -0.034 mean: -0.0357 sigma: 0.001247 min: -0.037 max: -0.034 range: 0.004
22:09:26.045: 9 of 10: z: -0.034 mean: -0.0356 sigma: 0.001238 min: -0.037 max: -0.034 range: 0.004
22:09:28.548: 10 of 10: z: -0.035 mean: -0.0355 sigma: 0.001175 min: -0.037 max: -0.034 range: 0.004
22:09:28.549: Finished!
22:09:28.551: Mean: -0.035537 Min: -0.037 Max: -0.034 Range: 0.004
22:09:28.552: Standard Deviation: 0.001175
22:09:28.557: X:100.00 Y:100.00 Z:7.17 E:1475.82 Count X:18080 Y:16000 Z:15207
**22:09:38.479: N28677 G29**
22:10:09.507: Bilinear Leveling Grid:
22:10:09.508: 0 1 2
22:10:09.510: 0 **-0.332** -0.222 -0.197
22:10:09.510: 1 -0.189 -0.081 -0.109
22:10:09.514: 2 +0.053 +0.208 +0.142
22:10:09.833: X:206.00 Y:147.00 Z:5.00 E:1475.82 Count X:35040 Y:23520 Z:10600
22:10:09.834: N28678 G1 X100 Y100
22:10:09.840: N28679 M48 V4
22:10:09.844: M48 Z-Probe Repeatability Test
22:10:09.846: Positioning the probe...
22:10:15.584: Bed X: 68.000 Y: 100.000 Z: -0.168
22:10:18.086: 1 of 10: **z: -0.162** mean: -0.1618 sigma: 0.000000 min: -0.162 max: -0.162 range: 0.000
22:10:20.591: 2 of 10: z: -0.162 mean: -0.1620 sigma: 0.000236 min: -0.162 max: -0.162 range: 0.000
22:10:23.094: 3 of 10: z: -0.163 mean: -0.1623 sigma: 0.000385 min: -0.163 max: -0.162 range: 0.001
22:10:25.599: 4 of 10: z: -0.163 mean: -0.1625 sigma: 0.000527 min: -0.163 max: -0.162 range: 0.001
22:10:28.103: 5 of 10: z: -0.164 mean: -0.1628 sigma: 0.000667 min: -0.164 max: -0.162 range: 0.002
22:10:30.607: 6 of 10: z: -0.161 mean: -0.1624 sigma: 0.000930 min: -0.164 max: -0.161 range: 0.003
22:10:33.113: 7 of 10: z: -0.162 mean: -0.1624 sigma: 0.000889 min: -0.164 max: -0.161 range: 0.003
22:10:35.617: 8 of 10: z: -0.162 mean: -0.1623 sigma: 0.000832 min: -0.164 max: -0.161 range: 0.003
22:10:38.123: 9 of 10: z: -0.163 mean: -0.1624 sigma: 0.000795 min: -0.164 max: -0.161 range: 0.003
22:10:40.626: 10 of 10: z: -0.163 mean: -0.1625 sigma: 0.000795 min: -0.164 max: -0.161 range: 0.003
22:10:40.627: Finished!
22:10:40.628: Mean: -0.162471 Min: -0.164 Max: -0.161 Range: 0.003
22:10:40.630: Standard Deviation: 0.000795
22:10:40.635: X:100.00 Y:100.00 Z:7.05 E:1475.82 Count X:18080 Y:16000 Z:14936
**22:10:40.636: N28680 G29**
22:11:11.687: Bilinear Leveling Grid:
22:11:11.688: 0 1 2
22:11:11.690: 0 **-0.453** -0.340 -0.315
22:11:11.690: 1 -0.306 -0.198 -0.226
22:11:11.693: 2 -0.067 +0.087 +0.023
22:11:12.014: X:206.00 Y:147.00 Z:5.00 E:1475.82 Count X:35040 Y:23520 Z:10600
22:11:12.015: N28681 G1 X100 Y100
22:11:12.022: N28682 M48 V4
22:11:12.025: M48 Z-Probe Repeatability Test
22:11:12.028: Positioning the probe...
[... 16 lines cut ...]
22:13:24.035: Bilinear Leveling Grid:
22:13:24.036: 0 1 2
22:13:24.038: 0 -0.459 -0.349 -0.321
22:13:24.038: 1 -0.310 -0.206 -0.234
22:13:24.040: 2 -0.073 +0.081 +0.017
22:13:24.361: X:206.00 Y:147.00 Z:5.00 E:1475.82 Count X:35040 Y:23520 Z:10600
22:13:24.361: N28684 G1 X100 Y100
22:13:24.367: N28685 M48 V4
22:13:24.371: M48 Z-Probe Repeatability Test
22:13:24.374: Positioning the probe...
22:13:30.108: Bed X: 68.000 Y: 100.000 Z: -0.161
22:13:32.610: 1 of 10: z: -0.159 mean: -0.1585 sigma: 0.000000 min: -0.159 max: -0.159 range: 0.000
22:13:35.115: 2 of 10: z: -0.156 mean: -0.1571 sigma: 0.001415 min: -0.159 max: -0.156 range: 0.003
22:13:37.618: 3 of 10: z: -0.156 mean: -0.1568 sigma: 0.001238 min: -0.159 max: -0.156 range: 0.003
22:13:40.122: 4 of 10: z: -0.157 mean: -0.1567 sigma: 0.001074 min: -0.159 max: -0.156 range: 0.003
22:13:42.627: 5 of 10: z: -0.157 mean: -0.1568 sigma: 0.000971 min: -0.159 max: -0.156 range: 0.003
22:13:45.132: 6 of 10: z: -0.158 mean: -0.1569 sigma: 0.000930 min: -0.159 max: -0.156 range: 0.003
22:13:47.635: 7 of 10: z: -0.155 mean: -0.1566 sigma: 0.001155 min: -0.159 max: -0.155 range: 0.004
22:13:50.142: 8 of 10: z: -0.159 mean: -0.1569 sigma: 0.001248 min: -0.159 max: -0.155 range: 0.004
22:13:52.646: 9 of 10: z: -0.156 mean: -0.1567 sigma: 0.001234 min: -0.159 max: -0.155 range: 0.004
22:13:55.159: 10 of 10: z: -0.157 mean: -0.1567 sigma: 0.001171 min: -0.159 max: -0.155 range: 0.004
22:13:55.160: Finished!
22:13:55.161: Mean: -0.156717 Min: -0.159 Max: -0.155 Range: 0.004
22:13:55.162: Standard Deviation: 0.001171
22:13:55.166: X:100.00 Y:100.00 Z:7.05 E:1475.82 Count X:18080 Y:16000 Z:14950

boelle, 2019-03-17:
i can't click close only you can with the botton below (i can't see it since i did not create the issue)

but i can say with 100% confidence that your printer is ok and firmware is ok

viebvo, 2019-03-17:
I'll accept this as standard behaviour and colse the issue.
Again thanks for your patience!

closed 2019-03-17 by viebvo