MarlinFirmware/Marlin #13426: [FR] Different in/out acceleration

ArtemKuchin, 2019-03-17:
I ask myself why acceleration should be the same when accelerating and deceleration?
For example, take 90 degree corner. If we decelerate jantly and accelerate fast there will be no overshooting on the corner because acceleration is low before the turn and the corner will be sharp, because acceleration is hight after the turn.

I realize, that S-curve acc  and LIN_ADVANNCE help in the describe case, but i think, having separate decelaration/acceleration  values might help even more.

qwewer0, 2021-06-02:
I support the idea for having an option for separate acceleration and deceleration values for printing moves. (it might not be useful for retraction and travel moves)

By my understanding, if for example, we use a well tuned printer, that can print without ringing/echoing:
- If we double the acceleration and by that the deceleration too, then that will cause ringing/echoing on the printed parts, but
- If we only double the acceleration, but keep the deceleration as it was, then there won't be ringing/echoing on the printed parts, or not as much

And a lover deceleration - relative to the acceleration - can reduce the top speed of short zig-zag moves, and thus reduce printer vibration from these moves

espr14, 2023-12-19:
Makes sense. On the other hand, limits for the printer should be different. Printer can decelerate faster because friction adds extra deceleration force. In case of the Z axis the gravitation helps accelerate downwards and decelerate upwards. This could be handeled as well.