MarlinFirmware/Marlin #13427: [FR] Controller Fan not activated by HeatBed, if HeatBed is driven by Solid State Relais or External MosFet

GeorgZiegler, 2019-03-17:
### Description
There is no need to activate the Controllerboard Fan, if the Heatbed is supplied by
an external Solid State Relais or an external Mos Fet Driver.
### Steps to Reproduce
So i suggest following:
in Configuration_adv.h
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 17    
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disable
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
**#define CONTROLLERFAN_HAS_EXT_HEATBED //No need for Controller Fan, if Heat Bed is  supplied by seperate external Solid State Relay or MosFET Driver**
#endif
and in controllerfan.cpp ( Marlin 2.0 ) or in Marlin_Main.cpp ( Marlin 1.1.9 )
void controllerfan_update() {
  static millis_t lastMotorOn = 0, // Last time a motor was turned on
                  nextMotorCheck = 0; // Last time the state was checked
  const millis_t ms = millis();
  if (ELAPSED(ms, nextMotorCheck)) {
    nextMotorCheck = ms + 2500UL; // Not a time critical function, so only check every 2.5s
   // If any of the drivers or the bed are enabled...
    if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON
**#ifndef CONTROLLERFAN_HAS_EXT_HEATBED
  //We don't need the fan if Heated Bed is driven by an external Solid State Relais or MosFet**
		#if HAS_HEATED_BED
          || thermalManager.soft_pwm_amount_bed > 0 
		#endif
	  #endif
'

reloxx13, 2019-03-18:
Configuration_adv.h
```
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 17
  #define CONTROLLERFAN_SECS 60 // Duration in seconds for the fan to run after all motors are  disable
  #define CONTROLLERFAN_SPEED 255 // 255 == full speed
  //mod add define CONTROLLERFAN_HAS_EXT_HEATBED:
  #define CONTROLLERFAN_HAS_EXT_HEATBED //No need for Controller Fan, if Heat Bed is supplied by seperate external Solid State Relay or MosFET Driver
#endif
```

 controllerfan.cpp ( Marlin 2.0 ) or in Marlin_Main.cpp ( Marlin 1.1.9 )
```
void controllerfan_update() {
  static millis_t lastMotorOn = 0, // Last time a motor was turned on
  nextMotorCheck = 0; // Last time the state was checked
  const millis_t ms = millis();

  if (ELAPSED(ms, nextMotorCheck)) {
    nextMotorCheck = ms + 2500UL; // Not a time critical function, so only check every 2.5s
    // If any of the drivers or the bed are enabled...
    if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON
      //mod add ifndef:
      #ifndef CONTROLLERFAN_HAS_EXT_HEATBED
        //We don't need the fan if Heated Bed is driven by an external Solid State Relais or MosFet
        #if HAS_HEATED_BED || thermalManager.soft_pwm_amount_bed > 0
        ...
        #endif
      #endif
    ... 
  }
  ...
}
```

boelle, 2019-03-20:
you will still need to cool the stepper drivers right?

ManuelMcLure, 2019-03-20:
The fix really should really just change the
``      #if HAS_HEATED_BED``
line with
``      #if HAS_HEATED_BED && DISABLED(CONTROLLERFAN_HAS_EXT_HEATBED)``
That will still keep the drivers cooled when enabled but will have the desired effect with not keeping the fan on to try to cool the hotbed MOSFET.

GeorgZiegler, 2019-03-21:
@boelle : Yes, i still need to cool the stepper drivers and the Hot End MosFET.
But due the 'external' , 'not on the Motherboard' placed Solid State Relais ( or an external Mosfet Driver )
the origin MosFET  on the Motherboard has not to drive a high current.

thisiskeithb, 2021-11-28:
Duplicate of #11253

closed 2021-11-28 by thisiskeithb