MarlinFirmware/Marlin #13428: Z Safe Homing ignored

Tailslide, 2019-03-18:
Hi, I am having trouble with Z safe homing on the latest 2.0 bugfix version on a new machine.
I am using the feature successfully with the release version of marlin on an older machine.

Whenever I try to auto home with G28, the machine homes X and Y properly then ignores the safe homing spot for Z and tries to home Z while still at 0,0.  I ran it with debugging enabled and the output is below. It's strange because the debug output says it is moving to the correct spot but it doesn't actually move.   I can manually jog the bed to that spot no problem if I home X and Y. One thing different about this machine compared to my old one is it has dual Y steppers and endstops.  

SENT: M111 S247
READ: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
READ: ok
SENT: M105
READ: echo:M105
M105
SENT: G28
READ: echo:G28
G28
READ: >>> G28
READ: Machine Type: Cartesian
READ: Probe: FIX_MOUNTED_PROBE
READ: Probe Offset X:0 Y:0 Z:0.00 (Aligned With & Same Z as Nozzle)
READ: Auto Bed Leveling: LINEAR
READ:  (disabled)
READ: 
READ: Raise Z (before homing) to 5.00
READ: >>> do_blocking_move_to(0.00, 0.00, 5.00)
READ: <<< do_blocking_move_to
READ: >>> homeaxis(X)
READ: Home 1 Fast:
READ: >>> do_homing_move(X, -637.50, [30.00])
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(X)
READ: Move Away:
READ: >>> do_homing_move(X, 5.00, [30.00])
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(X)
READ: Home 2 Slow:
READ: >>> do_homing_move(X, -10.00, 15.00)
READ: <<< do_homing_move(X)
READ: >>> set_axis_is_at_home(X)
READ: Axis X home_offset = 0.00 position_shift = 0.00
READ: > home_offset[X] = 0.00
READ:   current_position=(0.00, 0.00, 5.00) : 
READ: <<< set_axis_is_at_home(X)
READ:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
READ:   current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
READ: <<< homeaxis(X)
READ: >>> homeaxis(Y)
READ: Home 1 Fast:
READ: >>> do_homing_move(Y, -1350.00, [30.00])
READ: <<< do_homing_move(Y)
READ: Move Away:
READ: >>> do_homing_move(Y, 5.00, [30.00])
READ: echo:busy: processing
busy: processing
READ: <<< do_homing_move(Y)
READ: Home 2 Slow:
READ: >>> do_homing_move(Y, -10.00, 15.00)
READ: <<< do_homing_move(Y)
READ: >>> set_axis_is_at_home(Y)
READ: Axis Y home_offset = 0.00 position_shift = 0.00
READ: > home_offset[Y] = 0.00
READ:   current_position=(0.00, 0.00, 5.00) : 
READ: <<< set_axis_is_at_home(Y)
READ:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
READ:   current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
READ: <<< homeaxis(Y)
READ: Z_SAFE_HOMING >>>
READ:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
READ:   destination=(420.00, 1.00, 5.00) : Z_SAFE_HOMING
READ: >>> do_blocking_move_to(420.00, 1.00, 5.00)
READ: <<< do_blocking_move_to
READ: >>> homeaxis(Z)
READ:   current_position=(420.00, 1.00, 5.00) : set_probe_deployed
READ: deploy: 1
READ: do_probe_raise(10.00)
READ: >>> do_blocking_move_to(420.00, 1.00, 10.00)
READ: <<< do_blocking_move_to
READ: echo:busy: processing
busy: processing
READ: >>> do_blocking_move_to(420.00, 1.00, 10.00)
READ: <<< do_blocking_move_to
READ: Home 1 Fast:
READ: >>> do_homing_move(Z, -150.00, [3.00])
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2975930/Configuration.zip)

ManuelMcLure, 2019-03-18:
As of d20eab4 ``Z_SAFE_HOMING`` works fine for me with the default ``Z_SAFE_HOMING_[X|Y]_POINT`` settings, so it's not a general problem with ``Z_SAFE_HOMING``.

Tailslide, 2019-03-18:
Can you please post your configuration and configuration_adv?  I maybe there is some setting that is different between us that is causing the issue for me.

ManuelMcLure, 2019-03-18:
I just posted my configuration in #13429 along with a video showing ``Z_SAFE_HOMING`` working for me.

reloxx13, 2019-03-18:
do u have xyz MAX endstop?
cuz its enabled in ur config.

```
//FSIGAP enabled next three
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
```

If yes, is direction (inverted/non-inverted) correct?

Tailslide, 2019-03-19:
Yeah that was the problem.. I am using Y max as the second Ymin endstop for the dual steppers but the other endstops were enabled. When it went to move to the safe homing spot the X max endstop showed up triggered so it just did a bed crash as 0,0 instead of giving up or showing a message.  Thanks!

reloxx13, 2019-03-19:
great, pls close this issue 👍

Tailslide, 2019-03-19:
Whoops!

closed 2019-03-19 by Tailslide