MarlinFirmware/Marlin #13429: [BUG] Weird interaction between PRINTER_EVENT_LEDS and PROBING_HEATERS_OFF/WAIT_FOR_BED_HEATER

ManuelMcLure, 2019-03-18:
### Description

If you preheat the bed before homing with a probe (``Z_SAFE_HOMING`` enabled), and have ``PRINTER_EVENT_LEDS``, ``PROBING_HEATERS_OFF``, ``WAIT_FOR_BED_HEATER`` and ``MULTIPLE_PROBING`` enabled, while the probing is happening the status LEDs flash purple and blue several times while the homing is in progress.

### Steps to Reproduce

1. Enable ``Z_SAFE_HOMING``, ``PRINTER_EVENT_LEDS``, ``PROBING_HEATERS_OFF`` and ``MULTIPLE_PROBING`` (I used 2 for ``MULTIPLE_PROBING``)
2. Execute the following g-code:
```
M190 S65 ; preheat bed
G4 S20 ; wait for bed to stabilize
G28 ; home
M109 S205 ; heat the hotend
```
3. Watch the printer LEDs. 

**Expected behavior:**
I would expect the LEDs to turn from blue to purple as the bed heats, remain purple throughout the homing process, and then fade from purple to red as the hotend heats.

**Actual behavior:** [What actually happens]
The LEDs will initially slowly turn from blue to purple (correct), but during Z homing will change several times back and forth between blue and purple (incorrect). Once homing is complete the LEDs go back to purple and then fade to red as the hotend heats up (correct). See the video linked below.

#### Additional Information
Marlin version: bugfix-2.0.x as of d20eab4
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2976398/configuration.zip)
[Video of the behavior](https://www.youtube.com/watch?v=PDweqwy3JN8)
The probe is an 8mm sensing distance NPN inductive sensor.

Obviously this is not a high priority issue - this has no effect on the actual print.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@ManuelMcLure is this still an issue?

ManuelMcLure, 2019-09-27:
Yes. I still see this on the latest 2.0.x-bugfix. However, this is not anything that should stop the 2.0 release - it can be moved on to the next release milestone.

boelle, 2019-10-21:
came back to this one, do you have an idea what is wrong and where?

ManuelMcLure, 2019-10-21:
Not really. I expect it's something like the firmware changing states between "heating" (with the fading colors) and "printing" (which switches back to whatever color was set when heating started up) but I haven't dug that deeply into the code yet.

ManuelMcLure, 2019-10-21:
Perhaps we need a way to flag issues as "future release" so they can be filtered out?

boelle, 2019-10-21:
we can, we have milestones, but that means they need to be fixed before a certain version number is released

firepower9966, 2019-12-03:
Also if you preheat pla/abs there is no PRINTER_EVENT_LEDS (neopixels don't change color).

Tannoo, 2020-05-02:
I remember something like this back a while ago. I thought it was something to do with the scaling being a bit off. I was working on a fix, but life got in the way.

arifbalik, 2020-05-02:
Well I'm guessing the reason for that is Marlin calls `wait_for_bed_heating` in `do_homing_move` 
```C
#if HOMING_Z_WITH_PROBE && HAS_HEATED_BED && ENABLED(WAIT_FOR_BED_HEATER)
    // Wait for bed to heat back up between probing points
    if (axis == Z_AXIS && distance < 0)
      thermalManager.wait_for_bed_heating();
  #endif
```
and it calls (only if axis is Z) the `onBedHeating` in `printer_event_leds.cpp`.

So since you immediately call `M109`, Marlin calls `onBedHeating` AND `onHotendHeating`, which causes the change in color.

Normally `wait_for_bed_heating` only works if bed is heating, but the function who checks that simply looks at the target and current temperatures which constantly changes. I see at some point at your video just before leds going crazy, bed's pwm led stops, therefore temperature drops a little below target temperature, therefore you enter `wait_for_bed_heating`.

To be sure you could comment out those lines in `motion.cpp` so it can not call `wait_for_bed_heating` in the `do_homing_move` function.

Or you could remove `M109` line and see if leds alternate between blue and purple.

Tannoo, 2020-05-06:
I have updated and finally got my printer working with the latest release. I have not loaded the bugfix version yet.
In 2.0.5.3, the RGB LEDs work fine for me.

ManuelMcLure, 2020-05-06:
You have WAIT_FOR_BED_HEATING enabled?

Tannoo, 2020-05-06:
Let me get back to you on that after I get home and check.

Tannoo, 2020-05-06:
I found "WAIT_FOR_BED_HEATER" and that has to do with probing heaters. It is off for me.
I cannot find "WAIT_FOR_BED_HEATING".
But, there is M190 to do the same thing.. that sets the target temp and wait for it to reach it.

Tannoo, 2020-05-06:
That is what happens with the sliced code from Cura does for me. Calls M190 for the bed and then calls M109 for the extruder.

While waiting for the bed temp, the leds start at blue and fade to purple.
While waiting for the hotend, the leds fade from purple to red.
This is how they are supposed to work.

Tannoo, 2020-05-06:
there must be something in the bugfix branch that I haven't looked at in over a year.

Tannoo, 2020-05-06:
In the file "printer_events_leds.cpp':
`#if HAS_TEMP_HOTEND

  void PrinterEventLEDs::onHotendHeating(const float &start, const float &current, const float &target) {
    const uint8_t blue = pel_intensity(start, current, target);
    if (blue != old_intensity) {
      old_intensity = blue;
      pel_set_rgb(255, 0, 255 - blue);
    }
  }

#endif

#if HAS_HEATED_BED

  void PrinterEventLEDs::onBedHeating(const float &start, const float &current, const float &target) {
    const uint8_t red = pel_intensity(start, current, target);
    if (red != old_intensity) {
      old_intensity = red;
      pel_set_rgb(red, 0, 255);
    }
  }
#endif
`

This is where the issue would be if there is one. But this is not from bugfix. I will take a look at that one.

ManuelMcLure, 2020-05-06:
I meant ``WAIT_FOR_BED_HEATER`` - that's the crux of the problem and I think what is confusing the LEDs.

Tannoo, 2020-05-06:
This could be an issue with that.

`static inline void onHeatingDone() { leds.set_white(); }`

ManuelMcLure, 2020-06-21:
I no longer use WAIT_FOR_BED_HEATER in my start gcode, but I still see similar behavior sometimes when heating up the nozzle for a filament swap, so this hasn’t been fixed.

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

ManuelMcLure, 2020-06-29:
This does still happen with today's bugfix-2.0.x as of commit bfcf7ac2fd164cdd079d419ad5ae48a5649de1f5.
This no longer affects me directly since on my printer I no longer use ``PROBING_HEATERS_OFF``/``WAIT_FOR_BED_HEATER`` but I re-enabled them to do the test.

Tygrys-1, 2020-07-28:
Maybe it is from the way how neopixels works. They require the exact, tight timing when setting colors and during the waiting for heated bed/nozzle to reach the temperatures, the timer interrupt is used to check the temp. This may interfere with the neopixels timing, based frequently on the CPU cycle counting. Interrupts cannot be blocked for the neopixels programing time, because it may affect i.e. the soft serial timing for steppers. I do not see the good way to resolve this.

ManuelMcLure, 2020-07-28:
No - this also happens with PWM-driven RGB LEDs. When I originally reported it I was using those, and later I upgraded to NeoPixels and saw the same behavior.

gordo3di, 2020-07-28:
@ManuelMcLure Going through the same issues over here. Using biqu skr v1.4 turbo and neopixels. Just sets the leds to white during heat up instead of going from blue to red. I tried playing with the several things in `printer_event_leds.cpp` and `Printer_event_leds.h` but it's almost like its not even being called. 

Even if I bypass the color change and just set the strip to a color, red for example, in the `PrinterEventLEDs::onHotendHeating(` function it just ignores it.

I've spent a bit of time on this and can't figure it out. I had it working in a previous marlin compile but I can't figure out what I changed. The startup RGB cycle works and I can set the light color from the TFT35 E3 V3 touch screen menu.

Also tried setting the neopixel as the case light with no luck. Tested so much at this point I'm not even sure anymore. Weird.

gordo3di, 2020-07-28:
@ManuelMcLure @Tygrys-1  Interesting... in `printer_event_leds.h` I changed the LED color during printing to red and it stayed white...

```
  #if HAS_TEMP_HOTEND || HAS_HEATED_BED
    static inline void onHeatingDone() { leds.set_red(); } // make it red
    static inline void onPidTuningDone(LEDColor c) { leds.set_color(c); }
  #endif
```

ManuelMcLure, 2020-07-28:
@gordo3di That seems like a different issue. As of bugfix a couple of days ago (last time I did a print) ``PRINTER_EVENT_LEDS`` worked fine at the start of the print. Note that it only takes effect for an ``M109`` or ``M104`` - i.e. the commands that wait for the temperature before returning. The ``M190``/``M140`` commands or setting the temperature from the LCD does not make ``PRINTER_EVENT_LEDS`` work.

ManuelMcLure, 2020-07-28:
I just tested with a quick ``M190 S50`` and my Neopixels turned blue.

gordo3di, 2020-07-28:
@ManuelMcLure Yeah I must have something else going on then. I've tested it by setting M109 S199 and also waited for the print to reach temp and start... I'm sure I turned something off during all my testing I just need to figure out where I did it!

I actually played with neopixel stuff a yew years ago and had it working very well. It would heat from blue to red pixel by pixel and during the print, the light bar acted like a progress bar from white to green so you could see it from far away but all that code is very old... https://github.com/MarlinFirmware/Marlin/issues/8915

ManuelMcLure, 2020-07-28:
One thing to test - I do have _disabled_  ``NEOPIXEL_IS_SEQUENTIAL`` because I think I tried it and it didn't work for me. Try that and see if the results change.

gordo3di, 2020-07-28:
@ManuelMcLure Ah! Turns out it's because I'm using the TFT 35 V3 touch screen. It doesnt like the heating commands coming from it. It worked in an older version of the touchscreen firmware but I guess not the latest and I didn't check to see if that was the issue. When i switch back to the Marlin interface it works now... I'm gonna work on the TFT firmware then... Thanks for the help.

closed 2020-09-03 by github-actions[bot]