MarlinFirmware/Marlin #13430: [FR] Dual motor One Extruder

Simur, 2019-03-18:
Bonjour tout le monde ! 
J'aurais besoin de votre aide ! :D 
J'ai une boite où sont stocké mes filaments. J'aimerais ajouter un moteur pour aider l'extrudeur.
Seulement, comme ils n'ont pas le même ratio, ils n'ont pas les mêmes step/mm.
J'aimerais brancher ce moteur sur E1.
Avez-vous une idée pour paramétrer marlin correctement ?
Merci pour votre aide ! :D 

Hello everyone !
I need your help! : D
I have a box where my filaments are stored. I would like to add a motor to help my extruder.
Only, since they do not have the same ratio, they do not have the same step / mm.
I would like to connect this motor to E1.
Do you have an idea to set marlin correctly?
Thanks for your help ! : D

-Anet A8
-MKS Gen L V1.0
-DRV8825

InsanityAutomation, 2019-03-18:
Multi extruder duplication should accomplish this. We just extended that functionality last week as well. Its essentially doing what these dual stepper functions do and is typically used for multiple hotends on a single x gantry.

Simur, 2019-03-18:
> La duplication multi-extrudeuse devrait accomplir cela. Nous avons également étendu cette fonctionnalité la semaine dernière également. Il fait essentiellement ce que font ces doubles fonctions pas à pas et est généralement utilisé pour plusieurs adresses sur un même portique x.

Et comment le configurer ? 
And how to configure it?

#define E0_DUAL_STEPPER_DRIVERS
#if ENABLED(E0_DUAL_STEPPER_DRIVERS)
  //#define E0_DUAL_ENDSTOPS
  #if ENABLED(E0_DUAL_ENDSTOPS)
    #define E1_USE_ENDSTOP _XMAX_
    #define E1_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif
****

InsanityAutomation, 2019-03-18:
Just set MULTI_NOZZLE_DUPLICATION and use M605

See syntax here

https://github.com/MarlinFirmware/Marlin/pull/13373

Simur, 2019-03-18:
Merci beaucoup, je vais tester ça de suite ! :D 
Thanks ! I go test it ! :D

InsanityAutomation, 2019-03-25:
If the feature above does indeed satisfy this request, please close the issue.

Thanks

closed 2019-05-19 by thinkyhead