MarlinFirmware/Marlin #13432: BLT 3 - compile errors

RudyBenner, 2019-03-18:
Moving my config files to include new code for BLT 3.0

I get compile errors.

Screen shot included in attched ZIP file.

Before and after config files included.

[March 18 BLT 3.zip](https://github.com/MarlinFirmware/Marlin/files/2978233/March.18.BLT.3.zip)

InsanityAutomation, 2019-03-18:
There is a followup PR open and it includes this fix

https://github.com/MarlinFirmware/Marlin/pull/13422

RudyBenner, 2019-03-18:
I will not upload to my board until this is sorted.

InsanityAutomation, 2019-03-18:
Understandable. I dont see any reason for this to not get merged sometime soon.

RudyBenner, 2019-03-18:
I did not know how to merge this, until this morning.
Never let ignorance stand in the way of education.

Compiles and runs fine with my older BLT, I guess 2.2? Plastic pin.
MKS_GEN_L

Thanks for the help.

RudyBenner, 2019-03-18:
Is there a compelling reason other than it's new and SHINY for me to upgrade? Does it do any new tricks?

InsanityAutomation, 2019-03-18:
V3 has an open collector mode for 3.3v or 5v boards to function without hardware changes, and a 1.6mm longer pin stroke. Reason to upgrade would be if you get a V3.0 sensor sometime in the future, current versions will just crash into the bed as the trigger method has changed a bit. For older versions, its totally irrelevant to update to the new code here. New code as called out in the mentioned PR will allow swapping between the two seamlessly.

closed 2019-03-18 by RudyBenner