MarlinFirmware/Marlin #13433: [FR] Ability to disable Z endstop

Mixer9, 2019-03-18:
### Description

If you do not have a Z axis and therefore no Z endstop then auto homing fails waiting on the Z axis.

### Steps to Reproduce

1. Build or purchase an XY plotter, with no Z axis.
2. Attempt to use the "auto home" function in the Marlin firmware. This will fail after timing out waiting on the Z endstop you don't have.

### Suggested Fix
I am suggesting that there should be a way to disable the Z MIN endstops if the "laser" function is enabled, so that the firmware will not timeout with a "printer killed" message.

The other suggestion if you do not wish you give the user a way to disable the Z endstop check during the "auto home" , would be to put a comment into the configuration_adv.h around the laser enable that if it is an XY machine with no Z that a simple jumper between Signal and GND for the Zmin pins will fulfill this check and auto home will work without a Z axis.

#### Additional Information

If you wish to see what type of machine I am referring to see the XY plotter with laser upgrade by makeblock @ https://www.youtube.com/watch?v=Idc2q0KSkAA

AnHardt, 2019-03-18:
`G28 XY`

FlailAway, 2019-03-18:
> `G28 XY`

Serendipity: I was just coming here to ask the same kind of question. Your answer only works if you have a computer connected. If you are using an LCD+encoder that will not work.

I am building a new corexy and I need to do a lot of up-down with the z-axis. I cannot have a computer where I am building the corexy so must only use the LCD+encoder. Currently I can move the axis up but never lower than the start point. I need to be able to change the Configuration.h for unconstrained movement while devleoping this. 

Yes I totally accept all responsibility for crashing into the hotbed or it flying off and hitting the ceiling etc. :)

How can I do that?

boelle, 2019-03-18:
//
// Individual Axis Homing
//
// Add individual axis homing items (Home X, Home Y, and Home Z) to the LCD menu.
//
//#define INDIVIDUAL_AXIS_HOMING_MENU

FlailAway, 2019-03-18:
> //#define INDIVIDUAL_AXIS_HOMING_MENU

Ahhh, OK, thanks. I had to also disable 
#define MIN_SOFTWARE_ENDSTOPS
#define MAX_SOFTWARE_ENDSTOPS

But it now allows me to move up/down anywhere I need. 
Awesome, thanks again.

boelle, 2019-03-18:
you are welcome :-D

reloxx13, 2019-03-18:
but a "Home XY" in INDIVIDUAL_AXIS_HOMING_MENU would still be nice and helpfull for some guys out there.

Alternativly u can write it as user menu item.

7eggert, 2019-03-20:
Another idea, UNTESTED and needs to be double-checked: There is a DIR pin for the Z stepper, and it will toggle between on and off. Maybe this pin can be connected to the Z endstop pin:  Move down == triggered; move up == released.

Nevertheless (and even if this might be low-priority as it can be circumvented for now), supporting two- or even one-dimensional machines one day should be in the code.

ncr, 2023-07-21:
> Another idea, UNTESTED and needs to be double-checked: There is a DIR pin for the Z stepper, and it will toggle between on and off. Maybe this pin can be connected to the Z endstop pin: Move down == triggered; move up == released.

Just tested it and it works, just needed to set this in Configuration.h

#define INVERT_Z_DIR true

Very cool idea @7eggert 👍🏻 Thanks!

thisiskeithb, 2024-09-06:
You can now disable `Z_DRIVER_TYPE` and then work through all the errors of specific defines that use arrays for their values, removing the third value/Z, and you should be set.

Here's a diff of the default config with those changes to give you an idea of some of the required changes (exact changes will vary depending on enabled options):

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 64187f4..ca8fb8e 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -151,7 +151,7 @@
  */
 #define X_DRIVER_TYPE  A4988
 #define Y_DRIVER_TYPE  A4988
-#define Z_DRIVER_TYPE  A4988
+//#define Z_DRIVER_TYPE  A4988
 //#define X2_DRIVER_TYPE A4988
 //#define Y2_DRIVER_TYPE A4988
 //#define Z2_DRIVER_TYPE A4988
@@ -1252,7 +1252,7 @@
  * Override with M92 (when enabled below)
  *                                      X, Y, Z [, I [, J [, K...]]], E0 [, E1[, E2...]]
  */
-#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 500 }
+#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 500 }
 
 /**
  * Enable support for M92. Disable to save at least ~530 bytes of flash.
@@ -1264,7 +1264,7 @@
  * Override with M203
  *                                      X, Y, Z [, I [, J [, K...]]], E0 [, E1[, E2...]]
  */
-#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }
+#define DEFAULT_MAX_FEEDRATE          { 300, 300, 25 }
 
 //#define LIMITED_MAX_FR_EDITING        // Limit edit via M203 or LCD to DEFAULT_MAX_FEEDRATE * 2
 #if ENABLED(LIMITED_MAX_FR_EDITING)
@@ -1277,7 +1277,7 @@
  * Override with M201
  *                                      X, Y, Z [, I [, J [, K...]]], E0 [, E1[, E2...]]
  */
-#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }
+#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 10000 }
 
 //#define LIMITED_MAX_ACCEL_EDITING     // Limit edit via M201 or LCD to DEFAULT_MAX_ACCELERATION * 2
 #if ENABLED(LIMITED_MAX_ACCEL_EDITING)
@@ -2302,7 +2302,7 @@
 #endif
 
 // Homing speeds (linear=mm/min, rotational=°/min)
-#define HOMING_FEEDRATE_MM_M { (50*60), (50*60), (4*60) }
+#define HOMING_FEEDRATE_MM_M { (50*60), (50*60) }
 
 // Validate that endstops are triggered on homing moves
 #define VALIDATE_HOMING_ENDSTOPS
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index 0cd5a64..4828209 100644
--- a/Marlin/Configuration_adv.h
+++ b/Marlin/Configuration_adv.h
@@ -928,8 +928,8 @@
 
 //#define SENSORLESS_BACKOFF_MM  { 2, 2, 0 }  // (linear=mm, rotational=°) Backoff from endstops before sensorless homing
 
-#define HOMING_BUMP_MM      { 5, 5, 2 }       // (linear=mm, rotational=°) Backoff from endstops after first bump
-#define HOMING_BUMP_DIVISOR { 2, 2, 4 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)
+#define HOMING_BUMP_MM      { 5, 5 }       // (linear=mm, rotational=°) Backoff from endstops after first bump
+#define HOMING_BUMP_DIVISOR { 2, 2 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)
 
 //#define HOMING_BACKOFF_POST_MM { 2, 2, 2 }  // (linear=mm, rotational=°) Backoff from endstops after homing
 //#define XY_COUNTERPART_BACKOFF_MM 0         // (mm) Backoff X after homing Y, and vice-versa
@@ -1223,7 +1223,7 @@
 
 // @section motion
 
-#define AXIS_RELATIVE_MODES { false, false, false, false }
+#define AXIS_RELATIVE_MODES { false, false, false }
 
 // Add a Duplicate option for well-separated conjoined nozzles
 //#define MULTI_NOZZLE_DUPLICATION
```

closed 2024-09-06 by thisiskeithb