MarlinFirmware/Marlin #13434: HEATED BED IS NOT HEATİNG

ebubekirn, 2019-03-18:
I am sending this code: M303 E-1 S50 C8, for heating my heatbed. But i get a heating failed error. My heatbed heating when i gave dc power. So there is no mechanical problem to heatbed.

Configuration.h codes like that:

//===========================================================================
//============================= PID > Bed Temperature Control ===============
//===========================================================================

/**
 * PID Bed Heating
 *
 * If this option is enabled set PID constants below.
 * If this option is disabled, bang-bang will be used and BED_LIMIT_SWITCHING will enable hysteresis.
 *
 * The PID frequency will be the same as the extruder PWM.
 * If PID_dT is the default, and correct for the hardware/configuration, that means 7.689Hz,
 * which is fine for driving a square wave into a resistive load and does not significantly
 * impact FET heating. This also works fine on a Fotek SSR-10DA Solid State Relay into a 250W
 * heater. If your configuration is significantly different than this and you don't understand
 * the issues involved, don't use bed PID until someone else verifies that your hardware works.
 */
#define PIDTEMPBED

//#define BED_LIMIT_SWITCHING

/**
 * Max Bed Power
 * Applies to all forms of bed control (PID, bang-bang, and bang-bang with hysteresis).
 * When set to any value below 255, enables a form of PWM to the bed that acts like a divider
 * so don't use it unless you are OK with PWM on your bed. (See the comment on enabling PIDTEMPBED)
 */
#define MAX_BED_POWER 255 // limits duty cycle to bed; 255=full current

#if ENABLED(PIDTEMPBED)

  //#define PID_BED_DEBUG // Sends debug data to the serial port.

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define DEFAULT_bedKp 15.00
  #define DEFAULT_bedKi .04
  #define DEFAULT_bedKd 305.4

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from pidautotune
  //#define DEFAULT_bedKp 97.1
  //#define DEFAULT_bedKi 1.41
  //#define DEFAULT_bedKd 1675.16

  // FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
#endif // PIDTEMPBED

Configuration_adv.h codes like that:

#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

#define WATCH_TEMP_PERIOD 30                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
#endif

#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 20    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius

#define WATCH_BED_TEMP_PERIOD 60                // Seconds
  #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius
#endif

#if ENABLED(PIDTEMP)
  // this adds an experimental additional term to the heating power, proportional to the extrusion speed.
  // if Kc is chosen well, the additional required power due to increased melting should be compensated.
  //#define PID_EXTRUSION_SCALING
  #if ENABLED(PID_EXTRUSION_SCALING)
    #define DEFAULT_Kc (100) //heating power=Kc*(e_speed)
    #define LPQ_MAX_LEN 50
  #endif
#endif

I am waiting for your advices. Thank you

ebubekirn, 2019-03-19:
Do not have any solution about it??

7eggert, 2019-03-19:
Please make sure it's not a mainboard side hardware error and read about how to create a bug report. You didn't even mention which printer you have nor which mainboard you're trying to use nor the things you changed on your wiring.

ebubekirn, 2019-03-19:
I am using DIY delta type 3d printer and ramps 1.4 with arduino mega. I changed my wires and controled mechanical settings. So when i tried to heat bed i am getting heating failed error.

Uwe11, 2019-03-20:
If you get a heating failed error from Marlin, the protection routine stopped heating. This is normally caused by wrong PID parameters for your bed.
You *could* also change the settings that detect if somethings going wrong and make them less sensitive. But that would also increase your risk of something really going wrong.

7eggert, 2019-03-20:
Checklist 1:
1. Does the LED on the heatbed turn on and off when trying to control heating?
1b. Measure the voltage on the pins on the mainboard.
2. Do a PID autotune
2b. Failing 2., change the setting to Bang-Bang or limit switching for testing purposes. It should not matter, but …
3. Is your printer in a very cold room? If the sensor is below a limit, it' assumed to be defective.
…

ebubekirn, 2019-03-20:
When trying to control heating led does not turn on or off and voltage value equal zero. My printer is in room temperature.

boelle, 2019-03-20:
so when bed is trying to heat no voltage to bed? then you have either a blown fuse or bad mosfet

boelle, 2019-03-20:
btw @ebubekirn can you edit title so it becomes [BUG] HEATED BED IS NOT HEATİNG

ebubekirn, 2019-03-21:
I tried different ramps 1.4 but i got same problem. So i think there is no
problem about my mainboard hardware. Thanks for your attention.

şunu yazdı:

7eggert, 2019-03-21:
Did you remember to supply power to the second power socket? The 10 A one powering the heatbed connector?

sorgens, 2019-04-12:
I assume that initially you changed heating bed from bang-bang mode to pid? Have you initialized EEPROM(G-code M502) after re-flashing  board?

shitcreek, 2019-05-15:
@ebubekirn any update?

katyo, 2019-05-25:
I have similar issue with Melzi.
When I connect oscilloscope to gate of heatbed transistor I see short needles of +5V only.
I also tried reflash Repetier back and it works fine with heated bed. It successfully heated to +110C and more.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle