MarlinFirmware/Marlin #13436: [BUG] PREHEAT_1_LABEL and PREHEAT_2_LABEL not defined for German language

Uwe11, 2019-03-19:
### Description

language_de.h uses PREHEAT_1_LABEL and PREHEAT_2_LABEL instead of the constants "PLA" and "ABS". But these two variables are not defined and lead to an error.

### Steps to Reproduce

set language to "de" in configuration.h
#define LCD_LANGUAGE de

Compile the sketch

#### Additional Information
This is my first comment on github. Please excuse me, if not everything is perfect.

Mav987, 2019-03-20:
Hi Uwe,

you can help yourself by entering 

#define PREHEAT_1_LABEL                     _UxGT("PLA")
#define PREHEAT_2_LABEL                     _UxGT("ABS")

in the file language_de.h

[image]

greeting andy

Uwe11, 2019-03-20:
Thanks Andy,
I already did. 
But shouldn't we solve this issue for all?

I don't know, if it was ment to make "PLA" and "ABS" configurable. Then I would expect the definition for 
PREHEAT_1_LABEL and PREHEAT_2_LABEL in configuration.h.
But as far as I've seen, only German language file uses this variables. I'm not sure why there were introduced. 
Either define the name of the two materials in configuration.h for all languages or keep the text strait and fix with PLA and ABS.

BR
Uwe

MoellerDi, 2019-03-20:
IMHO it's already in the configuration.h by default:

```
// @section temperature

// Preheat Constants
#define PREHEAT_1_LABEL       "PLA"
#define PREHEAT_1_TEMP_HOTEND 180
#define PREHEAT_1_TEMP_BED     70
#define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_2_LABEL       "ABS"
#define PREHEAT_2_TEMP_HOTEND 240
#define PREHEAT_2_TEMP_BED    110
#define PREHEAT_2_FAN_SPEED     0 // Value from 0 to 255

```

see https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h#L1386-L1395

Did you use the default configuration.h or one from an example config to build your firmware?

kAdonis, 2019-03-20:
I think PREHEAT_1_LABEL and PREHEAT_2_LABEL are only used in bugfix-2.x 
there is something mixed up, It should not be in language_de.h for bugfix-1.1.x
@Ludy87

reloxx13, 2019-03-20:
but again missing config files so we can just guess

the issue template is not just for fun written.

`* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.`

Ludy87, 2019-03-20:
I have inadvertently entered, apologized.

Bugfix 1.1.x is getting outdated, but I touched the file, but unfortunately with errors.

I will write a new PR afterwards.

Uwe11, 2019-03-20:
ok, Thanks Ludy87!

Yes, we're taking about bugfix-1.1.x in the original version, like here on github. Therefor I thought it's not necessary to include the files here.

kAdonis, 2019-03-20:
@Uwe11 I recommend switching to bugfix-2.x, as Ludy87 said, bugfix-1.1.x is getting outdated

Uwe11, 2019-03-20:
Yes, i know that 1.1 is getting outdated. Just updated from 1.1.1 to 1.1.9 (working) and tried also bugfix 1.1.x (not working)
Till yesterday I thougt 2.0 is ONLY for 32bit. Learned today, that it still supports 8bit.
But is it already so far to give it a try in real world?

kAdonis, 2019-03-20:
> But is it already so far to give it a try in real world?

Absolutely!

Ludy87, 2019-03-20:
PR was created

Ludy87, 2019-03-21:
@Uwe11, du kannst hier zu machen. Der Fehler ist mit der PR gefixt.

Uwe11, 2019-03-21:
besten Dank!

closed 2019-03-21 by Uwe11