MarlinFirmware/Marlin #13442: [BUG] Filament runout state not persisted

guestisp, 2019-03-21:
In v2, the filament runout value is not persisted across reboot.
It starts every time with "On". If I set it to "Off", save settings to EEPROM, on next boot, that flag is still back to on

In Marlin v1 this issue doesn't exists and filament runout value is properly stored and persisted.

InsanityAutomation, 2019-03-21:
In 1.1.9, it cannot disable runout if it's compiled in. It's always on. This was added as a temporary problem solve when a sensor is acting up. Given the limited space in the AVR eeprom, we didn't feel that it was necessary especially as there is a gcode command to disable from your starting script of desired as well.

guestisp, 2019-03-21:
I was referring to 1.1.8 that im currently using and i've disabled the filament sensor just one time. I'm pretty sure but i'll double check in a couple of hours, after the current print

marcio-ao, 2019-04-12:
@InsanityAutomation: I put filament runout stuff in EEPROM in this PR #13659 I am presently working on. It consumes space in EEPROM only if the runout option is enabled.

Oh, and I also added some bytes for ExtUI, which may interest you.

InsanityAutomation, 2019-04-12:
@marcio-ao sounds good. Initially I left it out as it was meant to be a problem solver and incentive to fix the problem but I dont have an issue with it being there if people have desire for it! If the eeprom utilization mentality is shifting to allow optionally consumed space im all for a few more things being able to get saved.

I was watching the other thread a bit as well since I had been told in the past a wrapper for that data wasnt desired so im definitely interested in where that lands!

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle