MarlinFirmware/Marlin #13443: G90/G91 and M82/M83 relative mode set

Msq001, 2019-03-21:
### Description
http://marlinfw.org/docs/gcode/M082.html According to the description of the M82 command in this document, the actual effect of the marlin2.0 firmware is inconsistent. it sat M82 "This command is used to override G91 and put the E axis into absolute mode independent of the other axes", But actually when you use G91 to set "`relative_mode = true"`, even use "M82" to set `axis_relative_modes[E_AXIS] = false`; Extruder coordinates are still relative mode. Because the relative or absolute mode judgment condition in the `void GcodeSuite::get_destination_from_command()` function is `axis_relative_modes[i] || relative_mode`

### So
So. should we modify the document or firmware?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

shitcreek, 2019-09-11:
Thanks for bringing this issue to light, we are rectifying it here: #15218 
From now on, G90/G91 will set all axis to the respective mode and override any M82/M83 that was previously called.

Msq001, 2019-09-11:
Thank you. will close this issue

closed 2019-09-11 by Msq001